nxhack/openwrt-node-packages #5: Chaos Calmer(15.05) still not supported?

The-Alchemist, 2017-01-01:
I was wondering what it would take to support Chaos Calmer.  The README says it's not supported.

Thanks!

nxhack, 2017-01-05:
@The-Alchemist 

Tested for Chaos Calmer. But for some reason only node.js v4.x.

https://github.com/nxhack/openwrt-node-packages/tree/for-15.05

The-Alchemist, 2017-01-05:
Thanks so much!

nxhack, 2017-01-06:
v6.x & upper, build fail in case of using uclibc and gcc(g++) 4.8. You should apply patch on gcc package.

https://github.com/artynet/openwrt-git/blob/openwrt-1505-setup-03/toolchain/gcc/patches/4.8-linaro/852-libstdcxx-uclibc-c99.patch

nxhack, 2017-01-10:
and also apply patches on uclibc package.

https://github.com/artynet/openwrt-git/blob/openwrt-1505-setup-03/toolchain/uClibc/patches-0.9.33.2/L001-uClibc-add-execvpe-function.patch
https://github.com/artynet/openwrt-git/blob/openwrt-1505-setup-03/toolchain/uClibc/patches-0.9.33.2/L002-libm_Add_missing_C99_float_ld_wrappers.patch
https://github.com/artynet/openwrt-git/blob/openwrt-1505-setup-03/toolchain/uClibc/patches-0.9.33.2/L003-force-obstack-free-strong-alias.patch

ZackMattor, 2017-02-08:
@The-Alchemist have you been able to compile for v6 or v7?

The-Alchemist, 2017-02-08:
@ZackMattor : Sorry, I'm short on time.  I haven't tried. :(

ZackMattor, 2017-02-08:
Alright, well we're working on it. I'll report back here if we get it. Might have some questions @nxhack 😉

ZackMattor, 2017-02-08:
@The-Alchemist not sure if you care anymore... But we seem to have it building, Master open-wrt and node v7.5.0 ;)

The-Alchemist, 2017-02-09:
@ZackMattor : That's excellent news!  I'm super excited!   Do you remember how big it was?

I'm hoping to be able to stick it on a 32MB Omega2+.

ZackMattor, 2017-02-10:
@The-Alchemist It will definitely fit within 32mb! The total size we ended up with before trimming stuff was 17mb for the whole image (1mb outside of our flash) by removing some unneeded stuff we got it below. I think the node size was around 6mb.

cooboyno111, 2017-06-13:
If you don't need icu&i18n .
you can remove DEPENDS "+NODEJS_ICU:icu"
rom size 10m+。

artynet, 2017-06-20:
@ZackMattor @cooboyno111 please rebuild the toolchain with the above patches and then build the node packages :

https://github.com/artynet/openwrt-git/blob/openwrt-1505-setup-04/package/linino/node/Makefile
https://github.com/artynet/openwrt-git/blob/openwrt-1505-setup-04/package/linino/node7/Makefile

Best, Arturo

cooboyno111, 2017-07-04:
I successfully compiled nodejs6. on Chaos Calmer(15.05) . You need to change toolchain from uclibc0.9 to uclibc-ng 1.0.6. You need to remove ICU support. The librpc version of the library that needs to be modified is 2015-11-04.

artynet, 2017-07-04:
@cooboyno111 the L00X patches are needed to add those functionalities to the legacy version of uClibc. I have also the patches to build the latest release of node v8 as well 👍 

https://github.com/artynet/openwrt-git/tree/openwrt-1505-setup-04/package/linino/node8/patches

take a look at the link above ;)

Best, Arturo

cooboyno111, 2017-07-05:
@artynet I'm going to test your patch. But I think uclibc 0.99 is too old.

nxhack, 2017-08-02:
for-15.05 branch updated. 

Aggregated the necessary patches for build latest node.js version .
https://github.com/nxhack/openwrt-node-packages/blob/for-15.05/for_building_latest_node.patch

These patches are based on the work of the OpenWrt/LEDE developers and the buildroot developers and @artynet

closed 2018-02-14 by nxhack