nxhack/openwrt-node-packages #6: Problem building in openwrt.

d30jeff, 2017-03-01:
Hi, 

I have added this line
`src-git node https://github.com/nxhack/openwrt-node-packages.git` to my feeds.conf.default file.

Ran:

```
./scripts/feeds update node
./scripts/feeds install -a -p node
```
But when I `make` openwrt, my node-v4.8.0 directory only contains one folder called 'npm-cache'.

Did I miss any steps?

Thanks

nxhack, 2017-03-01:
@d30jeff 
Perhaps you may be choosing 'node' of "openwrt/packages".

Please try following instruction.
```
./scripts/feeds update node
rm  ./package/feeds/packages/node
rm  ./package/feeds/packages/node-arduino-firmata
rm  ./package/feeds/packages/node-cylon
rm  ./package/feeds/packages/node-hid
rm  ./package/feeds/packages/node-serialport
./scripts/feeds install -a -p node
```

d30jeff, 2017-03-01:
@nxhack It worked. 
Thank you so much :'-)

closed 2017-03-01 by d30jeff