nxhack/openwrt-node-packages #8: Does this support linkit smart 7688?

yangpu, 2017-03-14:
You may save my time if this support linkit smart 7688.

nxhack, 2017-03-14:
I don't have linkit smart 7688. Please try it and Please report it.
Regards

nxhack, 2017-03-15:
@yangpu 
Compile was successful.
But because I don't have target machine, I am unable to run test.
```
LEDE trunk r3748-48ae44d
CONFIG_TARGET_BOARD="ramips"
CONFIG_TARGET_SUBTARGET="mt7688"
CONFIG_TARGET_PROFILE="DEVICE_LinkIt7688"
CONFIG_TARGET_ARCH_PACKAGES="mipsel_24kc"
```

jcrben, 2017-06-12:
@nxhack I noticed that Atheros AR933x has a caveat - but what hardware are you testing or developing on?

nxhack, 2017-06-12:
@jcrben 
It is running on "Arduino Yun". Of course kernel FPU emulation is enabled.
```
system type             : Atheros AR9330 rev 1
machine                 : Arduino Yun
cpu model               : MIPS 24Kc V7.4
```

nxhack, 2017-07-13:
There was a report that it was running at "ramips". So I will close this issue.

Cleanshooter, 2017-07-19:
I would love to test this I have a LInkit 7688 lying around but I'm not sure how to run the compiler... Any getting started tips? Would it require me to flash my LinkIt with LEDE or can it run directly on OpenWRT?

nxhack, 2017-07-19:
Hi @Cleanshooter 

Please read the nice documentations : https://lede-project.org/faq/before_installation
Read [this](https://github.com/nxhack/openwrt-node-packages/blob/master/README.md) for building custom packages for node.js.
Custom packages can be installed with luci.
Enjoy!

nxhack, 2017-07-20:
If you want to rebuild mediatek 's firmware.

Read 'LinkIt Smart 7688 Developer’s Guide', '7. How to Build Firmware and Bootloader'  : https://docs.labs.mediatek.com/resource/linkit-smart-7688/en/documentation

and use https://github.com/nxhack/openwrt-node-packages/tree/for-15.05

Hopefully it may work. I will try a some. wait a moment.

Cleanshooter, 2017-07-20:
@nxhack thanks for the reply... yeah I was looking at the first few links for lede and I'm just not well versed enough with the project to understand it's setup.  Plus all the getting started stuff looked like I needed to install it on my device which I'd rather avoid.  

I'm going to try rebuilding the MediaTek firmware just to ensure compatibility.  I'll let you know how it goes.

nxhack, 2017-07-21:
Hi @Cleanshooter 
MediaTek [Build the firmware from source codes](https://docs.labs.mediatek.com/resource/linkit-smart-7688/en/tutorials/firmware-and-bootloader/build-the-firmware-from-source-codes)  + add some modifications.
 
1. Install prerequisite packages to build the firmware: 
```
sudo apt-get install git g++ libncurses5-dev subversion libssl-dev gawk libxml-parser-perl unzip wget
```
 1-1. Make working directory
```
mkdir Build_Path_Some_Where
cd Build_Path_Some_Where
```
2. Download the OpenWrt CC source codes: 
```
git clone git://git.openwrt.org/15.05/openwrt.git
```
3. Prepare the default configuration file for feeds: 
```
cd openwrt
cp feeds.conf.default feeds.conf
```
4. Add the LinkIt Smart 7688 development board's feed: 
```
echo 'src-git linkit https://github.com/MediaTek-Labs/linkit-smart-7688-feed.git' >> feeds.conf
```
4-1. Add the node.js packages feed:
```
echo 'src-git node https://github.com/nxhack/openwrt-node-packages.git;for-15.05' >> feeds.conf
```
5. Update the feed information for all available packages to build the firmware: 
```
./scripts/feeds update
```
5-1. Change the packages installed as default: 
```
wget https://gist.githubusercontent.com/nxhack/7ced4d866a59ebc72737589b49a220f8/raw/1bebfe9f6091f55a5856bc4af00da456a4995b09/mtk-linkit.patch
patch -p1 < mtk-linkit.patch
```
5-2. Copy kernel objects for support kernel 3.18.45: 
        see : https://github.com/MediaTek-Labs/linkit-smart-7688-feed/issues/37
```
cp ./feeds/linkit/mtk-sdk-wifi/wifi_binary/mt_wifi.ko_3.18.44 ./feeds/linkit/mtk-sdk-wifi/wifi_binary/mt_wifi.ko_3.18.45
cp ./feeds/linkit/mtk-sdk-wifi/wifi_binary/mt_wifi.ko_3.18.44_all ./feeds/linkit/mtk-sdk-wifi/wifi_binary/mt_wifi.ko_3.18.45_all
```
5-3. Fix build error of depend on node.js version: 
```
wget https://gist.githubusercontent.com/nxhack/4357d51918ba8f3cb5cc00080ad0815c/raw/e96758224ade8cb224523aedf5ea3249a6a59425/MRAA.patch
patch -p1 < MRAA.patch
```
6. Install all packages: 
```
./scripts/feeds install -a
```
6-1. Use node.js custom packages: 
```
rm ./package/feeds/packages/node
rm ./package/feeds/packages/node-arduino-firmata
rm ./package/feeds/packages/node-cylon
rm ./package/feeds/packages/node-hid
rm ./package/feeds/packages/node-serialport
./scripts/feeds install -a -p node
```
7. Prepare the kernel configuration:
```
make menuconfig
```

> Select the following options:
> 
>  Target System: Ralink RT288x/RT3xxx
>  Subtarget: MT7688 based boards
>  Target Profile: LinkIt7688
> 

7-1. Select optional packages of node.js custom feed.

>  Save and exit (use the default configuration file without any modification)

8. Start the compilation process:
```
make V=99
```

Cleanshooter, 2017-07-22:
@nxhack You found some of the pieces I was missing... I got some of them but not all the stuff you did above. 

I just finished a build of my own and right now and I'm trying to flash it... but it just won't take.  

I'm going to start a new build with some of your suggestions and let it run over night...

Cleanshooter, 2017-07-22:
@nxhack how did you find out about those lovely patches ;)

Cleanshooter, 2017-07-22:
@nxhack I'm concerned by this output...
```
moto@openwrt-compiler:~/openwrt$ ./scripts/feeds install -a -p node
Collecting package info: done
Installing all packages from feed node.
Installing package 'node'
Installing package 'node-arduino-firmata'
Installing package 'node-cylon'
Installing package 'node-cylon-firmata'
Installing package 'node-cylon-gpio'
Installing package 'node-cylon-i2c'
Installing package 'node-serialport'
```

Shouldn't I be seeing the packages form this library?

Cleanshooter, 2017-07-22:
7-1. Select optional packages of node.js custom feed.

Save and exit (use the default configuration file without any modification)

How do I do this?  In the makeconfig?  What menu options do I need to use?
Do I go to Languages and select Node.js and then select all those libraries? 
Not super familiar with this utility.

nxhack, 2017-07-24:
Hi @Cleanshooter 
> How do I do this? In the makeconfig? What menu options do I need to use?
> Do I go to Languages and select Node.js and then select all those libraries?
> Not super familiar with this utility.

In the node custom feed, if there is a module you want, select it.
```
make menuconfig
```

Languages -> Node.js -> (Select module that you want)

If 'Y' it will be put into firmware. 'M' will be only built, not put into firmware.

Cleanshooter, 2017-07-24:
Hey I just wanted you to know this works great!  Would you have any objections to me posting the built image on my blog for people to download and flash to their firmware or would you rather host it?

Also, I'm curious though is it possible to do a build for v6?

nxhack, 2017-07-25:
> Hey I just wanted you to know this works great! Would you have any objections to me posting the built image on my blog for people to download and flash to their firmware or would you rather host it in the repo?

Congrats!　And Now, You are not a beginner!
Open your knowledge is wonderful.
But I don't know license issue of LInkit firmware. Please ask mediatek.

> I'm guessing this would mean another patch to the openwrt repo...

https://github.com/nxhack/openwrt-node-packages/blob/for-15.05/README.md
[Chaos Calmer(15.05) still not supported?](https://github.com/nxhack/openwrt-node-packages/issues/5)

Please try, Please challenge.

cooboyno111, 2017-07-25:
I can run nodejs v6 on mt7628 chip。
I don't have a mt7688 device on this side。

Cleanshooter, 2017-07-25:
@cooboyno111 

I'd love to know what changes to made to the repo to compile in node v6. 

Did/could you fork the repo so that I could look at your changes?

nxhack, 2017-07-27:
@Cleanshooter 
The 'for-15.05' branch is modified.
Prepared necessary patches for build latest node.js version.

Please following steps.
```
./scripts/feeds update node
make oldconfig
patch -p1 < ./feeds/node/for_building_latest_node.patch
make toolchain/clean
make
```
You can select version of node.js in menuconfig.
(default version is v6)

Cleanshooter, 2017-07-27:
DUDE!!! The updates you've made are great!  I'm running the build right now.  I'll let you know how it goes.  It took me a while to find the version selector in the make menuconfig but I see you defaulted to v6 (<-- Thx). 

@nxhack You made my day. I'm hoping with this I'll finally be able to run my app on my LinkIt 7688.   I was really bummed out when I saw that openWRT only had Node 0.12.7 OOTB and it took a lot of digging to find your wonderful repo.  Thank you so much for sharing it! A lot of my code and dependencies needed Node v6, which I only found out after a few failed compile attempts.

Cleanshooter, 2017-07-27:
Worked!  I'm about to test my app on it now.  I would note to others that you can't fit all the packages on the LinkIt 7688 though.  The entire thing has to be under 30MB (I think) and if you add too many node libraries it will blow up on you.  All I added was NPM and it came out to 25MB.

nxhack, 2017-07-27:
@Cleanshooter 
If you want extend target machine's filesystem, You can use extroot mechanizm.
https://docs.labs.mediatek.com/resource/linkit-smart-7688/en/tutorials/file-storage/mounting-the-root-fs-on-an-sd-card

cooboyno111, 2017-07-28:
@Cleanshooter remove ICU,remove NPM. LinkIt 7688 9.4M

nxhack, 2017-07-28:
Oh, libmraa, libupm build will fail in situation of node.js v8.x.
But I do not care at this point. ;)

Cleanshooter, 2017-07-31:
@cooboyno111 I might remove them but for some that might not be a viable option as npm is used by some packages/software during run time.

cooboyno111, 2017-08-01:
@Cleanshooter In ejdb, for example, you need to understand the file structure of node_modules and what can be streamlined.
i remove 3 line
add 1line
//var binary = require('node-pre-gyp');
//var path = require('path');
//var ejdblib_path = binary.find(path.resolve(path.join(__dirname, './package.json')));
var ejdblib = require('./ejdb_native.node');
Only three files were eventually committed to the firmware
ejdb.js ejdb_native.node package.json
Original program 45m。
Streamlined 1.5m

closed 2018-02-14 by nxhack

tcpipchip, 2018-03-24:
Does it works to LEDE ? I Got illegal instruction when runs Node

tcpipchip, 2018-03-24:
I mean node 8 on lede on linkit smart 7688

nxhack, 2018-03-24:
I have not tested, but let's hack and try hard by referring to this.

https://github.com/changyuheng/linkit-smart-7688-feed

And, Be sure to enable MIPS_FPU_EMULATOR kernel option.

tcpipchip, 2018-03-24:
I will try again soon! That procedure includes Node 8 package and libmraa ?
Thank you so much! I hope to use your amazing azure amazon ibm packages :)

nxhack, 2018-03-25:
> That procedure includes Node 8 package and libmraa ?

No.

enable node.js v8.x :
```
echo 'src-git node https://github.com/nxhack/openwrt-node-packages.git' >> feeds.conf
./scripts/feeds update node
rm ./package/feeds/packages/node
rm ./package/feeds/packages/node-*
./scripts/feeds install -a -p node
make menuconfig
```
Please do the related hack described above.

libmraa's node.js v8.x support seems to be difficult. hack please and Good luck!

tcpipchip, 2018-03-25:
Thanks! I will report you monday!

tcpipchip, 2018-03-27:
still getting ILLEGAL INTRUCTION when runs NODE

i enable the FPU EMULATION on make kernel_menuconfig

nxhack, 2018-03-28:
@tcpipchip 

Please tell me the result of executing this command on your target machine.
```
ls -l /sys/kernel/debug/mips/fpuemustats
```
```
node --version
```
```
npm --version
```

tcpipchip, 2018-03-28:
I compiled again node v8 to linkit smart 7688, but now on ubuntu 17. And i had to remove some package dependencies of node! I guess that now will works! https://github.com/MediaTek-Labs/linkit-smart-7688-feed?files=1
Tomorrow i will test!

tcpipchip, 2018-03-28:
Now all is working! Linkit running LEDE with NODE V8 :)

tcpipchip, 2018-03-28:
Btw, thank you for all the help! next step to see if libmraa compile

tcpipchip, 2018-03-29:
LIBMRAA not works on LEDE.
Did you get some progress there ?

tcpipchip, 2018-04-02:
Still no success on LIBMRAA here :(

btw, about 
node-node-red-contrib-gpio
How can i do the NODE-RED see this NODE in the NODE-RED WEB ?

nxhack, 2018-04-12:
@tcpipchip 

Please test this.
https://github.com/nxhack/openwrt-intel-iot-devkit

tcpipchip, 2018-04-12:
Doing now the compile!

tcpipchip, 2018-04-12:
I enabled the libmraa on make menuconfig, i show a new error on LIBHID

I will update -a all and try again now!

tcpipchip, 2018-04-12:
make[4]: Entering directory '/home/osboxes/Lede/linkit-smart-7688-lede/build_dir/target-mipsel_24kc_musl-1.1.16/hidapi-0.8.0-rc1/libusb'
make[4]: 'libhidapi-libusb.so' is up to date.
make[4]: Leaving directory '/home/osboxes/Lede/linkit-smart-7688-lede/build_dir/target-mipsel_24kc_musl-1.1.16/hidapi-0.8.0-rc1/libusb'
mv /home/osboxes/Lede/linkit-smart-7688-lede/build_dir/target-mipsel_24kc_musl-1.1.16/hidapi-0.8.0-rc1/libusb/libhidapi-libusb.so /home/osboxes/Lede/linkit-smart-7688-lede/build_dir/target-mipsel_24kc_musl-1.1.16/hidapi-0.8.0-rc1/libusb/libhidapi-libusb.so.0
mv: '/home/osboxes/Lede/linkit-smart-7688-lede/build_dir/target-mipsel_24kc_musl-1.1.16/hidapi-0.8.0-rc1/libusb/libhidapi-libusb.so' and '/home/osboxes/Lede/linkit-smart-7688-lede/build_dir/target-mipsel_24kc_musl-1.1.16/hidapi-0.8.0-rc1/libusb/libhidapi-libusb.so.0' are the same file
Makefile:77: recipe for target '/home/osboxes/Lede/linkit-smart-7688-lede/build_dir/target-mipsel_24kc_musl-1.1.16/hidapi-0.8.0-rc1/.built' failed
make[3]: *** [/home/osboxes/Lede/linkit-smart-7688-lede/build_dir/target-mipsel_24kc_musl-1.1.16/hidapi-0.8.0-rc1/.built] Error 1
make[3]: Leaving directory '/home/osboxes/Lede/linkit-smart-7688-lede/feeds/packages/libs/hidapi'
package/Makefile:105: recipe for target 'package/feeds/packages/hidapi/compile' failed
make[2]: *** [package/feeds/packages/hidapi/compile] Error 2
make[2]: Leaving directory '/home/osboxes/Lede/linkit-smart-7688-lede'
package/Makefile:101: recipe for target '/home/osboxes/Lede/linkit-smart-7688-lede/staging_dir/target-mipsel_24kc_musl-1.1.16/stamp/.package_compile' failed
make[1]: *** [/home/osboxes/Lede/linkit-smart-7688-lede/staging_dir/target-mipsel_24kc_musl-1.1.16/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/osboxes/Lede/linkit-smart-7688-lede'
/home/osboxes/Lede/linkit-smart-7688-lede/include/toplevel.mk:205: recipe for target 'world' failed
make: *** [world] Error 2

tcpipchip, 2018-04-12:
Second time

make[4]: Entering directory '/home/osboxes/Lede/linkit-smart-7688-lede/build_dir/target-mipsel_24kc_musl-1.1.16/hidapi-0.8.0-rc1/libusb'
make[4]: 'libhidapi-libusb.so' is up to date.
make[4]: Leaving directory '/home/osboxes/Lede/linkit-smart-7688-lede/build_dir/target-mipsel_24kc_musl-1.1.16/hidapi-0.8.0-rc1/libusb'
mv /home/osboxes/Lede/linkit-smart-7688-lede/build_dir/target-mipsel_24kc_musl-1.1.16/hidapi-0.8.0-rc1/libusb/libhidapi-libusb.so /home/osboxes/Lede/linkit-smart-7688-lede/build_dir/target-mipsel_24kc_musl-1.1.16/hidapi-0.8.0-rc1/libusb/libhidapi-libusb.so.0
mv: '/home/osboxes/Lede/linkit-smart-7688-lede/build_dir/target-mipsel_24kc_musl-1.1.16/hidapi-0.8.0-rc1/libusb/libhidapi-libusb.so' and '/home/osboxes/Lede/linkit-smart-7688-lede/build_dir/target-mipsel_24kc_musl-1.1.16/hidapi-0.8.0-rc1/libusb/libhidapi-libusb.so.0' are the same file
Makefile:77: recipe for target '/home/osboxes/Lede/linkit-smart-7688-lede/build_dir/target-mipsel_24kc_musl-1.1.16/hidapi-0.8.0-rc1/.built' failed
make[3]: *** [/home/osboxes/Lede/linkit-smart-7688-lede/build_dir/target-mipsel_24kc_musl-1.1.16/hidapi-0.8.0-rc1/.built] Error 1
make[3]: Leaving directory '/home/osboxes/Lede/linkit-smart-7688-lede/feeds/packages/libs/hidapi'
package/Makefile:105: recipe for target 'package/feeds/packages/hidapi/compile' failed
make[2]: *** [package/feeds/packages/hidapi/compile] Error 2
make[2]: Leaving directory '/home/osboxes/Lede/linkit-smart-7688-lede'
package/Makefile:101: recipe for target '/home/osboxes/Lede/linkit-smart-7688-lede/staging_dir/target-mipsel_24kc_musl-1.1.16/stamp/.package_compile' failed
make[1]: *** [/home/osboxes/Lede/linkit-smart-7688-lede/staging_dir/target-mipsel_24kc_musl-1.1.16/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/osboxes/Lede/linkit-smart-7688-lede'
/home/osboxes/Lede/linkit-smart-7688-lede/include/toplevel.mk:205: recipe for target 'world' failed
make: *** [world] Error 2

tcpipchip, 2018-04-12:
Removed HIDLIB from menuconfig and compiled. need to test the bin :)

tcpipchip, 2018-04-13:
libupm, libmraa installed!  I need to test with node-red

nxhack, 2018-04-18:
@tcpipchip 
Did the running test succeed?
If it works, I will send pull request it in the official package of OpenWrt.

tcpipchip, 2018-04-18:
@nxhack 
I still tryng to make python import the LIBUPM
When try to  install the LIBUPM lib, said that needs "libupm-" dependencies and not install.
If i install the the dependecies libupm- first, when i try to INSTALL libupm, said the

See
root@mylinkit:/# opkg install libupm_1.6.0-2_mipsel_24kc.ipk
Installing libupm (1.6.0-2) to root...
.
.
.* check_data_file_clashes: Package libupm wants to install file /usr/lib/python2.7/site-packages/upm/pyupm_wt5001.py
        But that file is already provided by package  * libupm-wt5001
 * check_data_file_clashes: Package libupm wants to install file /usr/lib/python2.7/site-packages/upm/pyupm_xbee.py
        But that file is already provided by package  * **libupm-xbee**
 * check_data_file_clashes: Package libupm wants to install file /usr/lib/python2.7/site-packages/upm/pyupm_yg1006.py
        But that file is already provided by package  * libupm-yg1006
 * check_data_file_clashes: Package libupm wants to install file /usr/lib/python2.7/site-packages/upm/pyupm_zfm20.py
        But that file is already provided by package  * libupm-zfm20
 * opkg_install_cmd: Cannot install package libupm.

for example, to the package **libupm-xbee**, what must i make as reference to 
python on import instruction ?

import **??????**

nxhack, 2018-04-19:
I do not have mt7688, so I can not confirm it.
Please check first that gpio is recognized by OS.
```
ls -l /sys/class/gpio/
```

tcpipchip, 2018-04-19:
Only 3 gpios! 
Is there some option to enable on make menuconfig ?

nxhack, 2018-04-19:
check
```
ls -l /sys/devices/platform
```

nxhack, 2018-04-19:
FYI:
https://github.com/intel-iot-devkit/mraa/issues/620
https://wiki.openwrt.org/doc/hardware/port.gpio

tcpipchip, 2018-04-19:
ls -l /sys/devices/platform
drwxr-xr-x   13 root     root             0 Jan  1  1970 10000000.palmbus
drwxr-xr-x    3 root     root             0 Jan  1  1970 10100000.ethernet
drwxr-xr-x    2 root     root             0 Jan  1  1970 10110000.esw
drwxr-xr-x    3 root     root             0 Jan  1  1970 10120000.usbphy
drwxr-xr-x    3 root     root             0 Jan  1  1970 10130000.sdhci
drwxr-xr-x    3 root     root             0 Jan  1  1970 101c0000.ehci
drwxr-xr-x    3 root     root             0 Jan  1  1970 101c1000.ohci
drwxr-xr-x    5 root     root             0 Jan  1  1970 10300000.wmac
drwxr-xr-x    2 root     root             0 Jan  1  1970 alarmtimer
drwxr-xr-x    2 root     root             0 Jan  1  1970 bootstrap
drwxr-xr-x    2 root     root             0 Jan  1  1970 clkctrl
drwxr-xr-x    2 root     root             0 Jan  1  1970 cpuintc@0
drwxr-xr-x    2 root     root             0 Jan  1  1970 gpio-keys-polled
drwxr-xr-x    2 root     root             0 Jan  1  1970 gpio-leds
drwxr-xr-x    2 root     root             0 Jan  1  1970 gpio-wifi
drwxr-xr-x    2 root     root             0 Jan  1  1970 pinctrl
drwxr-xr-x    2 root     root             0 Apr 19 10:50 regulatory.0
drwxr-xr-x    2 root     root             0 Jan  1  1970 rstctrl
drwxr-xr-x    2 root     root             0 Jan  1  1970 serial8250
-rw-r--r--    1 root     root          4096 Apr 19 10:50 uevent

tcpipchip, 2018-04-19:
I will update the KERNEL version to see if works!

tcpipchip, 2018-04-21:
no sucess! is there anybody here and make the LEDE works with LIBMRAA. Compiles, install, but not recognizers gpios, i2c, spi, pwm, etc :(

nxhack, 2018-04-22:
@tcpipchip 

Please continue to ...
https://github.com/nxhack/openwrt-intel-iot-devkit/issues/1

nxhack, 2018-07-04:
For the latest information, please refer to wiki.
https://github.com/nxhack/openwrt-node-packages/wiki

tcpipchip, 2018-07-16:
Did you port to  Chaos Calmer too ?

nxhack, 2018-07-16:
Official OpenWrt Chaos Calmer 15.05.1 isn't supported and deprecated.
I will never port my patches to official CC repository.

[1 more reply]