nxhack/openwrt-node-packages #9: Segmentation fault

cszp17, 2017-03-17:
hi,
     my cpu is ar9331.
     run node -v ,I can get return v6.10.0 .   But when i run others,like npm -v, return "segmentation fault",
i don't have any ideal,can you help me ?

nxhack, 2017-03-22:
@cszp17 
Upstream has been updated to v6.10.1. Please test it.

nxhack, 2017-06-15:
It worked on my environment.
```
# node -v
v6.10.3
# npm -v
3.10.10
```

possibility of memory shortage?

ReleaseBug, 2017-07-08:
hi,  I got the same problem.
My CPU is mt7620a,128MB RAMm,32MB Flash.
node v6.9.1.
run node -v return v6.9.1.
run npm -v return segmentation fault.
run node test.js(console.log("Hello World");) return segmentation fault.

 any idea about this？thx~

nxhack, 2017-07-09:
It may be related to https://github.com/nodejs/node/pull/13871

nxhack, 2017-07-11:
Please test and report at latest version.

ReleaseBug, 2017-07-11:
Good News! 
I tested v8.1.3, node and npm have no segmentation fault error anymore!
(CPU mt7620a,128MB RAM,32MB Flash)
test result:
run node -v return v8.1.3.
run npm -v return 5.3.?（oops,,,I  forgot.....）
run node test.js(console.log("Hello World");) return Hello World.
Everything looks good.
-----------------------------------------------------------
Your project is great , and it's very convenient to use.
Thx for your contribution and your reply.
-----------------------------------------------------------
My next step is t to use node-homebridge.
Hope everything is OK.

nxhack, 2017-07-13:
https://github.com/nodejs/node/commit/47e702059d2360f5e2dec9c05e829222b1c8f9aa
This patch applies to the next release v6.11.2. Please test it when v6.11.2 comes out.

nxhack, 2017-08-02:
@cszp17 @ReleaseBug 
v6.11.2 is released. Please test.

nxhack, 2017-08-04:
```
# node -v
v6.11.2
# npm -v
3.10.10
# node
> console.log("Hello World");
Hello World
undefined
```

nxhack, 2017-09-04:
No news is good news.

closed 2017-09-04 by nxhack

abo-github, 2018-08-01:
```
# node -v
v6.11.2
```
content of my **hello.js** file:

`console.log("Hello World");`

but there's still the same error:

```
# node hello.js
Segmentation fault
```

nxhack, 2018-08-01:
Hi @abo-github 
The latest version of v6.x is v6.14.3.