nxhack/openwrt-node-packages #10: Illegal instruction in bcm4709c0

rench, 2017-05-19:
hello,nxhack:
 i used your git repository to build nodejs v6 into lede,the soc is bcm4709c0，buid success，
but in shell,when run node app.js ,error happens. **Illegal instruction**
can you help me ?

rench, 2017-05-20:
CONFIG_TARGET_BOARD="bcm53xx"
CONFIG_TARGET_SUBTARGET="generic"
CONFIG_TARGET_PROFILE="DEVICE_phicomm-k3"
CONFIG_TARGET_ARCH_PACKAGES="arm_cortex-a9"
CONFIG_DEFAULT_TARGET_OPTIMIZATION="-Os -pipe -mcpu=cortex-a9"
CONFIG_CPU_TYPE="cortex-a9"
CONFIG_LINUX_4_9=y

toolchain-arm_cortex-a9_gcc-5.4.0_musl_eabi

nxhack, 2017-05-22:
@rench 

Please test kernel Floating point emulation

`make kernel_menuconfig`

and Select 'Floating point emulation' 
enable 'VFP-format floating point maths'

rench, 2017-05-22:
@nxhack   
u r right.
bcm4709c0 has no fpu or vfp features。
i just works it on node 0.10.x.

Do you know if other methods can bypass the latest version of V8 that requires the demand for fpu. I looked at some of the node project issue, as if there is no way.

K3 is my new to buy a router, his master chip is bcm4709c0, I would like to run in the above 'node', and then access some intelligent devices. But now it seems that there is no hope in the above run node 4.x above the version.
But I still have a model qihoo C301 router, his master chip is ar9344, MIPS 72Kc , you know whether this CPU can run node 4.x version of it?

thks!

rench, 2017-05-22:
K3 
root@LEDE:~# cat /proc/cpuinfo 
processor	: 0
model name	: ARMv7 Processor rev 0 (v7l)
BogoMIPS	: 1400.00
Features	: half thumb fastmult edsp tls 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x3
CPU part	: 0xc09
CPU revision	: 0

processor	: 1
model name	: ARMv7 Processor rev 0 (v7l)
BogoMIPS	: 1400.00
Features	: half thumb fastmult edsp tls 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x3
CPU part	: 0xc09
CPU revision	: 0

Hardware	: BCM5301X
Revision	: 0000
Serial		: 0000000000000000

qihoo C301:

root@OpenWrt:~# cat /proc/cpuinfo 
system type		: Atheros AR9344 rev 3
machine			: Qihoo 360 C301
processor		: 0
cpu model		: MIPS 74Kc V4.12
BogoMIPS		: 278.93
wait instruction	: yes
microsecond timers	: yes
tlb_entries		: 32
extra interrupt vector	: yes
hardware watchpoint	: yes, count: 4, address/irw mask: [0x0ffc, 0x0ffc, 0x0ffb, 0x0ffb]
isa			: mips1 mips2 mips32r1 mips32r2
ASEs implemented	: mips16 dsp dsp2
shadow register sets	: 1
kscratch registers	: 0
package			: 0
core			: 0
VCED exceptions		: not available
VCEI exceptions		: not available

nxhack, 2017-05-23:
@rench 
In the case of no-fpu MIPS cpu, I think that it will solve with kernel fpu emutation.

`make kernel_menuconfig`

Kernel type -> enable 'MIPS FPU Emulator'

rench, 2017-05-27:
@nxhack  
thanks,i work it out on my c301 router which with ar9344.

i install "node-homebridge" into router,but i find this package $(LN)  executeable 
"homebridge" to the wrong path.

```
define Package/node-homebridge/install
	$(INSTALL_DIR) $(1)/usr/lib/node
	$(CP) $(PKG_INSTALL_DIR)/usr/lib/node_modules/* $(1)/usr/lib/node/
	$(INSTALL_DIR) $(1)/usr/bin
	$(LN) ../lib/node_modules/homebridge/bin/homebridge $(1)/usr/bin/homebridge
endef
```

```
define Package/node-homebridge/install
	$(INSTALL_DIR) $(1)/usr/lib/node
	$(CP) $(PKG_INSTALL_DIR)/usr/lib/node_modules/* $(1)/usr/lib/node/
	$(INSTALL_DIR) $(1)/usr/bin
	$(LN) ../lib/node/homebridge/bin/homebridge $(1)/usr/bin/homebridge
endef

```

i update "node-homebridge" to 0.4.20.and fix the $(LN) path bug.
i make a pr,please check it.

nxhack, 2017-05-28:
@rench Marged. thank you!

closed 2017-05-29 by rench