nxhack/openwrt-node-packages #15: i come again!

rench, 2017-06-30:
hello, i want to know if there any way to modify node source code or kernel to support soft vfp3 on bcm4709 ?

ar9344 can enable MIPS FPU Emulator for soft PFU？
can  bcm53xx do this?

nxhack, 2017-07-03:
Hi @rench 
LEDE 17.01-head (linux-4.4.74)

* bcm4709 and  bcm53xx
   CONFIG_TARGET_BOARD="bcm53xx"
make kernel_menuconfig
-> Select Floating point emulation
-> Select VFP-format floating point maths

* ar9344
  CONFIG_TARGET_BOARD="ar71xx"
make kernel_menuconfig
-> Select Kernel type
-> Select MIPS FPU Emulator

nxhack, 2017-07-03:
Hi @rench 

LEDE 17.01-head / linux-4.4.74
```
config FPE_NWFPE
        bool "NWFPE math emulation"
        depends on (!AEABI || OABI_COMPAT) && !THUMB2_KERNEL
        ---help---
          Say Y to include the NWFPE floating point emulator in the kernel.
          This is necessary to run most binaries. Linux does not currently
          support floating point hardware so you need to say Y here even if
          your machine has an FPA or floating point co-processor podule.

          You may say N here if you are going to load the Acorn FPEmulator
          early in the bootup.
```
To forcibly enable it...
**I don't know how bad effects will come out.**

* bcm4709 and  bcm53xx
  CONFIG_TARGET_BOARD="bcm53xx"
make kernel_menuconfig
-> Select Kernel Features
---> Unselect Use the ARM EABI to compile the kernel
(back top)
-> Select Floating point emulation
---> Select NWFPE math emulation

* ar9344
  CONFIG_TARGET_BOARD="ar71xx"
make kernel_menuconfig
-> Select Kernel type
---> Select MIPS FPU Emulator

rench, 2017-07-04:
yesterday , i found this way.
i try to enable aeabi and oabi_compat , it won't effect.
i will try disable aeabi.

nxhack, 2017-07-04:
> i try to enable aeabi and oabi_compat , it won't effect.

Hmm...
What happens if you enable "VFP-format floating point maths"?

-> Select Floating point emulation
---> Select NWFPE math emulation
---> Select **VFP-format floating point maths**

rench, 2017-07-04:
Unfortunately, I disabled aeabi, the results of the router can not start.
 if you enable "VFP-format floating point maths",
when node it echo  Illegal instruction too.

i found some hard code in toolchains/gcc/common.mk
```
ifneq ($(CONFIG_SOFT_FLOAT),y)
  ifeq ($(CONFIG_arm),y)
    GCC_CONFIGURE+= \
		--with-float=hard
  endif
endif
```

do this trigger Illegal instruction ？

nxhack, 2017-07-04:
It is not caused by the float instruction generated by gcc. It is a problem that the float instruction generated by the V8 JIT compiler. So hardware FPU or kernel FPU emulation is required.

Unfortunately, it seems better to give up using node on ARM without FPU.

rench, 2017-07-04:
sad.i have 2 routers with bcm4709.

but how can MIPS FPU Emulator work good?
i see node/Makefile use --with-arm-float-abi=soft
but when i run `node --v8-options` in router,  i echo  `softfp`

nxhack, 2017-07-04:
> but how can MIPS FPU Emulator work good?

The reason is simple: The fpu emulator of MIPS work well, the fpu emulator of ARM does not work well.

> i see node/Makefile use --with-arm-float-abi=soft
> but when i run node --v8-options in router, i echo softfp

As the last challenge...

 kernel : AEABI=on, OABI_COMPAT=off, NWFPE=on, VFP=on
 node/Makefile:    $(if $(CONFIG_SOFT_FLOAT), --with-arm-float-abi=softfp, \              

Please tell me the result.

rench, 2017-07-04:
NWFPE depends on AEABI=off || OABI_COMPAT=on
i will try tonight，then tell you result。

nxhack, 2017-07-04:
> NWFPE depends on AEABI=off || OABI_COMPAT=on

sorry.
kernel : AEABI=on, OABI_COMPAT=on, NWFPE=on, VFP=on
NWFPE can enable.

rench, 2017-07-04:
i  will  build
1. AEABI=on, OABI_COMPAT=on, NWFPE=on, VFP=on --with-arm-float-abi=softfp
2. AEABI=on, OABI_COMPAT=on, NWFPE=on, VFP=off --with-arm-float-abi=softfp

how to pass `-mfpu=fpa` to compile node?

nxhack, 2017-07-04:
`TARGET_CFLAGS+=-mfpu=fpa `

but get
```
arm-openwrt-linux-muslgnueabi-g++: error: unrecognized argument in option '-mfpu=fpa'
arm-openwrt-linux-muslgnueabi-g++: note: valid arguments to '-mfpu=' are: crypto-neon-fp-armv8 fp-armv8 fpv4-sp-d16 fpv5-d16 fpv5-sp-d16 neon neon-fp-armv8 neon-fp16 neon-vfpv4 vfp vfp3 vfpv3 vfpv3-d16 vfpv3-d16-fp16 vfpv3-fp16 vfpv3xd vfpv3xd-fp16 vfpv4 vfpv4-d16
```

@rench  Let's close this issue.

rench, 2017-07-04:
Do you have any other communication tools?
There are problems we can discuss together.

nxhack, 2017-07-04:
OK Let's keep this thread because open discussion is desirable.

and AGAIN.
> It is not caused by the float instruction generated by gcc. It is a problem that the float instruction generated by the V8 JIT compiler. So hardware FPU or kernel FPU emulation is required.
> 
> Unfortunately, it seems better to give up using node on ARM without FPU.

"nodejs/node/issues/7343"  It is as you read.

rench, 2017-07-04:
yes i have read that issue before.
i try to search `--with-arm-float-abi` parameter in google, but nothing to be show.

I wonder if this parameter really works？

nxhack, 2017-07-04:
it is node.js configure parameter.

https://github.com/nodejs/node/blob/v6.x/configure#L295

rench, 2017-07-04:
it does not work.
i think i have wrong step.
the different choice build the same byte trx file.
three trx file has the same bytes.

rench, 2017-07-04:
1. the fpu emulator of ARM does not work well.
2. node v8 always use vfp3 and softfp

solve it by:
- try to make ARM's fpu emulator work well
- try to make node v8  soft-float.

below section is v8-options that build node with soft/softfp.
```
root@LEDE:~# ./node --v8-options
target arm v7 vfp3-d16 softfp
ARMv8=0 ARMv7=1 VFP3=1 VFP32DREGS=0 NEON=0 SUDIV=0 MLS=1UNALIGNED_ACCESSES=1 MOVW_MOVT_IMMEDIATE_LOADS=0 COHERENT_CACHE=0 USE_EABI_HARDFLOAT=0
```

nxhack, 2017-07-05:
hmm

> root@LEDE:~# ./node --v8-options
> target arm v7 vfp3-d16 softfp

What if we set node.js configure  "--with-arm-fpu = vfp"?

rench, 2017-07-05:
i have test it.but no help.
i think the makefile config is set to build node but not to v8.
node work well because if i run node -v，print v6.11.0.
if i run other cmd，node will invoke v8，so error happen.

if there some config can pass to build v8？

or make FPE work well？

i have no friend know lede，i have no idear now……

rench, 2017-07-05:
https://bugs.chromium.org/p/v8/issues/detail?id=4704

```
* v8 does not have an official soft-float support. This thing is not tested and we don't know if it works correctly in all scenarios. You might experience some problems which nobody can help you with

* We did some performance comparison between soft-float and hard-float on MIPS using google octane benchsuite. Soft-float is on average 5% faster that hard-float on a board without HW FPU. Is this performance difference enough to justify using soft-float?

* Even if v8 is compiled with soft-float, this setting only applies to V8 C code. Don't forget that v8 is a JIT compiler and most of the code executed is v8 JIT code, not v8 C code. v8 JIT code DOES generate FP instructions if they are needed, regardless of compiler settings.

```
after read that, i think the only way is to make arm NWFPE work well.

rench, 2017-07-05:
Found that the latest kernel does not exist / arm / arm / fastfpe directory, and nwfpe EABI should also be outdated: the algorithm for FPA, EABI support VFP should not be properly supported.

working and searching...

rench, 2017-07-05:
MIPS work well.
https://github.com/v8/v8/blob/51d38345f9158e23fe5b0fceb5c6c97e4254c8dc/src/mips/constants-mips.h#L76

closed 2018-02-14 by nxhack