nxhack/openwrt-node-packages #17: homebridge  undefined symbol error

ReleaseBug, 2017-07-11:
oops,,,,,,,,
i have installed node & node-npm & node-homebridge & node-mdns & libavahi-compat-libdnss.
(selected [M] when build to prevent too big firmware, and lately install them to a usb storage device).

when i run  homebridge undefined symbol error occurre.
What should I do? Thx~

LOGS:
root@OpenWrt:/usr/bin# homebridge 
/usr/lib/node/homebridge/node_modules/mdns/lib/dns_sd.js:35
    throw ex;
    ^

Error: /usr/lib/node/homebridge/node_modules/mdns/build/Release/dns_sd_bindings.node: undefined symbol: _ZN2v816FunctionTemplate3NewEPNS_7IsolateEPFvRKNS_20FunctionCallbackInfoINS_5ValueEEEENS_5LocalIS4_EENSA_INS_9SignatureEEEi
    at Object.Module._extensions..node (module.js:598:18)
    at Module.load (module.js:503:32)
    at tryModuleLoad (module.js:466:12)
    at Function.Module._load (module.js:458:3)
    at Module.require (module.js:513:17)
    at require (internal/module.js:11:18)
    at Object.<anonymous> (/usr/lib/node/homebridge/node_modules/mdns/lib/dns_sd.js:24:20)
    at Module._compile (module.js:569:30)
    at Object.Module._extensions..js (module.js:580:10)
    at Module.load (module.js:503:32)

nxhack, 2017-07-12:
Are you using the same build in the same environment?

Because mdns is packed in homebridge, is it necessary?

nxhack, 2017-07-12:
Have you changed the environment of V8 when 'dns_sd_bindings.node' was built?

ReleaseBug, 2017-07-12:
yes, the same build in the same environment.
built node package names are:
node_v8.1.3-1_ramips.ipk
node-npm_v8.1.3-1_ramips.ipk
node-mdns_2.3.3-7_ramips.ipk
node-homebridge_0.4.20-2_ramips.ipk

AND I did the following steps before built openwrt (trunk ) codes：
1:
Add follow line to feeds.conf or feeds.conf.default
src-git node https://github.com/nxhack/openwrt-node-packages.git
2:
Run 
./scripts/feeds update node
rm ./package/feeds/packages/node
rm ./package/feeds/packages/node-arduino-firmata
rm ./package/feeds/packages/node-cylon
rm ./package/feeds/packages/node-hid
rm ./package/feeds/packages/node-serialport
./scripts/feeds install -a -p node
3:
Run make menuconfig
then  chosen:
Languages->Node.js->[M]node->configuration->Version Selection->[X]8.x
Languages->Node.js->[M]node-homebridge
Languages->Node.js->[M]node-mdns
Languages->Node.js->[M]node-npm

4：install these packages via luci software.
node
node-npm
node-mdns
node-homebridge

5: Run homebridge  AND undefined symbol error occurre.

Did i missed(wrong) some steps?
AND……
How to  " changed the environment of V8 when 'dns_sd_bindings.node' was built"?

nxhack, 2017-07-12:
hmm it is correct step...

One more thing to check. Is there a possibility that "node/host" (under ./staging_dir/target-xxx/host/usr/bin/node etc) is old version?

ReleaseBug, 2017-07-12:
my full path is openwrt/staging_dir/target-mipsel_24kec+dsp_glibc-2.22/host/bin
AND under this path  i run:
$ ./node -v
v4.8.3
./npm -v 
2.15.11
Is old version.

why could be that?
How to deal with this? Thx~

nxhack, 2017-07-12:
> $ ./node -v
> v4.8.3
> ./npm -v
> 2.15.11

Bingo! It is old (not v8.1.3)

Please reinstall node/host.

ReleaseBug, 2017-07-12:
good .
one more thing , Please tell me how to reinstall node/host(v8.1.3) to my building environment.
As a beginner, I am not clearly about what to do.

nxhack, 2017-07-12:
The most certain thing is 
```
make clean
make
```
 It takes long time.

ReleaseBug, 2017-07-12:
get it!
It really takes long time(may be 5 hours) and  I will feedback a bit late.
And thanks again!

ReleaseBug, 2017-07-12:
oh, build error occur.

i run ,
sudo make clean
sudo make V=s TARGET_DEVICES=youku-yk1 FORCE_UNSAFE_CONFIGURE=1

LOGS:
npm info lifecycle ed25519@0.0.4~install: ed25519@0.0.4

> ed25519@0.0.4 install /home/ubuntu/openwrt/openwrt/build_dir/target-mipsel_24kec+dsp_glibc-2.22/node-homebridge-0.4.20/ipkg-install/usr/lib/node_modules/homebridge/node_modules/ed25519
> node-gyp rebuild

gyp info it worked if it ends with ok
gyp info using node-gyp@3.6.2
gyp info using node@8.1.3 | linux | x64
gyp ERR! configure error 
**gyp ERR! stack Error: EACCES: permission denied, mkdir '/home/ubuntu/openwrt/openwrt/build_dir/target-mipsel_24kec+dsp_glibc-2.22/node-homebridge-0.4.20/ipkg-install/usr/lib/node_modules/homebridge/node_modules/ed25519/build'**
gyp ERR! System Linux 4.8.0-58-generic
gyp ERR! command "/home/ubuntu/openwrt/openwrt/staging_dir/target-mipsel_24kec+dsp_glibc-2.22/host/bin/node" "/home/ubuntu/openwrt/openwrt/staging_dir/target-mipsel_24kec+dsp_glibc-2.22/host/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/ubuntu/openwrt/openwrt/build_dir/target-mipsel_24kec+dsp_glibc-2.22/node-homebridge-0.4.20/ipkg-install/usr/lib/node_modules/homebridge/node_modules/ed25519
gyp ERR! node -v v8.1.3
gyp ERR! node-gyp -v v3.6.2
gyp ERR! not ok 
npm info lifecycle ed25519@0.0.4~install: Failed to exec install script
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! ed25519@0.0.4 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the ed25519@0.0.4 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/ubuntu/openwrt/openwrt/staging_dir/target-mipsel_24kec+dsp_glibc-2.22/host/share/node/npm-cache/_logs/2017-07-12T09_42_40_751Z-debug.log

nxhack, 2017-07-13:
OK I will test build it.

> target-mipsel_24kec+dsp_glibc-2.22

By the way, node upstream was updated because of  security issue.
https://nodejs.org/en/blog/vulnerability/july-2017-security-releases/

Please update.

`./scripts/feeds update node`

nxhack, 2017-07-13:
hmm.

> gyp ERR! stack Error: EACCES: permission denied, mkdir '/home/ubuntu/openwrt/openwrt/build_dir/target-mipsel_24kec+dsp_glibc-2.22/node-homebridge-0.4.20/ipkg-install/usr/lib/node_modules/homebridge/node_modules/ed25519/build'

Please check it. Please correct directory permission.

nxhack, 2017-07-13:
> sudo make clean
> sudo make V=s TARGET_DEVICES=youku-yk1 FORCE_UNSAFE_CONFIGURE=1

Administrator rights are not required to build openwrt. Please correct and build with correct user privileges.

ReleaseBug, 2017-07-13:
OK, i would like to make a new building environment.
start over again with cloning openwrt trunk codes.
And then build with correct use(no sudo).

Thx~

ReleaseBug, 2017-07-13:
Great!! Everything's OK!
Now, Homebridge is running  in my router .

Thank you a lot !

closed 2017-07-13 by ReleaseBug

nxhack, 2017-07-13:
Wow! Excellent!