nxhack/openwrt-node-packages #18: node-homebridge Error

songjw0820, 2017-07-17:
npm info ok 
touch /home/sjw/lede_sdk/source/build_dir/target-mipsel_24kc_musl-1.1.16/node-homebridge-0.4.22/.built
mkdir -p /home/sjw/lede_sdk/source/bin/targets/ramips/mt7688/packages /home/sjw/lede_sdk/source/build_dir/target-mipsel_24kc_musl-1.1.16/node-homebridge-0.4.22/ipkg-mipsel_24kc/node-homebridge/CONTROL /home/sjw/lede_sdk/source/staging_dir/target-mipsel_24kc_musl-1.1.16/pkginfo
install -d -m0755 /home/sjw/lede_sdk/source/build_dir/target-mipsel_24kc_musl-1.1.16/node-homebridge-0.4.22/ipkg-mipsel_24kc/node-homebridge/usr/lib/node
cp -fpR /home/sjw/lede_sdk/source/build_dir/target-mipsel_24kc_musl-1.1.16/node-homebridge-0.4.22/ipkg-install/usr/lib/node_modules/* /home/sjw/lede_sdk/source/build_dir/target-mipsel_24kc_musl-1.1.16/node-homebridge-0.4.22/ipkg-mipsel_24kc/node-homebridge/usr/lib/node/
install -d -m0755 /home/sjw/lede_sdk/source/build_dir/target-mipsel_24kc_musl-1.1.16/node-homebridge-0.4.22/ipkg-mipsel_24kc/node-homebridge/usr/bin
ln -sf ../lib/node/homebridge/bin/homebridge /home/sjw/lede_sdk/source/build_dir/target-mipsel_24kc_musl-1.1.16/node-homebridge-0.4.22/ipkg-mipsel_24kc/node-homebridge/usr/bin/homebridge
find /home/sjw/lede_sdk/source/build_dir/target-mipsel_24kc_musl-1.1.16/node-homebridge-0.4.22/ipkg-mipsel_24kc/node-homebridge -name 'CVS' -o -name '.svn' -o -name '.#*' -o -name '*~'| xargs -r rm -rf
Package node-homebridge is missing dependencies for the following libraries:
libdns_sd.so
Makefile:69: recipe for target '/home/sjw/lede_sdk/source/bin/packages/mipsel_24kc/node/node-homebridge_0.4.22-1_mipsel_24kc.ipk' failed
make[2]: *** [/home/sjw/lede_sdk/source/bin/packages/mipsel_24kc/node/node-homebridge_0.4.22-1_mipsel_24kc.ipk] Error 1
make[2]: Leaving directory '/home/sjw/lede_sdk/source/feeds/node/node-homebridge'
package/Makefile:105: recipe for target 'package/feeds/node/node-homebridge/compile' failed
make[1]: *** [package/feeds/node/node-homebridge/compile] Error 2
make[1]: Leaving directory '/home/sjw/lede_sdk/source'
/home/sjw/lede_sdk/source/include/toplevel.mk:197: recipe for target 'package/feeds/node/node-homebridge/compile' failed
make: *** [package/feeds/node/node-homebridge/compile] Error 2

nxhack, 2017-07-17:
Hi @songjw0820 
> Package node-homebridge is missing dependencies for the following libraries:
> libdns_sd.so

Please check your LEDE config.  make sure that "libavahi-compat-libdnssd" package installed.

nxhack, 2017-07-28:
No news is good news. ;)

CLOSE!

closed 2017-07-28 by nxhack