nxhack/openwrt-node-packages #32: please add node-ejdb native lib

cooboyno111, 2017-07-28:
https://www.npmjs.com/package/ejdb

-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Looking for include file pthread.h
-- Looking for include file pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE
CMake Warning at src/CMakeLists.txt:64 (message):
Library 'librt' not FOUND

CMake Error at src/CMakeLists.txt:69 (message):
Library 'libm' not FOUND

I use openwrt-trunk .
nodejs version is 4.4.5 in openwrt feeds
my makefile
Copyright (C) 2014 Arduino LLC
This is free software, licensed under the GNU General Public License v2.
See /LICENSE for more information.

include $(TOPDIR)/rules.mk

PKG_NPM_NAME:=ejdb
PKG_NAME:=node-$(PKG_NPM_NAME)
PKG_VERSION:=1.2.12-42
PKG_RELEASE:=3

PKG_SOURCE:=$(PKG_NPM_NAME)-$(PKG_VERSION).tgz
PKG_SOURCE_URL:=http://registry.npmjs.org/$(PKG_NPM_NAME)/-/
PKG_MD5SUM:=9b665a1b9105c4cda36c825b3ddecae1

PKG_BUILD_DEPENDS:=node/host
PKG_NODE_VERSION:=4.4.5
PKG_MAINTAINER:=John Crispin blogic@openwrt.org
PKG_LICENSE:=Custom
PKG_LICENSE_FILES:=LICENSE

include $(INCLUDE_DIR)/package.mk

define Package/node-ejdb
DEPENDS:=+node +libpthread +librt +zlib +libm
SUBMENU:=Node.js
SECTION:=lang
CATEGORY:=Languages
TITLE:=Node.js ejdb
URL:=https://www.npmjs.org/package/ejdb
endef

define Package/node-ejdb/description
Node.js ejdb
endef

define Build/Prepare
/bin/tar xzf $(DL_DIR)/$(PKG_SOURCE) -C $(PKG_BUILD_DIR) --strip-components 1
$(Build/Patch)
endef

CPU:=$(subst x86_64,x64,$(subst i386,ia32,$(ARCH)))

EXTRA_LDFLAGS="-L$(TOOLCHAIN_DIR)/lib/ -Wl,-rpath-link $(TOOLCHAIN_DIR)/lib/" \

define Build/Compile
$(MAKE_FLAGS)
npm_config_arch=$(CONFIG_ARCH)
npm_config_nodedir=$(BUILD_DIR)/node-v$(PKG_NODE_VERSION)/
npm_config_cache=$(BUILD_DIR)/node-v$(PKG_NODE_VERSION)/npm-cache
PREFIX="$(PKG_INSTALL_DIR)/usr/"
npm install --build-from-source --target_arch=$(CPU) -g $(PKG_BUILD_DIR)
endef

define Package/node-ejdb/install
mkdir -p $(1)/usr/lib/node/
$(CP) $(PKG_INSTALL_DIR)/usr/lib/node_modules/* $(1)/usr/lib/node/
endef

$(eval $(call BuildPackage,node-ejdb))

nxhack, 2017-07-28:
Hi @cooboyno111 
Please check. I have not run tests. Please tell me result.

for CC:   https://github.com/nxhack/openwrt-node-packages/tree/for-15.05/node-ejdb
for LEDE: https://github.com/nxhack/openwrt-node-packages/tree/master/node-ejdb

cooboyno111, 2017-07-28:
@nxhack
npm ERR! code EINTEGRITY
npm ERR! sha1-DCi8ZpqFFiFwm/eghQMDS+44ErY= integrity checksum failed when using sha1: wanted sha1-DCi8ZpqFFiFwm/eghQMDS+44ErY= but got sha1-/5Lez6nmEQwA0qEMsl9CUbaReJg=. (1615252 bytes)

npm ERR! A complete log of this run can be found in:
npm ERR!     /media/lup/22e7aa1c-9dd5-4d35-ad44-7bfed156fb0d/mtklinkkit/openwrt/staging_dir/host/share/node/npm-cache/_logs/2017-07-28T08_57_00_661Z-debug.log
Makefile:63: recipe for target '/media/lup/22e7aa1c-9dd5-4d35-ad44-7bfed156fb0d/mtklinkkit/openwrt/build_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/node-ejdb-1.2.12-42/.built' failed
for openwrt CC

closed 2017-07-28 by cooboyno111

nxhack, 2017-07-28:
@cooboyno111 
Does it work well?

cooboyno111, 2017-07-31:
@nxhack MT7628/MT7620 work well
root@k-wifi:/tmp# cat /proc/cpuinfo
system type		: MediaTek MT7628AN ver:1 eco:2
machine			: KunTeng ZC9525A
processor		: 0
cpu model		: MIPS 24KEc V5.5
BogoMIPS		: 380.92
wait instruction	: yes
microsecond timers	: yes
tlb_entries		: 32
extra interrupt vector	: yes
hardware watchpoint	: yes, count: 4, address/irw mask: [0x0ffc, 0x0ffc, 0x0ffb, 0x0ffb]
isa			: mips1 mips2 mips32r1 mips32r2
ASEs implemented	: mips16 dsp
shadow register sets	: 1
kscratch registers	: 0
package			: 0
core			: 0
VCED exceptions		: not available
VCEI exceptions		: not available

root@k-wifi:/tmp# node t.js 
Grenny OID: 597ee11a0cfef12400000000
Bounty OID: 597ee11a0cfef12400000001
Found 1 parrots
Grenny likes toys!
use https://www.npmjs.com/package/ejdb test。
Ejdblib takes up 1.5M of space