nxhack/openwrt-node-packages #42: will you have packages install-able for node 8 like lede with build bot or something?

p3x-robot, 2017-08-04:


nxhack, 2017-08-06:
Hi @p3x-robot 
Please give me detail.

p3x-robot, 2017-08-07:
Hi!
How are you?
i just want to build a node 8 package but i do not idea how to start up with it.
Don't you have a script or something how you would do it?
I would like with LEDE stable.

Like:
* git clone github/lede
* delete packages/node*
* git clone nxhack/openwrt-node-packages
* copy to node feeds
* select node packages
* configure
* make

Is it something like this?

nxhack, 2017-08-07:
@p3x-robot 

```
mkdir Build_Path_Some_Where
cd Build_Path_Some_Where
git clone https://git.lede-project.org/source.git
cd source
cp feeds.conf.default feeds.conf
echo 'src-git node https://github.com/nxhack/openwrt-node-packages.git' >> feeds.conf
./scripts/feeds update
./scripts/feeds install -a
rm ./package/feeds/packages/node
rm ./package/feeds/packages/node-arduino-firmata
rm ./package/feeds/packages/node-cylon
rm ./package/feeds/packages/node-hid
rm ./package/feeds/packages/node-serialport
./scripts/feeds install -a -p node
```
```
make menuconfig
```
Select Your target.
Select node.js version.
Select node packages
```
make
```

Note: 
If your target machine is MIPS and there is no hardware FPU, be sure to enable "MIPS_FPU_EMULATOR",, before make.
```
make kernel_menuconfig
```

p3x-robot, 2017-08-07:
ahh, you are god, i am testing! thanks dude!

p3x-robot, 2017-08-07:
hey dude, it works, but i get a new menu, what do i choose after ```make kernel_menuconfig```

p3x-robot, 2017-08-07:
ok, i got and ipkg, but it is a node 6, how can i select 8?

p3x-robot, 2017-08-07:
is it in the ```make kernel_menuconfig``` to choose nodejs version 8?

p3x-robot, 2017-08-07:
never mind, got it testing, then closing

p3x-robot, 2017-08-07:
if i just need for the done LEDE latest stable, I do not need ```make kernel_menuconfig``` right?

p3x-robot, 2017-08-07:
do you know how to use npm? 
[image]

p3x-robot, 2017-08-07:
never minde, i got it :)

p3x-robot, 2017-08-07:
I did this:
```bash
./run.sh
make menuconfig
# Target System
#   MediaTek Ralink MIPS 
# Target Profile
#   MT7621
# The Languages => Node.js (Make sure it is a module M)
# Space
# Down to
# => Configuration => Version Selection => 8.x
# Select 
# Exit
# node-npm (Make sure it is a module M)
# Save
# Exit all

# Note:
# If your target machine is MIPS and there is no hardware FPU, be 
# => MIPS_FPU_EMULATOR
# Advanced configuration options
# sure to enable "MIPS_FPU_EMULATOR", before make.
make kernel_menuconfig
# Kernel type => MIPS FPU EMULATOR
# Save
# Exit
# Exit

make -j8
```

Result:
```text
root@upc:/opt/router-scripts-lede/packages/LEDE-Stable-D-Link-DIR-860L-B1# node
Illegal instruction
```

p3x-robot, 2017-08-07:
trying revert, first i do the kernel menu...

p3x-robot, 2017-08-07:
ok, i build a new firwmare with emulator.... I let you know..

p3x-robot, 2017-08-07:
too much hassle, with my linksys 1900acs v2 it works and i dont even need the other... so it is awesome, you are the god!!!
ciao! happy day!

closed 2017-08-07 by p3x-robot

nxhack, 2017-08-08:
MEMO:
linksys 1900acs v2 : Marvell Armada 385 88F6820
Dual-core Cortex A9 ARMv7 CPU with NEON SIMD and Floating Point Unit (FPU)

👍

nxhack, 2017-08-08:
MEMO:
D-Link DIR-860L B1 : MediaTek MT7621AT
ramips MIPS32 1004K
Definitely enable MIPS_FPU_EMULATOR (disabled by default kernel config)

p3x-robot, 2017-08-08:
dude i did it! for mips! nodejs works!
now, i make my own firmware lede 17.01.2 for linksys as well.
This is awesome!

https://cdn.corifeus.com/lede/17.01.2/packages/mipsel_24kc/node/