nxhack/openwrt-node-packages #43: Build failed on debian 9

josser, 2017-08-09:
Hi! I'm trying to build homebridge for my router but getting error during build:

```
configure: error: in `/home/josser/Загрузки/lede-sdk-ar71xx-generic_gcc-5.4.0_musl.Linux-x86_64/build_dir/target-mips_24kc_musl/avahi/dbus/avahi-0.6.32':
configure: error: The pkg-config script could not be found or is too old.  Make sure it
is in your PATH or set the PKG_CONFIG environment variable to the full
path to pkg-config.

Alternatively, you may set the environment variables DBUS_CFLAGS
and DBUS_LIBS to avoid the need to call pkg-config.
See the pkg-config man page for more details.

To get pkg-config, see <http://pkg-config.freedesktop.org/>.
See `config.log' for more details
Makefile:396: recipe for target '/home/josser/Загрузки/lede-sdk-ar71xx-generic_gcc-5.4.0_musl.Linux-x86_64/build_dir/target-mips_24kc_musl/avahi/dbus/avahi-0.6.32/.configured_68b329da9893e34099c7d8ad5cb9c940' failed
make[3]: *** [/home/josser/Загрузки/lede-sdk-ar71xx-generic_gcc-5.4.0_musl.Linux-x86_64/build_dir/target-mips_24kc_musl/avahi/dbus/avahi-0.6.32/.configured_68b329da9893e34099c7d8ad5cb9c940] Error 1
make[3]: Leaving directory '/home/josser/Загрузки/lede-sdk-ar71xx-generic_gcc-5.4.0_musl.Linux-x86_64/feeds/packages/libs/avahi'
package/Makefile:109: recipe for target 'package/feeds/packages/avahi/compile' failed
make[2]: *** [package/feeds/packages/avahi/compile] Error 2
make[2]: Leaving directory '/home/josser/Загрузки/lede-sdk-ar71xx-generic_gcc-5.4.0_musl.Linux-x86_64'
package/Makefile:105: recipe for target '/home/josser/Загрузки/lede-sdk-ar71xx-generic_gcc-5.4.0_musl.Linux-x86_64/staging_dir/target-mips_24kc_musl/stamp/.package_compile' failed
make[1]: *** [/home/josser/Загрузки/lede-sdk-ar71xx-generic_gcc-5.4.0_musl.Linux-x86_64/staging_dir/target-mips_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/josser/Загрузки/lede-sdk-ar71xx-generic_gcc-5.4.0_musl.Linux-x86_64'
/home/josser/Загрузки/lede-sdk-ar71xx-generic_gcc-5.4.0_musl.Linux-x86_64/include/toplevel.mk:200: ошибка выполнения рецепта для цели «world»
make: *** [world] Error 2
```

Any advices? 
Thank you!

nxhack, 2017-08-09:
Hi @josser 

Please update debian's packages
```
sudo apt-get update
sudo apt-get install pkg-config
```
If you can also upgrade other packages,
```
sudo apt-get upgrade
```

nxhack, 2017-08-10:
Hi @josser 
# Cancel and resend

Please test, Change from Cyrillic character to ASCII character of your build path (directory name) .

josser, 2017-08-14:
Cool! moving to non-cyrillic folder helps to solve this but now another error appearing: 
```
/home/josser/lede-sdk-ar71xx-generic_gcc-5.4.0_musl.Linux-x86_64/staging_dir/toolchain-mips_24kc_gcc-5.4.0_musl/bin/../lib/gcc/mips-openwrt-linux-musl/5.4.0/../../../../mips-openwrt-linux-musl/bin/ld: cannot find -ldns_sd
collect2: error: ld returned 1 exit status
dns_sd_bindings.target.mk:184: recipe for target 'Release/obj.target/dns_sd_bindings.node' failed
make[4]: *** [Release/obj.target/dns_sd_bindings.node] Error 1
make[4]: Leaving directory '/home/josser/lede-sdk-ar71xx-generic_gcc-5.4.0_musl.Linux-x86_64/build_dir/target-mips_24kc_musl/node-hap-nodejs-0.4.27/ipkg-install/usr/lib/node_modules/hap-nodejs/node_modules/mdns/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/home/josser/lede-sdk-ar71xx-generic_gcc-5.4.0_musl.Linux-x86_64/staging_dir/hostpkg/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:276:23)
gyp ERR! stack     at emitTwo (events.js:106:13)
gyp ERR! stack     at ChildProcess.emit (events.js:191:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:215:12)
gyp ERR! System Linux 4.9.0-3-amd64
gyp ERR! command "/home/josser/lede-sdk-ar71xx-generic_gcc-5.4.0_musl.Linux-x86_64/staging_dir/hostpkg/bin/node" "/home/josser/lede-sdk-ar71xx-generic_gcc-5.4.0_musl.Linux-x86_64/staging_dir/hostpkg/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/josser/lede-sdk-ar71xx-generic_gcc-5.4.0_musl.Linux-x86_64/build_dir/target-mips_24kc_musl/node-hap-nodejs-0.4.27/ipkg-install/usr/lib/node_modules/hap-nodejs/node_modules/mdns
gyp ERR! node -v v6.11.2
gyp ERR! node-gyp -v v3.4.0
gyp ERR! not ok 
npm info lifecycle mdns@2.3.3~install: Failed to exec install script
/home/josser/lede-sdk-ar71xx-generic_gcc-5.4.0_musl.Linux-x86_64/build_dir/target-mips_24kc_musl/node-hap-nodejs-0.4.27/ipkg-install/usr/lib
`-- (empty)
```
Thank you!

josser, 2017-08-14:
Will try to select 'y' =  '*' for libavahi-compat-libdnssd as it said in docs )

closed 2017-08-18 by josser