nxhack/openwrt-node-packages #46: node-pre-gyp: Permission denied

opjlmi, 2017-08-14:
Hi,
I create a DockerFile to build Linkit7688-nodejs-v8.* image.
I'm not sure which show error `permission denied` when running node-pre-gyp.
This is my Docker File -> [(here)](https://gist.github.com/opjlmi/12ed612f9a27d0e2978fe0dbb957bd79#file-dockerfile)

```
> serialport@4.0.7 install /root/openwrt/build_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/node-firmata-0.17.0/ipkg-install/usr/lib/node_modules/firmata/node_modules/serialport
> node-pre-gyp install --fallback-to-build

sh: 1: node-pre-gyp: Permission denied
npm info lifecycle serialport@4.0.7~install: Failed to exec install script
npm ERR! file sh
npm ERR! code ELIFECYCLE
npm ERR! errno ENOENT
npm ERR! syscall spawn
npm ERR! serialport@4.0.7 install: `node-pre-gyp install --fallback-to-build`
npm ERR! spawn ENOENT
npm ERR!
npm ERR! Failed at the serialport@4.0.7 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/openwrt/staging_dir/host/share/node/npm-cache/_logs/2017-08-14T04_51_08_182Z-debug.log
make[3]: *** [/root/openwrt/build_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/node-firmata-0.17.0/.built] Error 1
make[3]: Leaving directory `/root/openwrt/feeds/node/node-firmata'
make[2]: *** [package/feeds/node/node-firmata/compile] Error 2
npm ERR! file sh
npm ERR! code ELIFECYCLE
npm ERR! errno ENOENT
npm ERR! syscall spawn
npm ERR! serialport@4.0.7 install: `node-pre-gyp install --fallback-to-build`
npm ERR! spawn ENOENT
npm ERR!
npm ERR! Failed at the serialport@4.0.7 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/openwrt/staging_dir/host/share/node/npm-cache/_logs/2017-08-14T04_51_08_386Z-debug.log
make[3]: *** [/root/openwrt/build_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/node-cylon-firmata-0.24.0/.built] Error 1
make[3]: Leaving directory `/root/openwrt/feeds/node/node-cylon-firmata'
make[2]: *** [package/feeds/node/node-cylon-firmata/compile] Error 2
```

nxhack, 2017-08-14:
Hi @opjlmi 
Please build with correct user (not root) privileges.

opjlmi, 2017-08-15:
I builded again with normal user.

But it still builds error on this:
```
make[3]: Leaving directory `/home/ubuntu/openwrt/feeds/node/node-serialport'
make[2]: Leaving directory `/home/ubuntu/openwrt'
make[1]: *** [/home/ubuntu/openwrt/staging_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/stamp/.package_compile] Error 2
make[1]: Leaving directory `/home/ubuntu/openwrt'
Build failed - please re-run with -j1 to see the real error message
make: *** [world] Error 1
```

nxhack, 2017-08-15:
Try
```
make V=99 package/feeds/node/node-serialport/clean
make V=99 package/feeds/node/node-serialport/compile
```
What kind of error did you get?

opjlmi, 2017-08-15:
```
/home/ubuntu/openwrt/staging_dir/host/include/node/v8.h:21:19: fatal error: utility: No such file or directory
 #include <utility>
                   ^
compilation terminated.
make[3]: *** [Release/obj.target/serialport/src/serialport.o] Error 1
make[3]: Leaving directory `/home/ubuntu/openwrt/build_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/node-serialport-4.0.7/ipkg-install/usr/lib/node_modules/serialport/build'
gyp ERR! build error
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/home/ubuntu/openwrt/staging_dir/host/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:276:23)
gyp ERR! stack     at emitTwo (events.js:106:13)
gyp ERR! stack     at ChildProcess.emit (events.js:191:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:215:12)
gyp ERR! System Linux 4.10.0-27-generic
gyp ERR! command "/home/ubuntu/openwrt/staging_dir/host/bin/node" "/home/ubuntu/openwrt/staging_dir/host/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "build" "--fallback-to-build" "--module=/home/ubuntu/openwrt/build_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/node-serialport-4.0.7/ipkg-install/usr/lib/node_modules/serialport/build/Release/serialport.node" "--module_name=serialport" "--module_path=/home/ubuntu/openwrt/build_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/node-serialport-4.0.7/ipkg-install/usr/lib/node_modules/serialport/build/Release"
gyp ERR! cwd /home/ubuntu/openwrt/build_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/node-serialport-4.0.7/ipkg-install/usr/lib/node_modules/serialport
gyp ERR! node -v v6.11.2
gyp ERR! node-gyp -v v3.4.0
gyp ERR! not ok
node-pre-gyp ERR! build error
node-pre-gyp ERR! stack Error: Failed to execute '/home/ubuntu/openwrt/staging_dir/host/bin/node /home/ubuntu/openwrt/staging_dir/host/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/home/ubuntu/openwrt/build_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/node-serialport-4.0.7/ipkg-install/usr/lib/node_modules/serialport/build/Release/serialport.node --module_name=serialport --module_path=/home/ubuntu/openwrt/build_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/node-serialport-4.0.7/ipkg-install/usr/lib/node_modules/serialport/build/Release' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (/home/ubuntu/openwrt/build_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/node-serialport-4.0.7/ipkg-install/usr/lib/node_modules/serialport/node_modules/node-pre-gyp/lib/util/compile.js:83:29)
node-pre-gyp ERR! stack     at emitTwo (events.js:106:13)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:191:7)
node-pre-gyp ERR! stack     at maybeClose (internal/child_process.js:891:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:226:5)
node-pre-gyp ERR! System Linux 4.10.0-27-generic
node-pre-gyp ERR! command "/home/ubuntu/openwrt/staging_dir/host/bin/node" "/home/ubuntu/openwrt/build_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/node-serialport-4.0.7/ipkg-install/usr/lib/node_modules/serialport/node_modules/.bin/node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd /home/ubuntu/openwrt/build_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/node-serialport-4.0.7/ipkg-install/usr/lib/node_modules/serialport
node-pre-gyp ERR! node -v v6.11.2
node-pre-gyp ERR! node-pre-gyp -v v0.6.32
node-pre-gyp ERR! not ok
Failed to execute '/home/ubuntu/openwrt/staging_dir/host/bin/node /home/ubuntu/openwrt/staging_dir/host/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/home/ubuntu/openwrt/build_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/node-serialport-4.0.7/ipkg-install/usr/lib/node_modules/serialport/build/Release/serialport.node --module_name=serialport --module_path=/home/ubuntu/openwrt/build_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/node-serialport-4.0.7/ipkg-install/usr/lib/node_modules/serialport/build/Release' (1)
npm info lifecycle serialport@4.0.7~install: Failed to exec install script
npm ERR! Linux 4.10.0-27-generic
npm ERR! argv "/home/ubuntu/openwrt/staging_dir/host/bin/node" "/home/ubuntu/openwrt/staging_dir/host/bin/npm" "install" "--build-from-source" "--target_arch=mipsel" "-g" "/home/ubuntu/openwrt/dl/serialport-4.0.7.tgz"
npm ERR! node v6.11.2
npm ERR! npm  v3.10.10
npm ERR! code ELIFECYCLE

npm ERR! serialport@4.0.7 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the serialport@4.0.7 install script 'node-pre-gyp install --fallback-to-build'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the serialport package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-pre-gyp install --fallback-to-build
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs serialport
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls serialport
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/ubuntu/openwrt/feeds/node/node-serialport/npm-debug.log
make[2]: *** [/home/ubuntu/openwrt/build_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/node-serialport-4.0.7/.built] Error 1
make[2]: Leaving directory `/home/ubuntu/openwrt/feeds/node/node-serialport'
make[1]: *** [package/feeds/node/node-serialport/compile] Error 2
make[1]: Leaving directory `/home/ubuntu/openwrt'
make: *** [package/feeds/node/node-serialport/compile] Error 2
```

nxhack, 2017-08-15:
I also test the environment by making Linkit 7688 environment. It will take some time.

opjlmi, 2017-08-15:
I guess maybe it's the problem that I used ubuntu:14.04
I'll change to ubuntu:16.04 and build again.

nxhack, 2017-08-16:
In my environment node-serialport was able to build well.
Ubuntu 16.04.3 LTS / non Docker

Please check whether this file exists or it have correct access right.
```
staging_dir/toolchain-mipsel_24kec+dsp_gcc-4.8-linaro_uClibc-0.9.33.2/mipsel-openwrt-linux-uclibc/include/c++/4.8.3/utility
```

opjlmi, 2017-08-16:
Yes, the utility problem has been solved on ubuntu:16.04.
But there is a new problem:

```
make[3]: Leaving directory '/home/ubuntu/openwrt/package/network/utils/curl'
make[2]: Leaving directory '/home/ubuntu/openwrt'
package/Makefile:188: recipe for target '/home/ubuntu/openwrt/staging_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/stamp/.package_compile' failed
make[1]: *** [/home/ubuntu/openwrt/staging_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/ubuntu/openwrt'
Build failed - please re-run with -j1 to see the real error message
/home/ubuntu/openwrt/include/toplevel.mk:181: recipe for target 'world' failed
make: *** [world] Error 1
```

nxhack, 2017-08-16:
If an unknown error occurs in the build of the certain package, it may be resolved by trying to clean and compile it.
```
make V=99 package/network/utils/curl/clean
make V=99 package/network/utils/curl/compile
```

opjlmi, 2017-08-17:
Thanks a lot!
I'll try.

opjlmi, 2017-08-17:
Seems not the feeds problem.
I'm not sure why I don't have the .package_compile file...... Orz

the error msg is same as my last comment.
```
package/Makefile:188: recipe for target '/home/ubuntu/openwrt/staging_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/stamp/.package_compile' failed
make[1]: *** [/home/ubuntu/openwrt/staging_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/stamp/.package_compile] Error 2
```

opjlmi, 2017-08-19:
Hi, sorry for bothering  you again.....
I tried to debug many times,
but it still show the same error like this.

```
make[3]: Leaving directory '/home/ubuntu/openwrt/package/network/utils/curl'
make[2]: Leaving directory '/home/ubuntu/openwrt'
package/Makefile:188: recipe for target '/home/ubuntu/openwrt/staging_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/stamp/.package_compile' failed
make[1]: *** [/home/ubuntu/openwrt/staging_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/ubuntu/openwrt'
Build failed - please re-run with -j1 to see the real error message
/home/ubuntu/openwrt/include/toplevel.mk:181: recipe for target 'world' failed
make: *** [world] Error 1
```

and I tried to 
```
make V=99 package/network/utils/curl/clean
make V=99 package/network/utils/curl/compile
```
but there is no any error msg on the screen.

But if I `make` again.
it will show the different error like

```
Cloning into 'doxyport'...
remote: Counting objects: 55, done.
remote: Compressing objects: 100% (8/8), done.
remote: Total 55 (delta 0), reused 6 (delta 0), pack-reused 47
Unpacking objects: 100% (55/55), done.
Checking connectivity... done.
Submodule path 'doxyport': checked out 'db3e1a6eb8677d3166d90d82c3068689803ce547'
warning: unable to rmdir doxyport: Directory not empty
M	doxygen2jsdoc
Note: checking out 'd8024bb516bf2d895eaff6f92115835d7ef704a2'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by performing another checkout.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -b with the checkout command again. Example:

  git checkout -b <new-branch-name>

HEAD is now at d8024bb... mraa: Update to v0.9.6
fatal: reference is not a tree: 9cc90b7976252b2d14b7956230c5870097e1f008
Unable to checkout '9cc90b7976252b2d14b7956230c5870097e1f008' in submodule path 'doxygen2jsdoc'
Makefile:51: recipe for target '/home/ubuntu/openwrt/dl/libmraa-0.9.6-d8024bb516bf2d895eaff6f92115835d7ef704a2.tar.gz' failed
make[3]: *** [/home/ubuntu/openwrt/dl/libmraa-0.9.6-d8024bb516bf2d895eaff6f92115835d7ef704a2.tar.gz] Error 1
make[3]: Leaving directory '/home/ubuntu/openwrt/feeds/packages/libs/libmraa'
package/Makefile:191: recipe for target 'package/feeds/packages/libmraa/compile' failed
make[2]: *** [package/feeds/packages/libmraa/compile] Error 2
```

and sure, if I run clean/compile again, it still show the same error about the git branch problem.

nxhack, 2017-08-20:
There was a recent change in the repository of Intel mraa. Therefore, it seems that bug of git submodue handling with OpenWrt 15.05's Makefile became obvious.

Tyr
```
wget https://gist.githubusercontent.com/nxhack/78fa1df0a1224a168191dd1ab5b3336e/raw/94b6ded945c61809103e529ddbc41cb4cb757792/fix-git-submodule.patch
patch -p1 < fix-git-submodule.patch
make
```

opjlmi, 2017-08-21:
omg! It works!
thanks for helping me.
I'll share the Dockerfile and write a new post later.

closed 2017-09-03 by nxhack