nxhack/openwrt-node-packages #55: small question

p3x-robot, 2017-08-18:
Ciao!

How are you?

How come ```node-yarn``` uses ```/usr/lib/node/yarn/bin/yarn``` instead of ```/usr/lib/node_modules/yarn```?

Is it just to make it work? Is it easy to change it? I do it if you can help me a little where to do.
Looks like the packages are not in the proper paths.  You must have tons of work anyway or just relaxing in August...

By the way, isn't it simpler just to use ```npm install yarn```??? 

I don't understand why we would want ```node-yarn``` iif it is in ```NPM```...

nxhack, 2017-08-20:
The Makefile which is not the 'native module' of this repository is only for convenience of using packages. It may be better to use 'npm install' except for the 'native module'.

nxhack, 2017-08-20:
> How come node-yarn uses /usr/lib/node/yarn/bin/yarn instead of /usr/lib/node_modules/yarn?

This is based on how to use OpenWrt's node module package. It is to distinguish whether it is derived from package or npm.

closed 2017-08-21 by p3x-robot