nxhack/openwrt-node-packages #56: another small question about scandir

p3x-robot, 2017-08-19:
Ciao!

How are you?

Is this a patch for Node or something?

```
glob error { Error: ENOSYS: function not implemented, scandir '/usr/lib/node_modules/npm/node_modules/cmd-shim'
  errno: -38,
  code: 'ENOSYS',
  syscall: 'scandir',
  path: '/usr/lib/node_modules/npm/node_modules/cmd-shim' }
```

nxhack, 2017-08-20:
Please tell me the details.

p3x-robot, 2017-08-21:
Ciao NXHACK!

How are you?

That's all it is that the npm cmd-shim syscall scandir says:
```
{ Error: ENOSYS: function not implemented, scandir '/usr/lib/node_modules/npm/node_modules/cmd-shim'
  errno: -38,
  code: 'ENOSYS',
  syscall: 'scandir',
  path: '/usr/lib/node_modules/npm/node_modules/cmd-shim' }
```

It is werid becasue using again it is working, so might be not worth taking care about.

But it's weird, it is enough to install ```node``` and ```node-npm```, right?
(Except if I need dbus or other gyp based packages), but I was just using ```ncu``` , ```npm-check-updates```....

This error is weird, becasue it is working now, just threw once, since I use ```NodeJs``` non stop all day for everything intead ```bash```... I guess I close it, if it happens again I ask you. 
Thanks,
Patrik

p3x-robot, 2017-08-21:
i think it is the D-Link D860l B1 is throwsing errors sometimes, 128MB is not enough to use for even like ```npm-check-updates```, I am getting a ```WRT3200ACM```, so with that and my ```WRT1900ACSv2```, no problem at all.

closed 2017-08-21 by p3x-robot