nxhack/openwrt-node-packages #65: please add node-ffi native lib

cooboyno111, 2017-08-28:
https://github.com/node-ffi/node-ffi

nxhack, 2017-08-29:
Hi @cooboyno111 

I tried compiling with a very dirty way to succeed. Please test whether it will run.

https://github.com/nxhack/openwrt-node-packages/tree/dev/node-ffi

cooboyno111, 2017-08-30:
mt7620 LEDE
root@LEDE:/tmp# node t.js 
/usr/lib/node/ffi/node_modules/bindings/bindings.js:83
        throw e
        ^

Error: Error relocating /usr/lib/node/ffi/build/Release/ffi_bindings.node: ffi_type_uint64: symbol not found
    at Error (native)
    at Object.Module._extensions..node (module.js:597:18)
    at Module.load (module.js:487:32)
    at tryModuleLoad (module.js:446:12)
    at Function.Module._load (module.js:438:3)
    at Module.require (module.js:497:17)
    at require (internal/module.js:20:19)
    at bindings (/usr/lib/node/ffi/node_modules/bindings/bindings.js:76:44)
    at Object.<anonymous> (/usr/lib/node/ffi/lib/bindings.js:2:37)
    at Module._compile (module.js:570:32)

t.js 
var ffi = require('ffi');
 
var current = ffi.Library(null, {
  'atoi': [ 'int', [ 'string' ] ]
});
var v2=current.atoi('1234'); // 1234 
console.log(v2);

cooboyno111, 2017-08-30:
openwrt cc Cannot compile

nxhack, 2017-08-30:
@cooboyno111 

I changed it to explicitly use the shared library of libffi. Please check.

Thanks.

nxhack, 2017-08-30:
> openwrt cc Cannot compile

CC version is not prepared yet.
I will make it after the prospect of hack.

cooboyno111, 2017-08-30:
root@LEDE:/tmp# node t.js 
2
1234
root@LEDE:/tmp# cat t.js 
var ffi = require('ffi');
 
var libm = ffi.Library('libm', {
  'ceil': [ 'double', [ 'double' ] ]
});
var v1=libm.ceil(1.5); // 2 
console.log(v1);
// You can also access just functions in the current process by passing a null 
var current = ffi.Library(null, {
  'atoi': [ 'int', [ 'string' ] ]
});
var v2=current.atoi('1234'); // 1234 
console.log(v2);

work well。

nxhack, 2017-08-31:
@cooboyno111 

I made a CC version. CC's libffi version is old and vulnerable. [CVE-2017-1000376](https://nvd.nist.gov/vuln/detail/CVE-2017-1000376)

https://github.com/nxhack/openwrt-node-packages/tree/dev_CC/node-ffi

cooboyno111, 2017-08-31:
root@k-wifi:/tmp# node t.js 
/usr/lib/node/ffi/lib/dynamic_library.js:74
    throw new Error('Dynamic Linking Error: ' + err)
    ^

Error: Dynamic Linking Error: File not found
    at new DynamicLibrary (/usr/lib/node/ffi/lib/dynamic_library.js:74:11)
    at Object.Library (/usr/lib/node/ffi/lib/library.js:45:12)
    at Object.<anonymous> (/tmp/t.js:3:16)
    at Module._compile (module.js:570:32)
    at Object.Module._extensions..js (module.js:579:10)
    at Module.load (module.js:487:32)
    at tryModuleLoad (module.js:446:12)
    at Function.Module._load (module.js:438:3)
    at Module.runMain (module.js:604:10)
    at run (bootstrap_node.js:389:7)

mt7628 openwrt。
qca9531 openwrt。

nxhack, 2017-08-31:
Is the "libffi" package installed on the target machine?

cooboyno111, 2017-08-31:
root@k-wifi:/tmp# node t.js 
null
1234
root@k-wifi:/tmp# cat  t.js 
var ffi = require('ffi');
 
var current = ffi.Library(null, {
  'atoi': [ 'int', [ 'string' ] ]
});
var v2=current.atoi('1234'); // 1234 
console.log(v2);

nxhack, 2017-08-31:
Hmm
on the target machine
```
ln -s /lib/libm.so.0 /lib/libm.so
```

closed 2017-09-03 by nxhack