nxhack/openwrt-node-packages #95: Pre compiled

scenaristeur, 2017-09-11:
Is there a repo where we can download precompiled image, to test . 
Thx

nxhack, 2017-09-12:
Hi @scenaristeur 
The node v6.x is supported in the LEDE project now. [Here](http://downloads.lede-project.org/) you will find packages of various architectures. If there is a necessary 'native module', I will try to put new node modules into the official package of LEDE.

scenaristeur, 2017-09-14:
thxs, i'll try ;-)

closed 2017-09-14 by scenaristeur

radiolee, 2018-01-22:
@nxhack what's the name of node package in LEDE?

nxhack, 2018-01-23:
Hi @radiolee 
```
opkg update
opkg list|grep ^node
```
I will add what I need in my custom packages to the  official OpenWrt packages.
It takes time since my Pull Request is not merged quickly.

radiolee, 2018-01-23:
@nxhack  thanks , I can't find node* before because it's a release version source. finally it's here for ar71xx: http://downloads.openwrt.org/snapshots/packages/mips_24kc/packages/