nxhack/openwrt-node-packages #124: small question about my build

p3x-robot, 2017-10-05:
Ciao!

How are you?

Is it totally wrong? Because a few LEDE node packages are removed, though I am not using (only node ande npm, the lest form the registry), but still is it totally wrong, what do you say what I am doing?

```dockerfile
RUN cp feeds.conf.default feeds.conf
RUN echo 'src-git node https://github.com/nxhack/openwrt-node-packages.git' >> feeds.conf
RUN echo 'src-git darkmatter git://github.com/apollo-ng/luci-theme-darkmatter.git' >> feeds.conf
RUN echo 'src-git redis https://github.com/patrikx3/lede-redis.git' >> feeds.conf
RUN ./scripts/feeds update -a
RUN ./scripts/feeds install -a
RUN rm -rf ./package/feeds/packages/node*
RUN ./scripts/feeds install -a -p node
RUN ./scripts/feeds update -a -p darkmatter
RUN ./scripts/feeds install luci-theme-darkmatter
RUN ./scripts/feeds update -a -p redis
RUN ./scripts/feeds install redis
```

Should it be in another order of setting up the feeds or what build order is good for the ```Dockerfile```?

I just remove all LEDE Node packages and I just selected yours,. but some of it are missing, which I am not using, but still, I would like to know what I am doing.

Thanks,
Patrik

nxhack, 2017-10-06:
Hi @p3x-robot 

Did you check the contents of ".config" ?

p3x-robot, 2017-10-08:
Yeah, it works, I just don't know if should be like

```Dockerfile
RUN cp feeds.conf.default feeds.conf
RUN echo 'src-git node https://github.com/nxhack/openwrt-node-packages.git' >> feeds.conf
RUN echo 'src-git darkmatter git://github.com/apollo-ng/luci-theme-darkmatter.git' >> feeds.conf
RUN echo 'src-git redis https://github.com/patrikx3/lede-redis.git' >> feeds.conf

RUN rm -rf ./package/feeds/packages/node*

RUN ./scripts/feeds update -a
RUN ./scripts/feeds install -a
RUN ./scripts/feeds install -a -p node
```

or

```Dockerfile
RUN cp feeds.conf.default feeds.conf
RUN echo 'src-git node https://github.com/nxhack/openwrt-node-packages.git' >> feeds.conf
RUN echo 'src-git darkmatter git://github.com/apollo-ng/luci-theme-darkmatter.git' >> feeds.conf
RUN echo 'src-git redis https://github.com/patrikx3/lede-redis.git' >> feeds.conf
RUN ./scripts/feeds update -a
RUN ./scripts/feeds install -a

RUN rm -rf ./package/feeds/packages/node*

RUN ./scripts/feeds install -a -p node
```

```config
#
# Node.js
#
CONFIG_PACKAGE_node=m

#
# Configuration
#
# CONFIG_NODEJS_4 is not set
# CONFIG_NODEJS_6 is not set
CONFIG_NODEJS_8=y
# CONFIG_NODEJS_ICU is not set
CONFIG_PACKAGE_node-alexa-app=m
CONFIG_PACKAGE_node-alexa-sdk=m
CONFIG_PACKAGE_node-alljoyn=m
# CONFIG_PACKAGE_node-arduino-firmata is not set
CONFIG_PACKAGE_node-authenticate-pam=m
CONFIG_PACKAGE_node-autobahn=m
CONFIG_PACKAGE_node-aws-iot-device-sdk=m
CONFIG_PACKAGE_node-aws-sdk=m
CONFIG_PACKAGE_node-azure-event-hubs=m
CONFIG_PACKAGE_node-azure-iot-device-amqp=m
CONFIG_PACKAGE_node-azure-iot-device-http=m
CONFIG_PACKAGE_node-azure-iot-device-mqtt=m
CONFIG_PACKAGE_node-azure-iothub=m
CONFIG_PACKAGE_node-bignum=m
CONFIG_PACKAGE_node-binaryjs=m
CONFIG_PACKAGE_node-bleacon=m
CONFIG_PACKAGE_node-bleno=m
CONFIG_PACKAGE_node-bluetooth-hci-socket=m
CONFIG_PACKAGE_node-bower=m
CONFIG_PACKAGE_node-browserify=m
CONFIG_PACKAGE_node-coap=m
CONFIG_PACKAGE_node-cpx=m
# CONFIG_PACKAGE_node-cylon is not set
# CONFIG_PACKAGE_node-cylon-firmata is not set
# CONFIG_PACKAGE_node-cylon-gpio is not set
# CONFIG_PACKAGE_node-cylon-i2c is not set
CONFIG_PACKAGE_node-eddystone-beacon=m
CONFIG_PACKAGE_node-ejdb=m
CONFIG_PACKAGE_node-express=m
CONFIG_PACKAGE_node-ffi=m
CONFIG_PACKAGE_node-firmata=m
CONFIG_PACKAGE_node-forever=m
CONFIG_PACKAGE_node-fuse-bindings=m
CONFIG_PACKAGE_node-hashmap=m
CONFIG_PACKAGE_node-hashtable=m
CONFIG_PACKAGE_node-i2c-bus=m
CONFIG_PACKAGE_node-ibmiotf=m
CONFIG_PACKAGE_node-ideino-linino-lib=m
CONFIG_PACKAGE_node-iothub-explorer=m
CONFIG_PACKAGE_node-johnny-five=m
CONFIG_PACKAGE_node-jsonfile=m
CONFIG_PACKAGE_node-level=m
CONFIG_PACKAGE_node-leveldown=m
CONFIG_PACKAGE_node-levelup=m
CONFIG_PACKAGE_node-logfmt=m
CONFIG_PACKAGE_node-lwm2m-node-lib=m
CONFIG_PACKAGE_node-mknod=m
CONFIG_PACKAGE_node-mocha=m
CONFIG_PACKAGE_node-modbus=m
CONFIG_PACKAGE_node-mqtt=m
CONFIG_PACKAGE_node-muzzley-client=m
CONFIG_PACKAGE_node-nconf=m
CONFIG_PACKAGE_node-nino-io=m
CONFIG_PACKAGE_node-nitrogen=m
CONFIG_PACKAGE_node-nitrogen-cli=m
CONFIG_PACKAGE_node-nitrogen-leveldb-store=m
CONFIG_PACKAGE_node-noble=m
CONFIG_PACKAGE_node-node-expat=m
CONFIG_PACKAGE_node-node-hid=m
CONFIG_PACKAGE_node-node-lambda=m
CONFIG_PACKAGE_node-node-red=m
CONFIG_PACKAGE_node-node-red-contrib-gpio=m
CONFIG_PACKAGE_node-node-red-node-arduino=m
CONFIG_PACKAGE_node-node-static=m
CONFIG_PACKAGE_node-npm=m
CONFIG_PACKAGE_node-nsp=m
CONFIG_PACKAGE_node-onoff=m
CONFIG_PACKAGE_node-pm2=m
CONFIG_PACKAGE_node-reverse-wstunnel=m
CONFIG_PACKAGE_node-rimraf=m
CONFIG_PACKAGE_node-rpio=m
# CONFIG_PACKAGE_node-serialport is not set
CONFIG_PACKAGE_node-simple-xmpp=m
CONFIG_PACKAGE_node-sleep=m
CONFIG_PACKAGE_node-socket.io=m
CONFIG_PACKAGE_node-socket.io-client=m
CONFIG_PACKAGE_node-socket.io-client-legacy=m
CONFIG_PACKAGE_node-socket.io-legacy=m
CONFIG_PACKAGE_node-soracom=m
CONFIG_PACKAGE_node-sqlite3=m
CONFIG_PACKAGE_node-statvfs=m
CONFIG_PACKAGE_node-twilio=m
CONFIG_PACKAGE_node-uglify-js=m
CONFIG_PACKAGE_node-usb=m
CONFIG_PACKAGE_node-ws=m
CONFIG_PACKAGE_node-xmpp=m
CONFIG_PACKAGE_node-xmpp-client=m
CONFIG_PACKAGE_node-xmpp-component=m
CONFIG_PACKAGE_node-xmpp-server=m
CONFIG_PACKAGE_node-yarn=m
```

nxhack, 2017-10-10:
https://lede-project.org/docs/guide-developer/use-buildsystem

https://lede-project.org/docs/guide-developer/use-buildsystem#image_configuration

closed 2017-10-10 by p3x-robot