nxhack/openwrt-node-packages #125: Do you guys why it does with D-LINK DIR 860L B1 error: Z_BUF_ERROR ?

p3x-robot, 2017-10-06:
Ciao!

I upgraded to LEDE 17.01.3, with my Linksys it works as expected, but with D-LINK I get this error with ```Z_BUF_ERROR```.
Is it a kernel setting or what could be?

```text
root@d-link:/media/storage/backup/daily/4-Thursday# daily
npm info it worked if it ends with ok
npm info using npm@5.3.0
npm info using node@v8.6.0
npm info attempt registry request try #1 at 09:13:18 GMT+0200 (CEST)
npm http request GET http://npm.patrikx3.com:4873/npm
npm info attempt registry request try #1 at 09:13:19 GMT+0200 (CEST)
npm http request GET http://npm.patrikx3.com:4873/npm-check-updates
npm info attempt registry request try #1 at 09:13:19 GMT+0200 (CEST)
npm http request GET http://npm.patrikx3.com:4873/yarn
npm http 200 http://npm.patrikx3.com:4873/yarn
npm http 200 http://npm.patrikx3.com:4873/npm-check-updates
npm http 200 http://npm.patrikx3.com:4873/npm
npm info outdated updating [ { dep: 
npm info outdated      Node {
npm info outdated        path: '/usr/lib/node_modules/npm',
npm info outdated        realpath: '/usr/lib/node_modules/npm',
npm info outdated        error: null,
npm info outdated        id: 1,
npm info outdated        package: [Object],
npm info outdated        parent: [Object],
npm info outdated        isLink: false,
npm info outdated        children: [Array],
npm info outdated        loaded: false,
npm info outdated        requiredBy: [],
npm info outdated        requires: [Array],
npm info outdated        missingDeps: {},
npm info outdated        missingDevDeps: {},
npm info outdated        phantomChildren: {},
npm info outdated        location: '/npm',
npm info outdated        userRequired: false,
npm info outdated        save: false,
npm info outdated        saveSpec: null,
npm info outdated        isTop: false,
npm info outdated        fromBundle: false,
npm info outdated        isInLink: false },
npm info outdated     depname: 'npm',
npm info outdated     current: '5.3.0',
npm info outdated     wanted: '5.5.1',
npm info outdated     latest: '5.4.2',
npm info outdated     req: '^5.3.0',
npm info outdated     what: 'npm@5.5.1' } ]
npm http fetch GET 200 http://npm.patrikx3.com:4873/npm 11894ms
npm http fetch GET 200 http://npm.patrikx3.com:4873/npm/-/npm-5.5.1.tgz 21502ms
npm ERR! code Z_BUF_ERROR
npm ERR! errno -5
npm ERR! unexpected end of file

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2017-10-06T07_14_39_095Z-debug.log
```

nxhack, 2017-10-06:
try
```
npm cache clean --force
```

p3x-robot, 2017-10-06:
ok, sorry, i just deleted the old npm, some error, now it works, thanks so much dude!!!!

p3x-robot, 2017-10-06:
your repo is awesome!!!!

closed 2017-10-06 by p3x-robot