nxhack/openwrt-node-packages #126: npm ERR! cb() never called!

p3x-robot, 2017-10-06:
Ciao!
How are you?
This is weird.
With 8.6.0, with Linksys it works, simply.
But with D-Link Dir 860-L B1, I get this weird error.

But with ```npm -g update```:
```text
npm ERR! cb() never called!

npm ERR! This is an error with npm itself. Please report this error at:
npm ERR!     <https://github.com/npm/npm/issues>

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2017-10-06T10_23_23_254Z-debug.log
```
More:
```text
root@d-link:/media/storage/backup/daily/4-Thursday# npm -v
5.4.2
```
Is it a ```kernel``` settings?

Though both Linksys WRT and D-Link the same ```npm -v``` => ```5.4.2```, so it looks like works.
Just irritating. :)

Why is it trying to update when it is already udpated?
Is it an ```npm``` error or the D-Link 860L B1 ```node``` something is different than ```Linksys WRT```?

p3x-robot, 2017-10-06:
Ok, I am wrong, with Linksys it works like this:
```text
root@digi:~# npm -v
5.5.1
root@digi:~# node -v
v8.6.0
root@digi:~# 
```

With D-Link DIR 860l B1:
```text
root@d-link:/# node -v
v8.6.0
root@d-link:/# npm -v
5.4.2
root@d-link:/# npm install -g npm@5.5.1
npm ERR! cb() never called!

npm ERR! This is an error with npm itself. Please report this error at:
npm ERR!     <https://github.com/npm/npm/issues>

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2017-10-06T18_13_34_071Z-debug.log
root@d-link:/# 
```

p3x-robot, 2017-10-06:
is it just the RAM is too small even with swap????

nxhack, 2017-10-07:
@p3x-robot 
node.js, I recommend using swap, I think you need to run enough memory. I have enabled swap.

p3x-robot, 2017-10-08:
yeah, i am  using it, just weird on Windows it updates to 5.4.2 and Linksys is updates itself to 5.5.1, and D Link updates to 5.4.2. Weird, but I guess everything works. So, I close it, thanks for helping
Have a nice day.
Patrik

closed 2017-10-08 by p3x-robot