nxhack/openwrt-node-packages #129: Can't install the package

lam560, 2017-10-10:
Hello, i'v an error with "make" command.
I'm running the latest version of LEDE on a router TP Link archer C2600

`root@LEDE:/# make
ash: make: not found`

`root@LEDE:/# ./scripts/feeds update node
ash: ./scripts/feeds: not found`

nxhack, 2017-10-10:
Hi @lam560 

These are the environment for cross-compiling.

Please read documents
https://lede-project.org/docs/guide-developer/build-system

lam560, 2017-10-11:
in your link there is 

> 1-4 GB of RAM are needed for the compilation stage

So it mean it's impossible to use homebridge on any mainstream router which have a maximum of 512GB of RAM like mine ??
if yes I can abort this now and save my time buying a raspberry.

Thx for your help

nxhack, 2017-10-12:
Hi @lam560 

Please read the documents carefully once more.

Documents for other target machines are also helpful.
https://docs.labs.mediatek.com/resource/linkit-smart-7688/en/tutorials/firmware-and-bootloader/build-the-firmware-from-source-codes

Old OpenWrt related documents are helpful as well.
http://linksprite.com/wiki/index.php5?title=Tutorial_of_compiling_OpenWRT

closed 2018-02-14 by nxhack