nxhack/openwrt-node-packages #132: remove LEDE node packages

p3x-robot, 2017-10-17:
Ciao!

How are you?

You are saying:
```bash
./scripts/feeds update node
rm ./package/feeds/packages/node
rm ./package/feeds/packages/node-arduino-firmata
rm ./package/feeds/packages/node-cylon
rm ./package/feeds/packages/node-hid
rm ./package/feeds/packages/node-serialport
./scripts/feeds install -a -p node
```

But there are more that have collision:
[image]

There are 8 packages.

Is it good to do like:
```bash
./scripts/feeds update node
rm ./package/feeds/packages/node* -rf
./scripts/feeds install -a -p node
```

Or is it there, that should be kept?

nxhack, 2017-10-17:
Hi @p3x-robot 

This process removes the node related module in the official package of openwrt. There are only these five in the official package.

https://github.com/openwrt/packages/tree/master/lang

p3x-robot, 2017-10-18:
Ciao!

How are you?

Ok, thanks for responding, so I guess given I use LEDE, I should remove all ```node*``` packages, right?
Since LEDE provides 8 and I am thinking remove all that start with ```node*``` like that.
Correct?

nxhack, 2017-10-18:
> Ok, thanks for responding, so I guess given I use LEDE, I should remove all node* packages, right?
> Since LEDE provides 8 and I am thinking remove all that start with node* like that.
> Correct?

Yes.  They are symbolic links. The "-rf" option is not required.

```
rm ./package/feeds/packages/node*
```

p3x-robot, 2017-10-18:
ok, thanks.

closed 2017-10-18 by p3x-robot