nxhack/openwrt-node-packages #134: Build

p3x-robot, 2017-10-25:
Ciao!

How are you?

Do you have an idea to speed up the NodeJs build? It is so slow. In my full build for the firmware and just a few packages about 2-3 hours is just ```NodeJs```. Out of 4-5 hours...

Thanks,
Patrik

nxhack, 2017-10-25:
Hi @p3x-robot 

Are you using ccache?
The first build is very slow, but the second build is super fast.

https://forum.lede-project.org/t/how-do-i-use-ccache-in-my-builds/1774

p3x-robot, 2017-10-25:
Ok, I will test it, thanks so much!

p3x-robot, 2017-10-25:
But if I change 8.7 to 8.8 it cannot use the cache, it will re-compile doesn't it?
But the others will be already built, right?

p3x-robot, 2017-10-25:
So I cannot use the cache if it is 8.7 to 8.8, right?
Or some code is kept?
Becuase I already save everything build in a ```docker```, but the problems is that the ```Node``` compile is what is so slow (2-3 hours with 2 different firmwares...)

utya1988, 2017-10-25:
Hello. Maybe somebody has firmware with hap-nodejs for arduino yun. Could you upload this image. Thanks.

p3x-robot, 2017-10-25:
This is a totally different issue.

nxhack, 2017-10-26:
> So I cannot use the cache if it is 8.7 to 8.8, right?

Yes.
Cache will not work if target source code update date/time or path changes.

p3x-robot, 2017-10-26:
ok thanks.

closed 2017-10-26 by p3x-robot