nxhack/openwrt-node-packages #135: firmware for yun

utya1988, 2017-10-25:
Hello. Maybe somebody has firmware with hap-nodejs for arduino yun. Could you upload this image. Thanks.

nxhack, 2017-10-26:
Hi @utya1988 

I am running homebridge /  hap-nodejs on Arduino Yun. Although it is slow ;), but it is working.

I have a plan to adopt the homebridge package in the official package on OpenWrt.
Once adopted as the OpenWrt official package, you will be able to get packages with various targets.
There is a problem that I  have to solve, it seems to take a while.

stay tuned

For custom firmware of Arduino yun, build youre own firmware with reference to the following site.

http://www.egrep.jp/wiki/index.php/OpenWrt_for_Arduino_Yun_cheat_sheet
https://github.com/nxhack/openwrt-yun-build-extra
https://github.com/nxhack/openwrt-arduino-packages
https://forum.lede-project.org/t/ardiuno-yun-kernel-size-under-1280k-requirement/3948
https://github.com/lede-project/source/pull/1145

utya1988, 2017-10-26:
thank you. I subscribed :-)

core-c, 2018-02-28:
Hi,
We have a **Yun** running (and pairing) with **Homekit**, but we were not able to finish pairing in time when using the FHEM+NodeJS+Homebridge solution (It took 7 minutes for the pair-setup to complete, and minutes more for the pair-verify). We changed the **culfw** software a bit to make it run on a Yun's atmega32u4, created a shield with a 868MHz CC1101 radio, and got that prototype running pretty quickly.
To add support for Homekit we had to do a lot of research. Thanks to your info (like the links shown in your post above) we were able to create our custom **LEDE 17.01.4** with the required FPU-emulation, and all needed packages, to get NodeJS running on a Yun. Meanwhile we know a lot more about cross-compiling, bootloaders, flashing any size kernel, extroot the Yun easily, etc. If you wish, i can share some useful information about the things we found out. There is not so very much specific info out there on the net. We spent months learning, testing, debugging..  From our 10 available Yuns, i busted only 1 :) because i flashed some wrong code into bootloader-space by accident..
In the end we could not use Yun+NodeJS+homebridge because it runs too slow (too slow to pair in time with an iPhone). We found **PHK**, and after some fiddling with the code to get it running, it was able to pair with iOS devices. Meanwhile we've rewritten and extended that code _completely_, and now we have clean, very structural and ordered code: all in one easy to use HAP-class.

I attached a readme file from our github (private, because it belongs to a company). The section about cross-compiling LEDE is old (Node?!), but the kernel flashing information is acurate.
[README.zip](https://github.com/nxhack/openwrt-node-packages/files/1766955/README.zip)

Thanks for sharing!
_C.. & FastFrank_

nxhack, 2018-03-01:
Hi, @core-c 

I appreciate very nice information and great insights.
I wish you for the success of your project.
Please let me know if you can disclose information. I want to link to my Yun related information site.

Thank you very much.

core-c, 2018-03-01:
Thank you for your reply.
He just went on a holiday. :) good for him.
I will let you know more, after i’ve talked to my companion.
You understand..

You may use the readme i sent you as you like. No problem.
You have my permission to publish any part you like.

i attached a zip file, containing our configuration.
To this zip file conent is refered to in the readme i sent you before.
Most (if not all) Yun related files, are from your github. :) Thank you.
SETUP.zip may not be the very very last setup we used to cross-compile LEDE 17.01.4, but it a working set of config files.
We systematically added more packages (while keeping track of kernel & rootfs sizes), to find the minimal needed packages to make Homekit work on an Arduino Yún.
It was a learningful time for me. I am very familiar with the Atmel AVR MCUs, since 20 years or so.. but i had never before cross-compiled a complete firmware for a Yún, on a virtual machine Linux Debian running on a remote fat Apple Powercomp. Cool..

I would like to share our HAP.h & HAP.cpp, BUT i already sent those files to the maker of PHK. I want to await any reply from him first.
However.. I send you a copy of the files FOR PERSONAL USE ONLY please. Do not publish please.. it is just for ONLY YOU to examine (or use).
I send the HAP files, so you understand what i mean. It is working like a charm. But, at heart, it is still PHK.. 
PLEASE, do not share HAP. Thank you.

We also have the code to “bridge” with FHEM..    FHEM,h & FHEM.cpp    ..as we named those.
Those files i can not send.  Not now.

Our project is nearly at its finish. All is working great. We have two fully working prototypes now. The only thing the Yún now needs, is power.. and it serves well the iOS phones, the MAX! thermostats, shuttercontacts.. etc.
Our last change/addition to our yunCUL was an 128x32 OLED. There was just enough space on our shield to add an OLED display. There were some last-minute changes in the AVR software (related to I2C pins  already in use by the CC1101 radio, but needed for the OLED). We were able to relocate some pins/wires, adjust the software to use a different INT pin on the Yún, and the OLED integration was successful. Now it displays information (with little icons, and different fontsizes) when some status changes in FHEM/HAP.
The HAP part of our software has no sence of FHEM. It is a class on its own, it knows nothing about FHEM. The HAP class establishes connections with HAP-clients (people with their iPhones). It can handle Homekit communication. HAP knows about accessories, services, characteristics, crypting!, clients etc.. protocol-stuff,.. but on itself, it does not control any thermostat, or whatever sensor/actor.
The FHEM part of our software is much more related to the radio (CC1101 868MHz). We run the radio-software on the Yún’s atmega32u4, and we run the automation controller (FHEM) on the Linux side of the Yún. The only hardwired serial connection AVR<->Linux is Serial1: We changed the radio-code (culfw) to use that physical connection (AVR<->Linux) instead of the usual USB-port connected CUL (a CUL comes in the form of a USBstick, a radio onboard). Culfw is great! It supports many radio protocols (manufacturers).
We only slightly changed the original culfw firmware to make it work with our custom Yún-shield. After two weeks we got it running well.
FHEM is the class that handles the radio transmissions: communication between FHEM-Homekit/HAP-MAX!. Since our OLED has become part of the culfw software, FHEM also handles all OLED display output. Our OLED uses the default hardware pins for the I2C connection with the Yún, and we relocated the radio’s INTerupt pin to INT.6 on the Yún (that is Arduino pin 7).

Anyway,
there is a lot of research done, by us UJE, before we got it all up and running. We made a complete detour with that NodeJS path (we successfully went through), but ended up “just” compiling the latest LEDE/OpenWRT for our Yúns with the necessary packages/firmware.
We now use that ready-to-use—and-already-(cross)compiled LEDE to cross-compile our own (PHK derived) HAP package. (y)

Greetings,
Ron Driessen.

core-c, 2018-03-02:
Since this topic is about ***Yun firmware***, i'd like to add:
Although we've cross-compiled many (ever so slightly modified) versions of Yun firmware (orginal Yun, Alt.Adj, Chaos, LEDE... ), we were never able to make a working ***sysupgrade*** bin file.
We always flashed our kernel- & rootfs-bins seperately, using tftp from the bootloader prompt.
This is one issue for us to solve.. still.  ;) ..but as development continued, ..

nxhack, 2018-03-08:
> we were never able to make a working sysupgrade bin file.

The kernel size on LEDE-17.01 (openwrt-yun) must be within 1280 kb.
Have you disabled "CONFIG_KERNEL_KALLSYMS"?

core-c, 2018-03-08:
CONFIG_KERNEL_KALLSYMS=y

Although we did manage to create kernels that were <1280K, for our purpose (including FHEM+Homekit support) we could not fit all within that size. Our latest kernel is bigger: 1472k(kernel),14528k(rootfs).
```
setenv board "Yun";
setenv mtdparts "spi0.0:256k(u-boot)ro,64k(u-boot-env),1472k(kernel),14528k(rootfs),64k(art)ro";
setenv bootcmd "run addboard; run addtty;run addparts; run addrootfs; bootm 0x9f050000";
saveenv
```
To be able to do that (flash a >1280K kernel), it is neccessary to upgrade the default Yun bootloader. As you may know already, the default bootloader is not capable of saving any value into the u-boot-environment (saveenv).

We prepare to send you more information.
(did you receive the attached files i sent you per email? i already sent you our config files)

nxhack, 2018-03-08:
I actually create a sysupgrade-bin-image (for lede-17.01) and actually use it.
(The parameters of u-boot remain the original yun.)
https://github.com/nxhack/openwrt-yun-build-extra/tree/lede-17.01

Patch for kernel config:
https://github.com/nxhack/openwrt-yun-build-extra/blob/lede-17.01/patches/LEDE-17.01-MIPS24Kc%2BPCI%2BFPU_EMU.patch

Openwrt .config:
https://github.com/nxhack/openwrt-yun-build-extra/blob/lede-17.01/lede-17.01-yun-lininoos.config

core-c, 2018-03-13:
Thanks, i will look into it.
Meanwhile, i still have no response from China, so i say: publish what You want Hirokazu.  <<
if (ch--) jp++; // ;)
Greetings..

core-c, 2018-03-17:
I've posted more code, a Fritzing design, and some photos of our prototype on the FHEM forum:
[https://forum.fhem.de/index.php/topic,85688.0.html](url)

closed 2018-06-18 by nxhack