nxhack/openwrt-node-packages #140: Trace/breakpoint trap on MIPS 24KEc

mitsos1os, 2017-10-27:
Hi there!
First of all congrats for the work you have done here. It has been a great help for us during development of our LEDE based device.

We have a previous generation of the devices that successfully builds and runs NodeJS 6.9 version without any problems at all using your package.

We tried now upgrading the firmware because we needed the new functionality of Node 8. The build completes successfully, and the node binary is executable... However at certain periods of time the code exits, without any error but a message of **Trace/breakpoint trap**. Can you help me with this please?

Some info on the device:
**OS:** LEDE v17.01.4
**CPU**: MIPS 24KEc V5.0
**SYSTEM:** Mediatek MT7620A
**LEDE TARGET PROFILE**: ZBT-WE826
**NODE PACKAGE VERSION**: v8.8.1

Please note that in the previous version (6.9.1) it worked flawlessly!
Thank you for your time

nxhack, 2017-10-30:
Since there is a possibility that "soft-float" may not be taken into consideration on newer version , please disable  "CONFIG_SOFT_FLOAT" option and build and try it out?

```
make menuconfig
```
> Advanced configuration options (for developers)  --->
> Target Options  --->
> Use software floating point by default   [TURN OFF]

mitsos1os, 2017-10-30:
Thank you for taking the time to answer. But since the CPU hasn't got a hard FPU, shouldn't this be enabled for the V8 to work?

mitsos1os, 2017-11-02:
Just an update... we believe we narrowed down the issue to the level* modules...

mitsos1os, 2017-11-03:
The issue remains for the level* modules. As for the inspector, it is disabled if non-ICU installation is selected...

nxhack, 2017-11-06:
It looks like a complex event...
level* module has been updated to 2.0.0. I will deal with it.

mitsos1os, 2017-11-06:
Thank you for your help. The issue now is definitely in level down modules...we have isolated the reproduce. I have opened an issue [here](https://github.com/Level/leveldown/issues/403)
If you need anything please tell me

nxhack, 2018-01-17:
@mitsos1os 

How is your situation after that?
Does  node.js v8.x work?
There is a discussion that it is better to have the default version of node.js be v8.x. What do you think?

nxhack, 2018-01-17:
> Unfortunately we did not receive any support from the leveldown repo do we let it be.

sad...

The Greenkeeper bot is storming. :)

nxhack, 2018-01-17:
OK.
I will fix it so that the "small-icu" option can be used.

```
--with-intl=small-icu
```

nxhack, 2018-01-18:
@mitsos1os 
Updated.
You can select "small-icu" option.
That enables "v8_inspector".
Please check.

Best, regard

mitsos1os, 2018-01-18:
Thanks a lot

nxhack, 2018-03-08:
As an experiment, I modified it to change leveldb to the latest version of upstream. It may be worth a try.

https://github.com/nxhack/openwrt-node-packages/tree/leveldb-master/node-leveldown

mitsos1os, 2018-03-12:
Thank you for the update, the truth is due to this kind of problems, we have abandoned MIPS as a Node v8+ Platform... If I find some time, I will give it a try and update you with the results!

nxhack, 2018-03-13:
The node.js official package of OpenWrt is now v8.x. Let's leave this issue open as this thread's bug is a concern.

nxhack, 2018-07-11:
I will close this issue. Please reopen if necessary.

closed 2018-07-11 by nxhack