nxhack/openwrt-node-packages #141: node 8.9.0 is here are you gonna add in 9.x.y as well??

p3x-robot, 2017-10-31:
ciao!

how are you?

node 8.9.0 is here are you gonna add in 9.x.y as well??

patrik

nxhack, 2017-11-01:
merged

p3x-robot, 2017-11-01:
thank you so much!

p3x-robot, 2017-11-01:
hey!

How are you?
Do you know what it could be?
```text
Package node-ejdb is missing dependencies for the following libraries:
liblzma.so.5
```

When:
[image]

p3x-robot, 2017-11-01:
Is it missing, add like this?

```Makefile
define Package/node-ejdb
   SUBMENU:=Node.js
   SECTION:=lang
   CATEGORY:=Languages
   TITLE:=EJDB NodeJS binding
   URL:=https://www.npmjs.com/package/ejdb
   DEPENDS:=+node +libbz2 +liblzma
endef
```

nxhack, 2017-11-06:
@p3x-robot 

Thank you. node-ejdb merged.

closed 2017-11-08 by p3x-robot