nxhack/openwrt-node-packages #192: What does the --build-from-source mean?

p3x-robot, 2018-01-26:
Ciao!

How are you?

I try to build everything from NODE @nxhack, but there are some libs that I cannot find in the ```.config```, for example:
* ffi (I found in ```.config``` as ```CONFIG_PACKAGE_libffi=m``` **GOOD**
* libavahi-compat-libdnssd (I found, I could addit in the firmware by default, with ```LEDE``` ```image-builder```) **GOOD**
* mdns (libavahi-compat-libdnssd auto selects it, so that is good, I think) **GOOD**
* ref (not in the ```.config```, like a  ```CONFIG_PACKAGE_libref```?)
* serialport (not in the ```.config```, like a ```CONFIG_PACKAGE_libserialport```?)
* utf-8-validate (not in the ```.config```, like a ```CONFIG_PACKAGE_libutf_8_validate```?)
* bufferutil (not in the ```.config```, like a  ```CONFIG_PACKAGE_libbufferutil```?)
* leveldown (not in the ```.config```, like a  ```CONFIG_PACKAGE_libleveldown```?)
* bcrypt  (not in the ```.config```, like a  ```CONFIG_PACKAGE_libbcrypt```?)

p3x-robot, 2018-01-26:
Hey! How are you?

So now I see these are ```NPM``` packages:
* serialport
* utf-8-validate
* bufferutil
* ffi
* ref
* mdns
* leveldown
* bcrypt

But how can I build it with cross-compilation?

I guess from from ```Debian Docker``` where I built, I cannot just install ```node``` and just install like:
```bash
npm i -g serialport utf-8-validate bufferutil ffi ref mdns leveldown bcrypt
```

So how do I do it? :)

p3x-robot, 2018-01-26:
Just getting these errors, although I am root in ```Debian Docker```:
```text
prebuild-install info begin Prebuild-install version 2.5.0
prebuild-install info install --build-from-source specified, not attempting download.
gyp info it worked if it ends with ok
gyp info using node-gyp@3.6.2
gyp info using node@9.4.0 | linux | x64
gyp ERR! configure error 
gyp ERR! stack Error: EACCES: permission denied, mkdir '/build/source/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/node-node-hid-0.7.2/ipkg-install/usr/lib/node_modules/node-hid/build'
gyp ERR! System Linux 4.12.0-2-amd64
gyp ERR! command "/build/source/staging_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/host/bin/node" "/build/source/staging_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/host/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /build/source/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/node-node-hid-0.7.2/ipkg-install/usr/lib/node_modules/node-hid
gyp ERR! node -v v9.4.0
gyp ERR! node-gyp -v v3.6.2

gyp info it worked if it ends with ok
gyp info using node-gyp@3.6.2
gyp info using node@9.4.0 | linux | x64
gyp ERR! configure error 
gyp ERR! stack Error: EACCES: permission denied, mkdir '/build/source/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/node-node-expat-2.3.16/ipkg-install/usr/lib/node_modules/node-expat/build'
gyp ERR! System Linux 4.12.0-2-amd64
gyp ERR! command "/build/source/staging_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/host/bin/node" "/build/source/staging_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/host/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /build/source/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/node-node-expat-2.3.16/ipkg-install/usr/lib/node_modules/node-expat
gyp ERR! node -v v9.4.0
gyp ERR! node-gyp -v v3.6.2
gyp ERR! not ok 
npm info lifecycle node-expat@2.3.16~install: Failed to exec install script
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! node-expat@2.3.16 install: `node-gyp rebuild`
npm ERR! Exit status 1

prebuild info begin Prebuild version 2.9.2
prebuild ERR! configure error 
prebuild ERR! stack Error: EACCES: permission denied, mkdir '/build/source/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/node-mknod-1.1.0/ipkg-install/usr/lib/node_modules/mknod/build'
prebuild ERR! not ok 
prebuild ERR! build Error: EACCES: permission denied, mkdir '/build/source/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/node-mknod-1.1.0/ipkg-install/usr/lib/node_modules/mknod/build'
npm info lifecycle mknod@1.1.0~install: Failed to exec install script
npm ERR! code ELIFECYCLE
npm ERR! errno 2
npm ERR! mknod@1.1.0 install: `prebuild --download`
npm ERR! Exit status 2
```

Is this possible with Linksys WRT with LEDE  17.01.4?

nxhack, 2018-01-26:
Hi @p3x-robot 

There are modules in node.js that consist only of javascript, and others that need to build a native binary that needs to be compiled. Since the library that requires link is described in DEPENDS of Makefile, setting of .config necessary for OpenWrt is done automatically.
In my Makefile there is also a "--build-from-source" option for modules without native binary, but this is incorrect.

Perhaps it is better to delete javascript-only modules in this repository. It only needs to "npm install" on the target machine.

>     serialport
>     utf-8-validate
>     bufferutil
>     ffi
>     ref
>     mdns
>     leveldown
>     bcrypt
> 
serialport :  	node-serialport package exist
ffi: node-ffi package exist
mdns: node-mdns package exist
leveldown : node-leveldown package exist

If necessary I made a Makefile so please tell me which module.

nxhack, 2018-01-26:
@p3x-robot 

> gyp ERR! stack Error: EACCES: permission denied, mkdir '/build/source/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/node-node-hid-0.7.2/ipkg-install/usr/lib/node_modules/node-hid/build'

Check directory permisson.

nxhack, 2018-01-26:
maybe TOPDIR env not setting.
Please check your openwrt/lede cross compile environment.

p3x-robot, 2018-01-26:
What is strange is, that it tries to create directory:

```Error: EACCES: permission denied,  mkdir '/build/source/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/node-homebridge-0.4.36/ipkg-install/usr/lib/node_modules/homebridge/node_modules/mdns/build'```

Shoudn't create it ```mkdir -p``` ?

nxhack, 2018-01-26:
Please check cross compile environment.
I think that "rules.mk" is not exist in your cross build environment.

p3x-robot, 2018-01-26:
Hello!
How are you?

This is weird I have a ```rules.mk``` in the source. It looks like when it install with ```npm install``` it uses a different directory, multiple:
* ```/build/source/staging_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/host/lib/node_modules```
* ```/build/source/staging_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/root-mvebu/usr/lib/node_modules``` 

Something with the ```node-gyp``` is trying different directories like,because it is a uses ```node_modules```:

```PREFIX="/build/source/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/node-usb-1.3.1/ipkg-install/usr/" npm install --build-from-source --target_arch=arm -g /build/source/dl/usb-1.3.1.tgz```

So now we have 3 ```node_modules```, 

If I do this:
```make package/feeds/node/node-usb/{clean,prepare,compile} package/index V=s```

I get this error:

```Error: EACCES: permission denied, mkdir '/build/source/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/node-usb-1.3.1/ipkg-install/usr/lib/node_modules/usb/build'```

Of course 
```/build/source/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/node-usb-1.3.1/ipkg-install/usr/lib/node_modules/``` is not existing

It is only not working if there is ```node-gyp```, so werid.

#  My ```rules.mk``` is:

It is the default ```LEDE``` rules.

```mk
#
# Copyright (C) 2006-2010 OpenWrt.org
# Copyright (C) 2016 LEDE Project
#
# This is free software, licensed under the GNU General Public License v2.
# See /LICENSE for more information.
#

ifneq ($(__rules_inc),1)
__rules_inc=1

ifeq ($(DUMP),)
  -include $(TOPDIR)/.config
endif
include $(TOPDIR)/include/debug.mk
include $(TOPDIR)/include/verbose.mk

ifneq ($(filter check,$(MAKECMDGOALS)),)
CHECK:=1
DUMP:=1
endif

export TMP_DIR:=$(TOPDIR)/tmp

qstrip=$(strip $(subst ",,$(1)))
#"))

empty:=
space:= $(empty) $(empty)
comma:=,
merge=$(subst $(space),,$(1))
confvar=$(call merge,$(foreach v,$(1),$(if $($(v)),y,n)))
strip_last=$(patsubst %.$(lastword $(subst .,$(space),$(1))),%,$(1))

paren_left = (
paren_right = )
chars_lower = a b c d e f g h i j k l m n o p q r s t u v w x y z
chars_upper = A B C D E F G H I J K L M N O P Q R S T U V W X Y Z

define sep
[... 346 lines cut ...]

# file extension
ext=$(word $(words $(subst ., ,$(1))),$(subst ., ,$(1)))

all:
FORCE: ;
.PHONY: FORCE

check: FORCE
	@true

val.%:
	@$(if $(filter undefined,$(origin $*)),\
		echo "$* undefined" >&2, \
		echo '$(subst ','"'"',$($*))' \
	)

var.%:
	@$(if $(filter undefined,$(origin $*)),\
		echo "$* undefined" >&2, \
		echo "$*='"'$(subst ','"'\"'\"'"',$($*))'"'" \
	)

endif #__rules_inc
```

nxhack, 2018-01-26:
OK, OpenWrt/LEDE cross compile, It is recommended that you do so with user permissions, not root.
Therefore, the authority of the user is required for the director to use. Check the directory and file permissions.

nxhack, 2018-01-26:
```
cd "your OpenWrt/LEDE build env"
make menuconfig
make
```

p3x-robot, 2018-01-26:
yes thanks, i was easier with root, but i still have the comments, so now i use a docker user.
i hope that will work.
Thanks so much!
I think that was it!
In docker to build you need a user, root is not good!

p3x-robot, 2018-01-26:
If it works I will close it, thanks so much again! take care @nxhack !

p3x-robot, 2018-01-27:
Hi!

How are you?

Ok it works, I built it on Debian Jessie Docker, only these were not building;

```text
CONFIG_PACKAGE_node-alljoyn=m
CONFIG_PACKAGE_node-ffi=m
CONFIG_PACKAGE_node-ideino-linino-lib=m
CONFIG_PACKAGE_node-xmpp=m
CONFIG_PACKAGE_node-xmpp-client=m
CONFIG_PACKAGE_node-xmpp-component=m
CONFIG_PACKAGE_node-xmpp-server=m
```

I am investigating, why it didn't build it. Maybe I will have questions.

p3x-robot, 2018-01-27:
Ciao @nxhack !

How are you?
So I list the errors.

# node-alljoyn

## make

```bash
make package/feeds/node/node-alljoyn/{clean,prepare,compile} package/index V=s
```
## error output

Looks like the ```tar``` error.

```text
docker@d09e95869b86:/build/source$ make package/feeds/node/node-alljoyn/{clean,prepare,compile} package/index V=s
make[1]: Entering directory '/build/source'
make[2]: Entering directory '/build/source/feeds/node/node-alljoyn'
rm -f /build/source/staging_dir/target-arm_cortex-a53+neon-vfpv4_musl-1.1.16_eabi/stamp/.node-alljoyn_installed
rm -f /build/source/staging_dir/target-arm_cortex-a53+neon-vfpv4_musl-1.1.16_eabi/packages/node-alljoyn.list /build/source/staging_dir/host/packages/node-alljoyn.list
rm -rf /build/source/build_dir/target-arm_cortex-a53+neon-vfpv4_musl-1.1.16_eabi/node-alljoyn-1.0.1
make[2]: Leaving directory '/build/source/feeds/node/node-alljoyn'
make[1]: Leaving directory '/build/source'
make[1]: Entering directory '/build/source'
make[2]: Entering directory '/build/source/feeds/node/node-alljoyn'
. /build/source/include/shell.sh; xzcat /build/source/dl/node-alljoyn-1.0.1.tar.xz | tar -C /build/source/build_dir/target-arm_cortex-a53+neon-vfpv4_musl-1.1.16_eabi/node-alljoyn-1.0.1/.. -xf - 
tar: This does not look like a tar archive
tar: Exiting with failure status due to previous errors
Makefile:73: recipe for target '/build/source/build_dir/target-arm_cortex-a53+neon-vfpv4_musl-1.1.16_eabi/node-alljoyn-1.0.1/.prepared_4a2106016bd26944f4cd02c3a7987caa' failed
make[2]: *** [/build/source/build_dir/target-arm_cortex-a53+neon-vfpv4_musl-1.1.16_eabi/node-alljoyn-1.0.1/.prepared_4a2106016bd26944f4cd02c3a7987caa] Error 2
make[2]: Leaving directory '/build/source/feeds/node/node-alljoyn'
package/Makefile:105: recipe for target 'package/feeds/node/node-alljoyn/prepare' failed
make[1]: *** [package/feeds/node/node-alljoyn/prepare] Error 2
make[1]: Leaving directory '/build/source'
/build/source/include/toplevel.mk:198: recipe for target 'package/feeds/node/node-alljoyn/prepare' failed
make: *** [package/feeds/node/node-alljoyn/prepare] Error 2
```

# node-ffi

## make

```bash
make package/feeds/node/node-ffi/{clean,prepare,compile} package/index V=s
```
## error output

Looks like ```recipe for target 'Release/obj.target/ffi_bindings/src/ffi.o' failed``` error.

```text
/build/source/staging_dir/target-arm_cortex-a53+neon-vfpv4_musl-1.1.16_eabi/usr/include/node/v8config.h:318:3: note: in definition of macro 'V8_DEPRECATED'
   declarator __attribute__((deprecated(message)))
   ^
/build/source/staging_dir/target-arm_cortex-a53+neon-vfpv4_musl-1.1.16_eabi/usr/include/node/v8.h:3115:29: note:   no known conversion for argument 3 from 'v8::PropertyAttribute' to 'v8::Local<v8::Value>'
                 Maybe<bool> ForceSet(Local<Context> context, Local<Value> key,
                             ^
/build/source/staging_dir/target-arm_cortex-a53+neon-vfpv4_musl-1.1.16_eabi/usr/include/node/v8config.h:318:3: note: in definition of macro 'V8_DEPRECATED'
   declarator __attribute__((deprecated(message)))
   ^
ffi_bindings.target.mk:95: recipe for target 'Release/obj.target/ffi_bindings/src/ffi.o' failed
make[3]: *** [Release/obj.target/ffi_bindings/src/ffi.o] Error 1
make[3]: Leaving directory '/build/source/build_dir/target-arm_cortex-a53+neon-vfpv4_musl-1.1.16_eabi/node-ffi-2.2.0/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/build/source/staging_dir/target-arm_cortex-a53+neon-vfpv4_musl-1.1.16_eabi/host/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:258:23)
gyp ERR! stack     at ChildProcess.emit (events.js:160:13)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:209:12)
gyp ERR! System Linux 4.12.0-2-amd64
[... 69 lines cut ...]
## make

Have to not, that ```prepere``` is not working, so I removed that like below:

```bash
make package/feeds/node/node-xmpp/{clean,compile} package/index V=s
```
## error output

Looks like the ```tar``` error.

```text
make[2]: Entering directory '/build/source/feeds/node/node-xmpp'
. /build/source/include/shell.sh; xzcat /build/source/dl/node-xmpp-1.1.0.tar.xz | tar -C /build/source/build_dir/target-arm_cortex-a53+neon-vfpv4_musl-1.1.16_eabi/node-xmpp-1.1.0/.. -xf - 
tar: This does not look like a tar archive
tar: Exiting with failure status due to previous errors
Makefile:100: recipe for target '/build/source/build_dir/target-arm_cortex-a53+neon-vfpv4_musl-1.1.16_eabi/node-xmpp-1.1.0/.prepared_87312aadc8bd5693a63d6baf5cdb038e' failed
make[2]: *** [/build/source/build_dir/target-arm_cortex-a53+neon-vfpv4_musl-1.1.16_eabi/node-xmpp-1.1.0/.prepared_87312aadc8bd5693a63d6baf5cdb038e] Error 2
make[2]: Leaving directory '/build/source/feeds/node/node-xmpp'
package/Makefile:105: recipe for target 'package/feeds/node/node-xmpp/compile' failed
make[1]: *** [package/feeds/node/node-xmpp/compile] Error 2
make[1]: Leaving directory '/build/source'
/build/source/include/toplevel.mk:198: recipe for target 'package/feeds/node/node-xmpp/compile' failed
make: *** [package/feeds/node/node-xmpp/compile] Error 2
```

p3x-robot, 2018-01-27:
@nxhack of course all rest is built in 3 devices types. (Linksys WRT, D-LINK DIR 860L B1, R-PI3) All same errors.

closed 2018-01-27 by p3x-robot