nxhack/openwrt-node-packages #193: node-alljoyn-1.0.1

p3x-robot, 2018-01-27:
Ciao!

How are you

The Makefile is using ```.tar.xz```, it should end with ```.tar.gz```.

Then it can download via GIT and builds.

```Makefile
PKG_SOURCE:=$(PKG_NAME)-$(PKG_VERSION).tar.gz
```

Instead of 
```Makefile
PKG_SOURCE:=$(PKG_NAME)-$(PKG_VERSION).tar.xz
```

p3x-robot, 2018-01-27:
Pull merge: https://github.com/nxhack/openwrt-node-packages/pull/194

nxhack, 2018-01-27:
Hi @p3x-robot 
In OpenWrt packages, it is required to specify xz when downloading with git for the purpose of compressing the size.
Did you have any problems?

p3x-robot, 2018-01-28:
Hi!

How are you?

That's weird, because I couldn't build it with ```LEDE```, only if I change it to ```.gz```.
**Isn't ```OpenWrt``` is obsolete?**
In the ```GitHub```, the provided sources have no ```.xz```, only ```.gz```, or I don't know, but it doesn't work with ```LEDE```. I had to change it to build, that's all.

nxhack, 2018-01-28:
On OpenWrt/LEDE build system, when download source via git, first time it use 'git clone' command and packing it locally (tar.xz is recommended), Second time use that local archive.

p3x-robot, 2018-01-28:
Well, check out the build error, it is exactly the ```tar.xz```, when I change it to ```gz```, it works.

See for below:
https://github.com/nxhack/openwrt-node-packages/issues/192#issuecomment-360975527

nxhack, 2018-01-28:
I can build it normally in my environment(LEDE-17.01, node.js v9.x).

p3x-robot, 2018-01-28:
17.01 or 17.01.4? Because it shows that error, what it could be then??? Weird :+1:

nxhack, 2018-01-28:
I used lede-17.01 (head) branch.
```
make[2]: Entering directory '/home/ubuntu/17.01/openwrt/feeds/node/node-alljoyn'
mkdir -p /home/ubuntu/17.01/openwrt/dl
SHELL= flock /home/ubuntu/17.01/openwrt/tmp/.node-alljoyn-1.0.1.tar.xz.flock -c '       /home/ubuntu/17.01/openwrt/scripts/download.pl "/home/ubuntu/17.01/openwrt/dl" "node-alljoyn-1.0.1.tar.xz" "f29ae0b2e11d4a9ccedbe8cb752091988b19c3a4197e801c463b4a337e9d121a" "" || (  echo "Checking out files from the git repository..."; mkdir -p /home/ubuntu/17.01/openwrt/tmp/dl && cd /home/ubuntu/17.01/openwrt/tmp/dl && rm -rf node-alljoyn-1.0.1 && [ \! -d node-alljoyn-1.0.1 ] && git clone  https://github.com/octoblu/alljoyn.git node-alljoyn-1.0.1 && (cd node-alljoyn-1.0.1 && git checkout a74003fa25af1d0790468bf781a4d49347ec05c4 && git submodule update --init --recursive) && echo "Packing checkout..." && export TAR_TIMESTAMP=`cd node-alljoyn-1.0.1 && git log -1 --format='\''@%ct'\''` && rm -rf node-alljoyn-1.0.1/.git &&  tar --numeric-owner --owner=0 --group=0 --sort=name ${TAR_TIMESTAMP:+--mtime="$TAR_TIMESTAMP"} -c node-alljoyn-1.0.1 |        xz -zc -7e > /home/ubuntu/17.01/openwrt/tmp/dl/node-alljoyn-1.0.1.tar.xz && mv /home/ubuntu/17.01/openwrt/tmp/dl/node-alljoyn-1.0.1.tar.xz /home/ubuntu/17.01/openwrt/dl/ && rm -rf node-alljoyn-1.0.1;  )    '
+ curl -f --connect-timeout 20 --retry 5 --location --insecure http://sources.lede-project.org/node-alljoyn-1.0.1.tar.xz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:--  0:00:01 --:--:--     0
curl: (22) The requested URL returned error: 404 Not Found
Download failed.
+ curl -f --connect-timeout 20 --retry 5 --location --insecure http://mirror2.openwrt.org/sources/node-alljoyn-1.0.1.tar.xz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (22) The requested URL returned error: 404 Not Found
Download failed.
+ curl -f --connect-timeout 20 --retry 5 --location --insecure http://downloads.openwrt.org/sources/node-alljoyn-1.0.1.tar.xz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   184  100   184    0     0    277      0 --:--:-- --:--:-- --:--:--   277
  0     0    0     0    0     0      0      0 --:--:--  0:00:01 --:--:--     0
curl: (22) The requested URL returned error: 404 Not Found
Download failed.
No more mirrors to try - giving up.
Checking out files from the git repository...
Cloning into 'node-alljoyn-1.0.1'...
remote: Counting objects: 7502, done.
remote: Total 7502 (delta 0), reused 0 (delta 0), pack-reused 7502
Receiving objects: 100% (7502/7502), 18.42 MiB | 1.25 MiB/s, done.
Resolving deltas: 100% (3514/3514), done.
Checking connectivity... done.
Note: checking out 'a74003fa25af1d0790468bf781a4d49347ec05c4'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by performing another checkout.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -b with the checkout command again. Example:

  git checkout -b <new-branch-name>

HEAD is now at a74003f... v1.0.1 build pull requests
Packing checkout...
```
finally git clone succeeded. and build succeeded.

p3x-robot, 2018-01-28:
Ok, I found it.
If you pre-download the files, like:
```bash
make download V=s
IGNORE_ERRORS=1 make V=s
```

That's when it not working, because it download a zero file ```.tar.xz```, but if it is ```.tar.gz```, it works...
That's all.
I found it to pre-download is faster...

p3x-robot, 2018-01-28:
Besides, i think it will work always...
Weird, isn't it? :+1:

p3x-robot, 2018-01-28:
https://lede-project.org/docs/guide-developer/use-buildsystem#make_tips