nxhack/openwrt-node-packages #195: make package/feeds/node/node-ffi/{clean,prepare,compile} package/index V=s

p3x-robot, 2018-01-27:
Ciao!

How are you?

I haven't found the solution but this is the log, does it tell you anything?

```text
0 info it worked if it ends with ok
1 verbose cli [ '/build/source/staging_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/host/bin/node',
1 verbose cli   '/build/source/staging_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/host/bin/npm',
1 verbose cli   'install',
1 verbose cli   '--build-from-source',
1 verbose cli   '--target_arch=arm',
1 verbose cli   '-g',
1 verbose cli   '/build/source/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/node-ffi-2.2.0' ]
2 info using npm@5.6.0
3 info using node@v9.4.0
4 verbose npm-session 1d27ae147d808bb9
5 silly install loadCurrentTree
6 silly install readGlobalPackageData
7 silly pacote directory manifest for undefined@file:/build/source/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/node-ffi-2.2.0 fetched in 8ms
8 silly install loadIdealTree
9 silly install cloneCurrentTreeToIdealTree
10 silly install loadShrinkwrap
11 silly install loadAllDepsIntoIdealTree
12 silly resolveWithNewModule ffi@2.2.0 checking installable status
13 http fetch GET 200 https://registry.npmjs.org/debug 5162ms
14 http fetch GET 200 https://registry.npmjs.org/ref-struct 5161ms
15 http fetch GET 200 https://registry.npmjs.org/bindings 5167ms
16 http fetch GET 200 https://registry.npmjs.org/ref 5167ms
17 silly pacote range manifest for bindings@~1.2.0 fetched in 5180ms
18 silly resolveWithNewModule bindings@1.2.1 checking installable status
19 silly pacote range manifest for debug@2 fetched in 5169ms
20 silly resolveWithNewModule debug@2.6.9 checking installable status
21 silly pacote range manifest for ref@1 fetched in 5170ms
22 silly resolveWithNewModule ref@1.3.5 checking installable status
23 silly pacote range manifest for ref-struct@1 fetched in 5170ms
24 silly resolveWithNewModule ref-struct@1.1.0 checking installable status
25 http fetch GET 200 https://registry.npmjs.org/nan 10180ms
26 silly pacote range manifest for nan@2 fetched in 10182ms
27 silly resolveWithNewModule nan@2.8.0 checking installable status
28 http fetch GET 200 https://registry.npmjs.org/ms 40ms
29 silly pacote version manifest for ms@2.0.0 fetched in 48ms
30 silly resolveWithNewModule ms@2.0.0 checking installable status
31 silly currentTree lib
32 silly idealTree lib
32 silly idealTree `-- ffi@2.2.0
32 silly idealTree   +-- bindings@1.2.1
32 silly idealTree   +-- debug@2.6.9
32 silly idealTree   +-- ms@2.0.0
32 silly idealTree   +-- nan@2.8.0
32 silly idealTree   +-- ref-struct@1.1.0
32 silly idealTree   `-- ref@1.3.5
33 silly install generateActionsToTake
34 silly diffTrees action count 7
35 silly diffTrees add bindings@1.2.1
36 silly diffTrees add ms@2.0.0
37 silly diffTrees add debug@2.6.9
38 silly diffTrees add nan@2.8.0
39 silly diffTrees add ref@1.3.5
40 silly diffTrees add ref-struct@1.1.0
41 silly diffTrees add ffi@2.2.0
42 silly decomposeActions action count 54
43 silly decomposeActions fetch bindings@1.2.1
44 silly decomposeActions extract bindings@1.2.1
45 silly decomposeActions preinstall bindings@1.2.1
46 silly decomposeActions build bindings@1.2.1
47 silly decomposeActions install bindings@1.2.1
48 silly decomposeActions postinstall bindings@1.2.1
49 silly decomposeActions finalize bindings@1.2.1
[... 873 lines cut ...]
gyp ERR! node-gyp -v v3.6.2
gyp ERR! not ok 
npm info lifecycle ffi@2.2.0~install: Failed to exec install script
npm WARN rollback Rolling back bindings@1.2.1 failed (this is probably harmless): /build/source/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/node-ffi-2.2.0/node_modules/bindings is not a child of /build/source/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/node-ffi-2.2.0/ipkg-install/usr/lib
npm WARN rollback Rolling back ms@2.0.0 failed (this is probably harmless): /build/source/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/node-ffi-2.2.0/node_modules/ms is not a child of /build/source/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/node-ffi-2.2.0/ipkg-install/usr/lib
npm WARN rollback Rolling back debug@2.6.9 failed (this is probably harmless): /build/source/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/node-ffi-2.2.0/node_modules/debug is not a child of /build/source/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/node-ffi-2.2.0/ipkg-install/usr/lib
npm WARN rollback Rolling back nan@2.8.0 failed (this is probably harmless): /build/source/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/node-ffi-2.2.0/node_modules/nan is not a child of /build/source/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/node-ffi-2.2.0/ipkg-install/usr/lib
npm WARN rollback Rolling back ref@1.3.5 failed (this is probably harmless): /build/source/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/node-ffi-2.2.0/node_modules/ref is not a child of /build/source/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/node-ffi-2.2.0/ipkg-install/usr/lib
npm WARN rollback Rolling back ref-struct@1.1.0 failed (this is probably harmless): /build/source/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/node-ffi-2.2.0/node_modules/ref-struct is not a child of /build/source/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/node-ffi-2.2.0/ipkg-install/usr/lib
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! ffi@2.2.0 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the ffi@2.2.0 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /build/source/tmp/npm-cache/_logs/2018-01-27T18_55_54_012Z-debug.log
Makefile:76: recipe for target '/build/source/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/node-ffi-2.2.0/.built' failed
make[2]: *** [/build/source/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/node-ffi-2.2.0/.built] Error 1
make[2]: Leaving directory '/build/source/feeds/node/node-ffi'
package/Makefile:105: recipe for target 'package/feeds/node/node-ffi/compile' failed
make[1]: *** [package/feeds/node/node-ffi/compile] Error 2
```

p3x-robot, 2018-01-27:
It is Docker Debian Jessie.

nxhack, 2018-01-28:
Currently, the npm version of node-ffi has a problem with node.js v9.x. It has been fixed in github. I will take countermeasures.

nxhack, 2018-01-29:
@p3x-robot 
merged

closed 2018-02-02 by p3x-robot