nxhack/openwrt-node-packages #206: node-homebridge: update to 0.4.38

p3x-robot, 2018-02-08:
Hello!

How are you?
I get a new error from homebridge:
```text
Package node-homebridge is missing dependencies for the following libraries:
libdns_sd.so
```

Do you know what it could be?
Do I have to install an ```apt``` library?

Thanks,
Patrik

p3x-robot, 2018-02-08:
Or other packages are built so it is not missing the ```libdns_sd.so```, because my all packages are missing only these packages, like this (a new build for Linksys WRT):

So you can see, only these packages are missing exactly. Weird.

```text
Every 60,0s: echo It was Linksys 103 && find /var/www/cdn.corifeus.com/public/lede/17.01.4/p...  server: Thu Feb  8 15:04:17 2018

It was Linksys 103
101

R-PI 3 Was 103
103

D-LINK was 103
103

Total was 14630

Now 14628
```

p3x-robot, 2018-02-08:
I rebuild both routers now, maybe it was during a new update or something. But wierd, hey?

nxhack, 2018-02-09:
Hmm. It works for lede-17.01 and node v8.9.4. Next, I will check with node v9.5.0.

p3x-robot, 2018-02-09:
it works, it was updating after a build (a weird build) and it started a new build a new version of your ```repo```, so there was a new commit and updated the feeds then when it started to build a new target (1900ACS => 3200ACM), I know you can do it at once, but then i didnt know, so it was rebuilding during an update (given where if you re-build and some is done it is fast kinda) in the ```openwrt-node-packages``` and that was a problem, now builds perfectly. thanks and sorry for the issue. closed now.

closed 2018-02-09 by p3x-robot