nxhack/openwrt-node-packages #207: unable to get node in, FPU-emu works

bastup, 2018-02-13:
Hello NXhack, I am trying to run your work on an Archer C7 v2, ar71xx-arch.

I am able to get the image up and running on the device but node is not available on the image, even with your guide. 'opkg update node' tells me that it cannot find the mips24 node archive over at openwrt. Which is true, it does not exist. 

I am running the scripts as you point out under usage before i make menuconfig and make, and they return no errors. When I run make menuconfig there is a * next to node but node does not show up on the sysupgrade image. Any clue what has gone wrong?

nxhack, 2018-02-13:
Hi @bastup 

Have you selected the node package?
Please refer to this for details.

https://openwrt.org/doc/howto/build?s[]=buildroot#make_menuconfig

bastup, 2018-02-13:
Hello Hirokazu,

Yes, when I run make menuconfig the node package is marked with a * under
image configuration.

I am sure I've done something wrong. This is my method
within LEDE 17 trunk

./scripts/feeds update
./scripts/feeds install -a
add src-git node https://github.com/nxhack/openwrt-node-packages.git to
feeds.default.conf

./scripts/feeds update node
rm the node packages
./scripts/feeds install -a -p

This returns no error about node.

then I run make menuconfig ->chose my model ->add FPU-emulation -> check
image configuration that node is marked with a *, it is.
make

The image produced works on my archer C7 but node is not there. opkg update
and opkg install node throws error

Downloading
http://downloads.lede-project.org/snapshots/packages/mips_24kc/node/Packages.gz
*** Failed to download the package list from
http://downloads.lede-project.org/snapshots/packages/mips_24kc/node/Packages.gz

Any help would be appreciated. Thanks! /Jonatan

bastup, 2018-02-13:
> 24kc/node/Packages.gz

To clarify

./scripts/feeds install -a -p NODE

thanks

nxhack, 2018-02-13:
Check ./bin/packages/mips_24kc/node/ 
The packages (.ipk)  should be there.

> Downloading http://downloads.lede-project.org/snapshots/packages/mips_24kc/node/Packages.gz *** Failed to download the package list from http://downloads.lede-project.org/snapshots/packages/mips_24kc/node/Packages.gz 

Of course you do not have packages you build, not on site (downloads.lede-project.org). There is only in your hard disk.
>

nxhack, 2018-02-13:
& Your firmware is in ./bin/targets/ar71xx/generic/

bastup, 2018-02-13:
Hello Harokazu!

I have now gotten it to work. I misunderstood things in the make
menuconfig. I now have an image with node.js built in and it runs, slow,
but it runs.

It seems buggy though. When I run the example server from the documentation
the server runs but when I try to connect with the browser I get server not
found not the expected Hello World

Do you know why? have this happened to you? I chose the 9.5 version with
make menuconfig.

Thank you! /Jonatan

-----------------
const http = require('http');

const hostname = '127.0.0.1';const port = 8082;
const server = http.createServer((req, res) => {
  res.statusCode = 200;
  res.setHeader('Content-Type', 'text/plain');
  res.end('Hello World\n');});

server.listen(port, hostname, () => {
  console.log(`Server running at http://${hostname}:${port}/`);});

root@LEDE:/tmp# node test.js
Server running at http://127.0.0.1:8082/

nxhack, 2018-02-13:
> const hostname = '127.0.0.1';const port = 8082; 

You are binding to loopback address. In this case, you can not connect from the external network.

bastup, 2018-02-14:
Hello Hirokazu

Dear god, I am such an idiot, setting it to 192.168.1.1 works, of course.
Things seem a little slow but that is because of my device.

Thank you dearly /Jonatan

closed 2018-02-14 by nxhack