nxhack/openwrt-node-packages #213: node-ffi - r-pi 3

p3x-robot, 2018-02-17:
ciao!

how are you?
on the latest openwrt, node-ffi doesn't build, i will provide more info, but it is building right now and linksys wrt compatible and d-link dir860l b1 ramips compatible built, but with r-pi 3 cortex-a53  is wasn't compiling. 

i will compile by itself and comment the error.

take care,
patrik

p3x-robot, 2018-02-17:
Full R-PI 3 output:

```text
Cloning into 'node-ffi-2.2.1'...
remote: Counting objects: 4775, done.
remote: Compressing objects: 100% (2/2), done.
remote: Total 4775 (delta 1), reused 0 (delta 0), pack-reused 4773
Receiving objects: 100% (4775/4775), 3.03 MiB | 2.93 MiB/s, done.
Resolving deltas: 100% (2479/2479), done.
Checking connectivity... done.
Note: checking out '60282c29eb29eafb5d9b969b794a5ddf3e31e9d5'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by performing another checkout.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -b with the checkout command again. Example:

  git checkout -b new_branch_name

HEAD is now at 60282c2... travis: drop Node 0.10 and 0.12, add Node 8 and 9
Packing checkout...
touch /build/source/build_dir/target-aarch64_cortex-a53_musl/node-ffi-2.2.1/.prepared_ddd989775ee8ac1502fd924adb3956e6_6664517399ebbbc92a37c5bb081b5c53_check
. /build/source/include/shell.sh; xzcat /build/source/dl/node-ffi-2.2.1.tar.xz | tar -C /build/source/build_dir/target-aarch64_cortex-a53_musl/node-ffi-2.2.1/.. -xf - 
[ ! -d ./src/ ] || cp -fpR ./src/* /build/source/build_dir/target-aarch64_cortex-a53_musl/node-ffi-2.2.1

Applying ./patches/0001-do_not_build_libffi.patch using plaintext: 
patching file binding.gyp

Applying ./patches/0002-support_mips.patch using plaintext: 
patching file src/ffi.cc
touch /build/source/build_dir/target-aarch64_cortex-a53_musl/node-ffi-2.2.1/.prepared_ddd989775ee8ac1502fd924adb3956e6_6664517399ebbbc92a37c5bb081b5c53
rm -f /build/source/build_dir/target-aarch64_cortex-a53_musl/node-ffi-2.2.1/.configured_*
rm -f /build/source/staging_dir/target-aarch64_cortex-a53_musl/stamp/.node-ffi_installed
(cd /build/source/build_dir/target-aarch64_cortex-a53_musl/node-ffi-2.2.1/./; if [ -x ./configure ]; then find /build/source/build_dir/target-aarch64_cortex-a53_musl/node-ffi-2.2.1/ -name config.guess | xargs -r chmod u+w; find /build/source/build_dir/target-aarch64_cortex-a53_musl/node-ffi-2.2.1/ -name config.guess | xargs -r -n1 cp --remove-destination /build/source/scripts/config.guess; find /build/source/build_dir/target-aarch64_cortex-a53_musl/node-ffi-2.2.1/ -name config.sub | xargs -r chmod u+w; find /build/source/build_dir/target-aarch64_cortex-a53_musl/node-ffi-2.2.1/ -name config.sub | xargs -r -n1 cp --remove-destination /build/source/scripts/config.sub; AR="aarch64-openwrt-linux-musl-gcc-ar" AS="aarch64-openwrt-linux-musl-gcc -c -Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap/build/source/build_dir/target-aarch64_cortex-a53_musl/node-ffi-2.2.1:node-ffi-2.2.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/build/source/build_dir/target-aarch64_cortex-a53_musl/libffi-3.2.1/build" LD=aarch64-openwrt-linux-musl-ld NM="aarch64-openwrt-linux-musl-gcc-nm" CC="aarch64-openwrt-linux-musl-gcc" GCC="aarch64-openwrt-linux-musl-gcc" CXX="aarch64-openwrt-linux-musl-g++" RANLIB="aarch64-openwrt-linux-musl-gcc-ranlib" STRIP=aarch64-openwrt-linux-musl-strip OBJCOPY=aarch64-openwrt-linux-musl-objcopy OBJDUMP=aarch64-openwrt-linux-musl-objdump SIZE=aarch64-openwrt-linux-musl-size CFLAGS="-Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap/build/source/build_dir/target-aarch64_cortex-a53_musl/node-ffi-2.2.1:node-ffi-2.2.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/build/source/build_dir/target-aarch64_cortex-a53_musl/libffi-3.2.1/build " CXXFLAGS="-Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap/build/source/build_dir/target-aarch64_cortex-a53_musl/node-ffi-2.2.1:node-ffi-2.2.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/build/source/build_dir/target-aarch64_cortex-a53_musl/libffi-3.2.1/build " CPPFLAGS="-I/build/source/staging_dir/target-aarch64_cortex-a53_musl/usr/include -I/build/source/staging_dir/target-aarch64_cortex-a53_musl/include -I/build/source/staging_dir/toolchain-aarch64_cortex-a53_gcc-5.5.0_musl/usr/include -I/build/source/staging_dir/toolchain-aarch64_cortex-a53_gcc-5.5.0_musl/include/fortify -I/build/source/staging_dir/toolchain-aarch64_cortex-a53_gcc-5.5.0_musl/include " LDFLAGS="-L/build/source/staging_dir/target-aarch64_cortex-a53_musl/usr/lib -L/build/source/staging_dir/target-aarch64_cortex-a53_musl/lib -L/build/source/staging_dir/toolchain-aarch64_cortex-a53_gcc-5.5.0_musl/usr/lib -L/build/source/staging_dir/toolchain-aarch64_cortex-a53_gcc-5.5.0_musl/lib -znow -zrelro "   ./configure --target=aarch64-openwrt-linux --host=aarch64-openwrt-linux --build=x86_64-pc-linux-gnu --program-prefix="" --program-suffix="" --prefix=/usr --exec-prefix=/usr --bindir=/usr/bin --sbindir=/usr/sbin --libexecdir=/usr/lib --sysconfdir=/etc --datadir=/usr/share --localstatedir=/var --mandir=/usr/man --infodir=/usr/info --disable-nls  ; fi; )
touch /build/source/build_dir/target-aarch64_cortex-a53_musl/node-ffi-2.2.1/.configured_68b329da9893e34099c7d8ad5cb9c940
rm -f /build/source/build_dir/target-aarch64_cortex-a53_musl/node-ffi-2.2.1/.built
[... cut ...]
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /build/source/tmp/npm-cache/_logs/2018-02-17T19_26_59_333Z-debug.log
Makefile:76: recipe for target '/build/source/build_dir/target-aarch64_cortex-a53_musl/node-ffi-2.2.1/.built' failed
make[3]: *** [/build/source/build_dir/target-aarch64_cortex-a53_musl/node-ffi-2.2.1/.built] Error 1
make[3]: Leaving directory '/build/source/feeds/node/node-ffi'
    ERROR: package/feeds/node/node-ffi failed to build.
```

nxhack, 2018-02-19:
@p3x-robot 
fixed. but It is not tested whether it will run on the target machine or not.

p3x-robot, 2018-02-21:
ciao @nxhack !
How are you?
I confirmed, it is building.
Thanks,
@patrikx3

closed 2018-02-21 by p3x-robot