nxhack/openwrt-node-packages #224: LEDE 17.01.4 - node-openzwave-shared and node-things-gateway

p3x-robot, 2018-02-24:
ciao!

how are you?

does it only build for the ```OpenWrt``` ```master``` / ```snapshot``` only?
On ```OpneWrt``` it builts, but the stable ```LEDE``` is not building (```17.01.4```).
Just asking.

Thanks,
Patrik

p3x-robot, 2018-02-24:
hm, maybe the ```.config``` is missing it, i am checking it....

p3x-robot, 2018-02-24:
hey,
how are you?

it is weird, i doesn't show in the ```make menuconfig``` at all.
I created the images 9 hours ago and these packages were created 21 hours ago.

Do you have an idea?

Thanks,
Patrik

p3x-robot, 2018-02-24:
puzzling

nxhack, 2018-02-25:
Sziasztok! @p3x-robot 

The packages related to openzwave is not supported by lede-17.01. My packages of mozilla-IoT-gateway has not worked yet.