nxhack/openwrt-node-packages #226: Package request: npm-check-updates

p3x-robot, 2018-02-25:
Ciao @nxhack !

Can I create pull merge for ```npm-check-updates```?

Is it viable?

Sincerely,
Patrik

nxhack, 2018-02-26:
Pull Request welcome!

p3x-robot, 2018-02-26:
Hey @nxhack !

I have added it in, but it is not working like with other packages:

https://github.com/patrikx3/openwrt-node-packages/tree/master/node-npm-check-updates

Trying:
```bash
make feeds/node/node-npm-check-updates/download V=s
make feeds/node/node-npm-check-updates/check FIXUP=1 V=s
cat feeds/node/node-npm-check-updates/Makefile | less
```

Error:
```text
docker@6691bd919010:/build/source$ make feeds/node/node-npm-check-updates/download V=s
make[1]: Entering directory '/build/source'
cd "/build/source"; git log --format=%h -1 toolchain > /build/source/tmp/.ver_check
cmp -s /build/source/tmp/.ver_check /build/source/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-5.5.0_musl_eabi/stamp/.ver_check || { \
	rm -rf /build/source/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi /build/source/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi /build/source/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-5.5.0_musl_eabi /build/source/build_dir/toolchain-arm_cortex-a9+vfpv3_gcc-5.5.0_musl_eabi; \
	mkdir -p /build/source/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-5.5.0_musl_eabi/stamp; \
	mv /build/source/tmp/.ver_check /build/source/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-5.5.0_musl_eabi/stamp/.ver_check; \
}
make[1]: *** No rule to make target 'feeds/node/node-npm-check-updates/download'.  Stop.
make[1]: Leaving directory '/build/source'
/build/source/include/toplevel.mk:216: recipe for target 'feeds/node/node-npm-check-updates/download' failed
make: *** [feeds/node/node-npm-check-updates/download] Error 2
```

Do you know how  I can fix up the ```PKG_HASH```?

Thanks,
Patrik

nxhack, 2018-02-26:
```
./scripts/feeds install node-npm-check-updates
make oldconfig
make V=s package/feeds/node/node-npm-check-updates/compile
```

p3x-robot, 2018-02-26:
Ok, it works, is it possible to add a file like this:

```.gitignore```

```text
/.idea
```

Sincerely,
Patrik

p3x-robot, 2018-02-26:
https://github.com/nxhack/openwrt-node-packages/pull/228

p3x-robot, 2018-02-26:
Updated ```node-yarn``` as well.

p3x-robot, 2018-02-26:
https://cdn.corifeus.com/archive/

nxhack, 2018-02-27:
merged

closed 2018-02-27 by nxhack