nxhack/openwrt-node-packages #243: node-homebridge doesn't build anymore with LEDE 17.01.4

p3x-robot, 2018-03-05:
# Ciao!

## How are you?

This is weird, there is a new package, ```node-epoll``` did build, but now I can see there is an update with the ```node-homebridge``` and there are new depends:
```Makefile
  DEPENDS:=+node +node-npm +avahi-dbus-daemon +libavahi-compat-libdnssd
```

# But still, the packages are built that are required:
```text
docker@14a80110d9c4:/build/source/bin$ find -iname '*avahi-dbus-daemon*'
./packages/arm_cortex-a9_vfpv3/packages/avahi-dbus-daemon_0.6.32-1_arm_cortex-a9_vfpv3.ipk
docker@14a80110d9c4:/build/source/bin$ find -iname '*libavahi-compat-libdnssd*'
./packages/arm_cortex-a9_vfpv3/packages/libavahi-compat-libdnssd_0.6.32-1_arm_cortex-a9_vfpv3.ipk
```

The previous version was built successfully.

# The error:
```text
Package node-homebridge is missing dependencies for the following libraries:
libdns_sd.so
```

So weird!

Thanks,
Patrik

p3x-robot, 2018-03-05:
Besides this is selected:
```text
 * libavahi-compat-libdnssd (*)         
```

Just before the previous version.

nxhack, 2018-03-05:
Hi @p3x-robot 

> But still, the packages are built that are required:

This is normal behavior. The required packages will be built.

> Package node-homebridge is missing dependencies for the following libraries: libdns_sd.so

For lede-17.01, you must **explicitly** set the built-in flag 'libavahi-compat-libdnssd' in make config.
Otherwise, libdns_sd.so will not be created in the cross-compilation environment.

nxhack, 2018-03-06:
OK
In order to eliminate the complexity, I reverted to previous way.

closed 2018-03-06 by nxhack

p3x-robot, 2018-03-06:
thank you @nxhack !!! :100:

p3x-robot, 2018-03-06:
ok, confirmed, works.