nxhack/openwrt-node-packages #252: failed to compile

muradurm, 2018-03-12:
Failed to compile nodejs. Error during compilation:

> In file included from ../src/node.cc:37:0:
> ../src/node_crypto.h:39:25: fatal error: openssl/ssl.h: No such file or directory compilation terminated.

I tried it with SDK for openwrt 17.04 and 17.01 for ar711 and for x86 under ubuntu 17.10

nxhack, 2018-03-12:
This is probably the case for host build.
If so,
```
apt-get install -y libssl-dev
```

muradurm, 2018-03-13:
For sure, I followed the instruction and installed libssl-dev before trying to compile nodejs:
sudo apt-get install -y libssl-dev
Reading package lists... Done
Building dependency tree       
Reading state information... Done
libssl-dev is already the newest version (1.0.2g-1ubuntu4.10).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

But still had the fatal error: openssl/ssl.h: No such file or directory compilation terminated.

The full error output is:
`ccache_cxx '-DNODE_ARCH="ia32"' '-DNODE_PLATFORM="linux"' '-DNODE_WANT_INTERNALS=1' '-DV8_DEPRECATION_WARNINGS=1' '-DNODE_OPENSSL_SYSTEM_CERT_PATH=""' '-DHAVE_INSPECTOR=0' '-D__POSIX__' '-DHAVE_OPENSSL=1' '-DNODE_USE_V8_PLATFORM=1' '-DHTTP_PARSER_STRICT=0' '-D_LARGEFILE_SOURCE' '-D_FILE_OFFSET_BITS=64' '-D_POSIX_C_SOURCE=200112' '-DNGHTTP2_STATICLIB' -I../src -I../tools/msvs/genfiles -I/home/murad/o17x86/build_dir/target-i386_pentium4_musl-1.1.16/node-v8.10.0/out/Release/obj/gen -I../deps/v8/include -I../deps/http_parser -I../deps/cares/include -I../deps/uv/include -I../deps/nghttp2/lib/includes  -pthread -Wall -Wextra -Wno-unused-parameter -m32 -O3 -fno-omit-frame-pointer -fno-rtti -fno-exceptions -std=gnu++0x -MMD -MF /home/murad/o17x86/build_dir/target-i386_pentium4_musl-1.1.16/node-v8.10.0/out/Release/.deps//home/murad/o17x86/build_dir/target-i386_pentium4_musl-1.1.16/node-v8.10.0/out/Release/obj.target/node/src/node.o.d.raw  -Os -pipe -march=pentium4 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap /home/murad/o17x86/build_dir/target-i386_pentium4_musl-1.1.16/node-v8.10.0:node-v8.10.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/home/murad/o17x86/staging_dir/target-i386_pentium4_musl-1.1.16/usr/include -I/home/murad/o17x86/staging_dir/target-i386_pentium4_musl-1.1.16/include -I/home/murad/o17x86/staging_dir/toolchain-i386_pentium4_gcc-5.4.0_musl-1.1.16/usr/include -I/home/murad/o17x86/staging_dir/toolchain-i386_pentium4_gcc-5.4.0_musl-1.1.16/include/fortify -I/home/murad/o17x86/staging_dir/toolchain-i386_pentium4_gcc-5.4.0_musl-1.1.16/include  -c -o /home/murad/o17x86/build_dir/target-i386_pentium4_musl-1.1.16/node-v8.10.0/out/Release/obj.target/node/src/node.o ../src/node.cc
In file included from ../src/node.cc:37:0:
../src/node_crypto.h:39:25: fatal error: openssl/ssl.h: No such file or directory
compilation terminated.
`
I found the directory openssl here: staging_dir/host/include/openssl

And copied it here: staging_dir/toolchain-i386_pentium4_gcc-5.4.0_musl-1.1.16/include/

Then I've got:  fatal error: zlib.h: No such file or directory
compilation terminated.

The full output:
`LD_LIBRARY_PATH=/home/murad/o17x86/build_dir/target-i386_pentium4_musl-1.1.16/node-v8.10.0/out/Release/lib.host:/home/murad/o17x86/build_dir/target-i386_pentium4_musl-1.1.16/node-v8.10.0/out/Release/lib.target:$LD_LIBRARY_PATH; export LD_LIBRARY_PATH; cd ../deps/v8/src/inspector; mkdir -p /home/murad/o17x86/build_dir/target-i386_pentium4_musl-1.1.16/node-v8.10.0/out/Release/obj/gen/src/inspector/protocol /home/murad/o17x86/build_dir/target-i386_pentium4_musl-1.1.16/node-v8.10.0/out/Release/obj/gen/include/inspector; python ../../third_party/inspector_protocol/CodeGenerator.py --jinja_dir ../../third_party --output_base "/home/murad/o17x86/build_dir/target-i386_pentium4_musl-1.1.16/node-v8.10.0/out/Release/obj/gen/src/inspector" --config inspector_protocol_config.json
  ccache_cxx '-DNODE_ARCH="ia32"' '-DNODE_PLATFORM="linux"' '-DNODE_WANT_INTERNALS=1' '-DV8_DEPRECATION_WARNINGS=1' '-DNODE_OPENSSL_SYSTEM_CERT_PATH=""' '-DHAVE_INSPECTOR=0' '-D__POSIX__' '-DHAVE_OPENSSL=1' '-DNODE_USE_V8_PLATFORM=1' '-DHTTP_PARSER_STRICT=0' '-D_LARGEFILE_SOURCE' '-D_FILE_OFFSET_BITS=64' '-D_POSIX_C_SOURCE=200112' '-DNGHTTP2_STATICLIB' -I../src -I../tools/msvs/genfiles -I/home/murad/o17x86/build_dir/target-i386_pentium4_musl-1.1.16/node-v8.10.0/out/Release/obj/gen -I../deps/v8/include -I../deps/http_parser -I../deps/cares/include -I../deps/uv/include -I../deps/nghttp2/lib/includes  -pthread -Wall -Wextra -Wno-unused-parameter -m32 -O3 -fno-omit-frame-pointer -fno-rtti -fno-exceptions -std=gnu++0x -MMD -MF /home/murad/o17x86/build_dir/target-i386_pentium4_musl-1.1.16/node-v8.10.0/out/Release/.deps//home/murad/o17x86/build_dir/target-i386_pentium4_musl-1.1.16/node-v8.10.0/out/Release/obj.target/node/src/node.o.d.raw  -Os -pipe -march=pentium4 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap /home/murad/o17x86/build_dir/target-i386_pentium4_musl-1.1.16/node-v8.10.0:node-v8.10.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/home/murad/o17x86/staging_dir/target-i386_pentium4_musl-1.1.16/usr/include -I/home/murad/o17x86/staging_dir/target-i386_pentium4_musl-1.1.16/include -I/home/murad/o17x86/staging_dir/toolchain-i386_pentium4_gcc-5.4.0_musl-1.1.16/usr/include -I/home/murad/o17x86/staging_dir/toolchain-i386_pentium4_gcc-5.4.0_musl-1.1.16/include/fortify -I/home/murad/o17x86/staging_dir/toolchain-i386_pentium4_gcc-5.4.0_musl-1.1.16/include  -c -o /home/murad/o17x86/build_dir/target-i386_pentium4_musl-1.1.16/node-v8.10.0/out/Release/obj.target/node/src/node.o ../src/node.cc
../src/node.cc:73:18: fatal error: zlib.h: No such file or directory
compilation terminated.
`
To solve it I copied 
[... cut ...]
make[2]: *** [package/feeds/node/node/compile] Error 2
make[2]: Leaving directory '/home/murad/o17x86'
package/Makefile:101: recipe for target '/home/murad/o17x86/staging_dir/target-i386_pentium4_musl-1.1.16/stamp/.package_compile' failed
make[1]: *** [/home/murad/o17x86/staging_dir/target-i386_pentium4_musl-1.1.16/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/murad/o17x86'
/home/murad/o17x86/include/toplevel.mk:191: recipe for target 'world' failed
make: *** [world] Error 2
murad@ubnt16:~/o17x86$
`

nxhack, 2018-03-14:
I do not have time to verify now, so I will inform you the link to refer to.

https://openwrt.org/docs/guide-developer/obtain.firmware.sdk

nxhack, 2018-03-14:
```
mkdir Build_Path_Some_Where
cd Build_Path_Some_Where
wget http://downloads.lede-project.org/releases/17.01.4/targets/ar71xx/generic/lede-sdk-17.01.4-ar71xx-generic_gcc-5.4.0_musl-1.1.16.Linux-x86_64.tar.xz
tar xfJ lede-sdk-17.01.4-ar71xx-generic_gcc-5.4.0_musl-1.1.16.Linux-x86_64.tar.xz
cd lede-sdk-17.01.4-ar71xx-generic_gcc-5.4.0_musl-1.1.16.Linux-x86_64
echo "src-git node https://github.com/nxhack/openwrt-node-packages.git" >> feeds.conf.default
./scripts/feeds update -a
./scripts/feeds install -a
rm ./package/feeds/packages/node
rm ./package/feeds/packages/node-*
./scripts/feeds install -a -p node
wget https://raw.githubusercontent.com/nxhack/packages/master/lang/node/patches/004-node_crypto-remove-std.patch
cp 004-node_crypto-remove-std.patch ./feeds/node/node/patches/v8.x/ 
make package/feeds/node/node/{download,prepare,compile} V=s
```

nxhack, 2018-03-14:
When using SDK, I am investigating why this patch (004-node_crypto-remove-std.patch) is necessary...

nxhack, 2018-03-14:
https://github.com/nodejs/node/pull/19196

This issue will be fix. I hope.

muradurm, 2018-03-14:
Thank you very much.

I've successfully complied node and node-npm.

But I also need node-serialport.

During compilation I've errors about node.cc again:

`  g++ '-DNODE_ARCH="x64"' '-DNODE_PLATFORM="linux"' '-DNODE_WANT_INTERNALS=1' '-DV8_DEPRECATION_WARNINGS=1' '-DNODE_OPENSSL_SYSTEM_CERT_PATH=""' '-DHAVE_INSPECTOR=0' '-D__POSIX__' '-DHAVE_OPENSSL=1' '-DNODE_USE_V8_PLATFORM=1' '-DHTTP_PARSER_STRICT=0' '-D_LARGEFILE_SOURCE' '-D_FILE_OFFSET_BITS=64' '-D_POSIX_C_SOURCE=200112' '-DNGHTTP2_STATICLIB' -I../src -I../tools/msvs/genfiles -I/home/murad/o17x86/build_dir/target-i386_pentium4_musl-1.1.16/host/node-v8.10.0/out/Release/obj/gen -I../deps/openssl/openssl/include -I../deps/v8/include -I../deps/http_parser -I../deps/cares/include -I../deps/uv/include -I../deps/nghttp2/lib/includes  -pthread -Wall -Wextra -Wno-unused-parameter -m64 -O3 -fno-omit-frame-pointer -fno-rtti -fno-exceptions -std=gnu++0x -MMD -MF /home/murad/o17x86/build_dir/target-i386_pentium4_musl-1.1.16/host/node-v8.10.0/out/Release/.deps//home/murad/o17x86/build_dir/target-i386_pentium4_musl-1.1.16/host/node-v8.10.0/out/Release/obj.target/node/src/node_crypto.o.d.raw -I/home/murad/o17x86/staging_dir/host/include -I/home/murad/o17x86/staging_dir/host/usr/include -I/home/murad/o17x86/staging_dir/target-i386_pentium4_musl-1.1.16/host/include  -c -o /home/murad/o17x86/build_dir/target-i386_pentium4_musl-1.1.16/host/node-v8.10.0/out/Release/obj.target/node/src/node_crypto.o ../src/node_crypto.cc
../src/node_crypto.cc: In function 'void node::crypto::PBKDF2(const v8::FunctionCallbackInfo<v8::Value>&)':
../src/node_crypto.cc:5626:43: error: 'isnan' was not declared in this scope
   if (raw_keylen < 0.0 || isnan(raw_keylen) || isinf(raw_keylen) ||
                                           ^
../src/node_crypto.cc:5626:43: note: suggested alternative:
In file included from /usr/include/c++/5/random:38:0,
                 from /usr/include/c++/5/bits/stl_algo.h:66,
                 from /usr/include/c++/5/algorithm:62,
                 from ../src/node_crypto.cc:46:
/usr/include/c++/5/cmath:641:5: note:   'std::isnan'
     isnan(_Tp __x)
     ^
../src/node_crypto.cc:5626:64: error: 'isinf' was not declared in this scope
   if (raw_keylen < 0.0 || isnan(raw_keylen) || isinf(raw_keylen) ||
                                                                ^
../src/node_crypto.cc:5626:64: note: suggested alternative:
In file included from /usr/include/c++/5/random:38:0,
                 from /usr/include/c++/5/bits/stl_algo.h:66,
                 from /usr/include/c++/5/algorithm:62,
                 from ../src/node_crypto.cc:46:
/usr/include/c++/5/cmath:621:5: note:   'std::isinf'
     isinf(_Tp __x)
     ^
node.target.mk:173: recipe for target '/home/murad/o17x86/build_dir/target-i386_pentium4_musl-1.1.16/host/node-v8.10.0/out/Release/obj.target/node/src/node_crypto.o' failed
`

nxhack, 2018-03-16:
@muradurm 
addresse the issue of v8.10.0 node_crypto definitions of isnan and isinf.

Please check
```
mkdir Build_Path_Some_Where
cd Build_Path_Some_Where
wget http://downloads.lede-project.org/releases/17.01.4/targets/ar71xx/generic/lede-sdk-17.01.4-ar71xx-generic_gcc-5.4.0_musl-1.1.16.Linux-x86_64.tar.xz
tar xfJ lede-sdk-17.01.4-ar71xx-generic_gcc-5.4.0_musl-1.1.16.Linux-x86_64.tar.xz
cd lede-sdk-17.01.4-ar71xx-generic_gcc-5.4.0_musl-1.1.16.Linux-x86_64
echo "src-git node https://github.com/nxhack/openwrt-node-packages.git" >> feeds.conf.default
./scripts/feeds update -a
./scripts/feeds install -a
rm ./package/feeds/packages/node
rm ./package/feeds/packages/node-*
./scripts/feeds install -a -p node
make package/feeds/node/node/{download,prepare,compile} V=s
make package/feeds/node/node-serialport/{download,prepare,compile} V=s
```

muradurm, 2018-03-19:
Thank you very much!

I managed to compile both node and node-serialport. 

I installed it on my [TP-Link TL-WR1043ND v1.x](https://openwrt.org/toh/hwdata/tp-link/tp-link_tl-wr1043nd_v1)

After executing node I can see "Illegal instruction"

By default SOFT_FLOAT is set to yes. I've set it to no, rebuilt and installed again. But still have "Illegal instruction".

nxhack, 2018-03-19:
Please make sure your kernel builds with MIPS_FPU_EMULATOR kernel option.

command:
```
ls -l /sys/kernel/debug/mips/fpuemustats
```
If you do not have the fpuemustats directory, you need to rebuild the kernel with MIPS_FPU_EMULATOR option.

muradurm, 2018-03-23:
I built a f/w for my router with MIPS_FPU_EMULATOR kernel option and node.js started to work. I think this issue can be closed now.

Thank you very much for your help.

closed 2018-03-24 by nxhack