nxhack/openwrt-node-packages #260: npm install WARN

Raccooon, 2018-03-27:
When I use "npm install" to install plugins
It always WARN me that 
......No package.json
......No description
......No repository field

What should I fix?

nxhack, 2018-03-28:
Please tell me the details.

Raccooon, 2018-03-28:
Like this
root@OpenWrt:~# npm install homebridge-mi-aqara
npm info it worked if it ends with ok
npm info using npm@5.6.0
npm info using node@v8.10.0
npm http fetch GET 200 https://registry.npmjs.org/homebridge-mi-aqara 2795ms
npm http fetch GET 200 https://registry.npmjs.org/homebridge-mi-aqara/-/homebridge-mi-aqara-0.6.8.tgz 5070ms
npm info lifecycle homebridge-mi-aqara@0.6.8~preinstall: homebridge-mi-aqara@0.6.8
npm info linkStuff homebridge-mi-aqara@0.6.8
npm info lifecycle homebridge-mi-aqara@0.6.8~install: homebridge-mi-aqara@0.6.8
npm info lifecycle homebridge-mi-aqara@0.6.8~postinstall: homebridge-mi-aqara@0.6.8
npm WARN saveError ENOENT: no such file or directory, open '/root/package.json'
npm info lifecycle undefined~preshrinkwrap: undefined
npm info lifecycle undefined~shrinkwrap: undefined
npm notice created a lockfile as package-lock.json. You should commit this file.
npm info lifecycle undefined~postshrinkwrap: undefined
npm WARN enoent ENOENT: no such file or directory, open '/root/package.json'
npm WARN root No description
npm WARN root No repository field.
npm WARN root No README data
npm WARN root No license field.

Raccooon, 2018-03-28:
I have fix it using npm init -f
But I don't think that's a good idea

nxhack, 2018-03-28:
Is the necessary service running?
```
/bin/ps |fgrep homebridge
/bin/ps |fgrep avahi-daemon
```
Check with avahi-browse.
```
opkg update
opkg install avahi-utils
avahi-browse -at|fgrep _hap._tcp
```

Raccooon, 2018-03-28:
root@OpenWrt:/usr/lib# /bin/ps |fgrep homebridge
 2655 root     64292 S    homebridge
 2667 root      1200 R    fgrep homebridge
root@OpenWrt:/usr/lib# /bin/ps |fgrep avahi-daemon
 1049 nobody    1924 S    avahi-daemon: running [OpenWrt.local]
 2686 root      1200 S    fgrep avahi-daemon

root@OpenWrt:/usr/lib# avahi-browse -at|fgrep _hap._tcp
+  wlan0 IPv6 Homebridge                                    _hap._tcp            local
+  wlan1 IPv6 Homebridge                                    _hap._tcp            local
+ br-lan IPv6 Homebridge                                    _hap._tcp            local
+ br-lan IPv4 Homebridge                                    _hap._tcp            local
+   eth0 IPv6 Homebridge                                    _hap._tcp            local

nxhack, 2018-03-28:
```
    wlan0 IPv6 Homebridge _hap._tcp local
    wlan1 IPv6 Homebridge _hap._tcp local
    br-lan IPv6 Homebridge _hap._tcp local
    br-lan IPv4 Homebridge _hap._tcp local
    eth0 IPv6 Homebridge _hap._tcp local
```
Which network would you want to access? Is the network IPv4 or IPv6?

nxhack, 2018-03-28:
If your client (maybe iPhone home.app) is accessing from wlan 0 and your wlan 0 network device is IPv4 only, you can not access in this situation. In that way, please deal with your environment.

nxhack, 2018-03-28:
If you want only need IPv4, you can change  in /etc/avahi/avahi-daemon.conf

```
[server]
 |
use-ipv4=yes
use-ipv6=no
 |
```

Raccooon, 2018-03-28:
Thanks a lot
I will try it tonight

Raccooon, 2018-03-28:
I have done it
now
root@OpenWrt:~# avahi-browse -at|fgrep _hap._tcp
+ br-lan IPv4 Homebridge                                    _hap._tcp            local

Wlan0 also disappeared
and I still can't pair it

nxhack, 2018-03-28:
Unfortunately, it is beyond the scope of this repository and can not be supported. This depends on your hardware configuration and your own network configuration.

Why do not you go back to the basic and try network connection tests?

Raccooon, 2018-03-28:
Sorry, I'm a novice. 
Can you tell me how to do it?
What should I test?
Thank you very much.

Raccooon, 2018-03-28:
Now I need to make Wlan0 to ipv4 not ipv6?
I know, I will try it first

nxhack, 2018-03-28:
https://openwrt.org/docs/guide-user/start
https://www.google.co.jp/search?q=OpenWrt+network+configuration

p3x-robot, 2018-03-29:
when you want to install a package, I think you want global, like:
```bash
npm install -g homebridge-mi-aqara
```

Raccooon, 2018-03-29:
root@OpenWrt:~# avahi-browse -at|fgrep _hap._tcp
+  wlan0 IPv4 Homebridge                                    _hap._tcp            local
+  wlan1 IPv4 Homebridge                                    _hap._tcp            local
+ br-lan IPv4 Homebridge                                    _hap._tcp            local
+   eth1 IPv4 Homebridge                                    _hap._tcp            local
I have make wlan0 and wlan1 all work on ipv4
the port is opened
root@OpenWrt:~# netstat -tunlp
Active Internet connections (only servers)
Proto Recv-Q Send-Q Local Address           Foreign Address         State       PID/Program name    
tcp        0      0 0.0.0.0:80              0.0.0.0:*               LISTEN      886/uhttpd
tcp        0      0 192.168.1.170:53        0.0.0.0:*               LISTEN      1557/dnsmasq
tcp        0      0 127.0.0.1:53            0.0.0.0:*               LISTEN      1557/dnsmasq
tcp        0      0 222.195.67.42:53        0.0.0.0:*               LISTEN      1557/dnsmasq
tcp        0      0 192.168.1.1:53          0.0.0.0:*               LISTEN      1557/dnsmasq
tcp        0      0 0.0.0.0:22              0.0.0.0:*               LISTEN      851/dropbear
tcp        0      0 :::80                   :::*                    LISTEN      886/uhttpd
tcp        0      0 :::51826                :::*                    LISTEN      6473/homebridge
tcp        0      0 ::1:53                  :::*                    LISTEN      1557/dnsmasq
tcp        0      0 fe80::c404:15ff:fe90:38d5:53 :::*                    LISTEN      1557/dnsmasq
tcp        0      0 fe80::c604:15ff:fe90:38d6:53 :::*                    LISTEN      1557/dnsmasq
tcp        0      0 fe80::c404:15ff:fe90:38d5:53 :::*                    LISTEN      1557/dnsmasq
tcp        0      0 :::22                   :::*                    LISTEN      851/dropbear
tcp        0      0 :::38841                :::*                    LISTEN      6473/homebridge
udp        0      0 192.168.1.170:53        0.0.0.0:*                           1557/dnsmasq
udp        0      0 127.0.0.1:53            0.0.0.0:*                           1557/dnsmasq
udp        0      0 222.195.67.42:53        0.0.0.0:*                           1557/dnsmasq
udp        0      0 192.168.1.1:53          0.0.0.0:*                           1557/dnsmasq
udp        0      0 0.0.0.0:67              0.0.0.0:*                           1557/dnsmasq
udp        0      0 0.0.0.0:9898            0.0.0.0:*                           6473/homebridge
udp        0      0 0.0.0.0:5353            0.0.0.0:*                           903/avahi-daemon: r
udp        0      0 ::1:53                  :::*                                1557/dnsmasq
udp        0      0 fe80::c404:15ff:fe90:38d5:53 :::*                                1557/dnsmasq
udp        0      0 fe80::c604:15ff:fe90:38d6:53 :::*                                1557/dnsmasq
udp        0      0 fe80::c404:15ff:fe90:38d5:53 :::*                                1557/dnsmasq

so sad

nxhack, 2018-03-30:
Is the client an iPhone and home.app?

Raccooon, 2018-03-30:
Right.
iOS 11.3 with home.app

nxhack, 2018-03-30:
Ping the iPhone from the OpenWrt host and check continuity.

https://www.iphonefaq.org/archives/971437

Raccooon, 2018-03-30:
You mean like this?
My iPhone's ipv4 address is 192.168.1.160

```
root@OpenWrt:~# ping 192.168.1.160
PING 192.168.1.160 (192.168.1.160): 56 data bytes
64 bytes from 192.168.1.160: seq=0 ttl=64 time=2.848 ms
64 bytes from 192.168.1.160: seq=1 ttl=64 time=98.557 ms
64 bytes from 192.168.1.160: seq=2 ttl=64 time=20.040 ms
64 bytes from 192.168.1.160: seq=3 ttl=64 time=43.039 ms
64 bytes from 192.168.1.160: seq=4 ttl=64 time=67.544 ms
^C
--- 192.168.1.160 ping statistics ---
5 packets transmitted, 5 packets received, 0% packet loss
round-trip min/avg/max = 2.848/46.405/98.557 ms
```

nxhack, 2018-03-30:
OK. There is no network connection problem. 
Next, check homebridge settings.

on openwrt host:
```
logread|fgrep homebridge
```
check homebridge log.
(Please do not copy & paste the setting information you want to keep secret.)

Raccooon, 2018-03-30:
quite strange bug
why homebridge don't go to ~/.homebrige/ to find persist?
```
Fri Mar 30 02:09:55 2018 daemon.err homebridge[1050]: npm info it worked if it ends with ok
Fri Mar 30 02:09:55 2018 daemon.err homebridge[1050]: npm info using npm@5.6.0
Fri Mar 30 02:09:55 2018 daemon.err homebridge[1050]: npm info using node@v8.10.0
Fri Mar 30 02:09:55 2018 daemon.err homebridge[1050]:
Fri Mar 30 02:09:55 2018 daemon.err homebridge[1050]: ┌──────────────────────────────────────────────┐
Fri Mar 30 02:09:55 2018 daemon.err homebridge[1050]: │           npm update check failed            │
Fri Mar 30 02:09:55 2018 daemon.err homebridge[1050]: │     Try running with sudo or get access      │
Fri Mar 30 02:09:55 2018 daemon.err homebridge[1050]: │     to the local update config store via     │
Fri Mar 30 02:09:55 2018 daemon.err homebridge[1050]: │ sudo chown -R $USER:$(id -gn $USER) /.config │
Fri Mar 30 02:09:55 2018 daemon.err homebridge[1050]: └──────────────────────────────────────────────┘
Fri Mar 30 02:09:55 2018 daemon.err homebridge[1050]: npm info ok
Fri Mar 30 02:09:56 2018 daemon.err homebridge[1050]: /usr/lib/node/homebridge/node_modules/mkdirp/index.js:90
Fri Mar 30 02:09:56 2018 daemon.err homebridge[1050]:                     throw err0;
Fri Mar 30 02:09:56 2018 daemon.err homebridge[1050]:                     ^
Fri Mar 30 02:09:56 2018 daemon.err homebridge[1050]:
Fri Mar 30 02:09:56 2018 daemon.err homebridge[1050]: Error: EACCES: permission denied, mkdir '/usr/share/homebridge/persist'
Fri Mar 30 02:09:56 2018 daemon.err homebridge[1050]:     at Object.fs.mkdirSync (fs.js:885:18)
Fri Mar 30 02:09:56 2018 daemon.err homebridge[1050]:     at Function.sync (/usr/lib/node/homebridge/node_modules/mkdirp/index.js:71:13)
Fri Mar 30 02:09:56 2018 daemon.err homebridge[1050]:     at LocalStorage.parseDirSync (/usr/lib/node/homebridge/node_modules/hap-nodejs/node_modules/node-persist/src/local-storage.js:642:20)
Fri Mar 30 02:09:56 2018 daemon.err homebridge[1050]:     at LocalStorage.parseDataDirSync (/usr/lib/node/homebridge/node_modules/hap-nodejs/node_modules/node-persist/src/local-storage.js:572:21)
Fri Mar 30 02:09:56 2018 daemon.err homebridge[1050]:     at LocalStorage.initSync (/usr/lib/node/homebridge/node_modules/hap-nodejs/node_modules/node-persist/src/local-storage.js:145:14)
Fri Mar 30 02:09:56 2018 daemon.err homebridge[1050]:     at Object.nodePersist.initSync (/usr/lib/node/homebridge/node_modules/hap-nodejs/node_modules/node-persist/src/node-persist.js:41:29)
Fri Mar 30 02:09:56 2018 daemon.err homebridge[1050]:     at Object.init (/usr/lib/node/homebridge/node_modules/hap-nodejs/index.js:31:13)
Fri Mar 30 02:09:56 2018 daemon.err homebridge[1050]:     at module.exports (/usr/lib/node/homebridge/lib/cli.js:24:7)
Fri Mar 30 02:09:56 2018 daemon.err homebridge[1050]:     at Object.<anonymous> (/usr/lib/node/homebridge/bin/homebridge:17:22)
Fri Mar 30 02:09:56 2018 daemon.err homebridge[1050]:     at Module._compile (module.js:652:30)
```

Raccooon, 2018-03-30:
here is my homebridge service file
```
#!/bin/sh /etc/rc.common

START=98
USE_PROCD=1

start_service() {
        [ -d /usr/share/homebridge ] || {
                mkdir -m 0755 -p /usr/share/homebridge
                chmod 0700 /usr/share/homebridge
                chown homebridge:homebridge /usr/share/homebridge
        }
        procd_open_instance
        procd_set_param command /usr/bin/homebridge -U /usr/share/homebridge
        procd_set_param user homebridge
        procd_set_param stdout 1
        procd_set_param stderr 1
        procd_close_instance
}
```

nxhack, 2018-03-30:
homebridge services is running with the normal user privilege of "homebridge".

Check the permissions of related files and directories and correct them.

nxhack, 2018-03-30:
Then put config.json in your homebridge configuration file under "/usr/share/homebridge" directory.

Raccooon, 2018-03-30:
```
root@OpenWrt:/usr/share# ls -al
drwxrwxrwx    2 root     root             0 Mar 30 04:06 homebridge
root@OpenWrt:/usr/share# chmod 777 /usr/share/homebridge
root@OpenWrt:/usr/share# chown homebridge:homebridge /usr/share/homebridge
```
I do it like this , but still
```
Error: EACCES: permission denied, mkdir '/usr/share/homebridge/persist'
```
which permissons should I give?

nxhack, 2018-03-30:
```
chown -R homebridge:homebridge /usr/share/homebridge
```

Raccooon, 2018-03-30:
yes, I have done this
but
root@OpenWrt:~# chown -R homebridge:homebridge /usr/share/homebridge
root@OpenWrt:/usr/share# ls -l
drwxrwxrwx    2 root     root             0 Mar 30 04:06 homebridge

nxhack, 2018-03-30:
?
check /etc/passwd & /etc/group

Raccooon, 2018-03-30:
I am working on this
user and group all exist
/etc/passwd
root:x:0:0:root:/root:/bin/ash
daemon:*:1:1:daemon:/var:/bin/false
ftp:*:55:55:ftp:/home/ftp:/bin/false
network:*:101:101:network:/var:/bin/false
nobody:*:65534:65534:nobody:/var:/bin/false
dnsmasq:x:453:453:dnsmasq:/var/run/dnsmasq:/bin/false
homebridge:x:65536:65536:homebridge:/var/run/homebridge:/bin/false

/etc/group
root:x:0:
daemon:x:1:
adm:x:4:
tty:x:5:
mail:x:8:
audio:x:29:
www-data:x:33:
ftp:x:55:
users:x:100:
network:x:101:
nogroup:x:65534:
dnsmasq:x:453:dnsmasq
homebridge:x:65536:homebridge

nxhack, 2018-03-30:
?
stop homebridge service then try change owner & group.

If fail, let's test whether owner and group can be changed correctly.
```
cd /tmp
mkdir HB
chown homebridge:homebridge HB
ls -ald HB
```

Raccooon, 2018-03-30:
it works
```
root@OpenWrt:/# cd tmp
root@OpenWrt:/tmp# mkdir HB
root@OpenWrt:/tmp# chown homebridge:homebridge HB
root@OpenWrt:/tmp# ls -ald HB
drwxr-xr-x    2 homebrid homebrid        40 Mar 30 05:03 HB
```
quite strange
```
root@OpenWrt:/usr/share# /etc/init.d/homebridge disable
root@OpenWrt:/usr/share# /etc/init.d/homebridge stop
root@OpenWrt:/usr/share# chown homebridge:homebridge homebridge
root@OpenWrt:/usr/share# ls -al
drwx------    2 root     root             0 Mar 30 04:06 homebridge
```

nxhack, 2018-03-30:
?????
try

backup your config.json, then
```
cd /usr/share
rm -rf homebridge
mkdir homebridge
chown homebridge:homebridge homebridge
```

Raccooon, 2018-03-30:
the same idea
```
root@OpenWrt:/usr/share# rm -rf homebridge/
root@OpenWrt:/usr/share# ls
acl.d      dbus-1     dnsmasq    fw3        libiwinfo  libubox    rpcd       udhcpc
root@OpenWrt:/usr/share# mkdir homebridge
root@OpenWrt:/usr/share# ls -al
drwxr-xr-x    2 root     root             0 Mar 30 05:08 homebridge
root@OpenWrt:/usr/share# chown homebridge:homebridge homebridge
root@OpenWrt:/usr/share# ls -al
drwxr-xr-x    2 root     root             0 Mar 30 05:08 homebridge
```

nxhack, 2018-03-30:
???????????
I will try to check google :) if there are similar issue. It will take time.

Raccooon, 2018-03-30:
that's quite strange problem
I will try to give the homebridge root

nxhack, 2018-03-30:
Is there a file system error log?
```
logread
```
(may not be easy...)

Raccooon, 2018-03-30:
I did it
no logs in logread

nxhack, 2018-03-30:
xattr ???

Raccooon, 2018-03-30:
I mean
no file system error in logs

Raccooon, 2018-03-30:
the rom I flash into the router(wndr3800)

ar71xx
select node.js-homebridge
select python
select luci
no other things

nxhack, 2018-03-30:
Since it is an unusual issue, for now, please do workaround with a way to run it as root. You already know the changes.

nxhack, 2018-03-30:
It is a fact that this package works without problems for me and other users.

Raccooon, 2018-03-30:
That’s what gives me hope
I will try it

nxhack, 2018-03-30:
Please give me another information. What is your OpenWrt version?

Raccooon, 2018-03-30:
OpenWrt SNAPSHOT r6550-83e1fce / LuCI Master (git-18.085.65030-648fd91)

nxhack, 2018-03-30:
As expected it was Head version so we will investigate with the bug line of kernel and system.
It may not be solved, but please do not feel bad.

Raccooon, 2018-03-30:
The source is from
https://git.openwrt.org/openwrt/openwrt.git
I thought it is trunk
Right？

nxhack, 2018-03-30:
> https://git.openwrt.org/openwrt/openwrt.git
> I thought it is trunk

yes

nxhack, 2018-03-30:
@Raccooon 
Would you please try it once?
Try homeidge's uid and gid to numbers that are not 65536, as 200.
65536 is suspicious.

Raccooon, 2018-03-30:
OK, I will try it later

nxhack, 2018-03-30:
Hmm... it is a 32-bit UID, it's okay. There was no problem with lede-17.01 kernel 4.4.108. Hmm...

Raccooon, 2018-03-30:
I have done it, make uid and gid 200, and everything becomes right
```
root@OpenWrt:/usr/share# ls -al
drwxr-xr-x    4 homebrid homebrid         0 Mar 30 08:32 homebridge
```
I try to pair, but failed.

now, I have done a full check about it
```
root@OpenWrt:/usr/bin# /bin/ps |fgrep homebridge
 2433 homebrid 56140 R    node /usr/bin/homebridge -U /usr/share/homebridge
 2445 root      1200 S    fgrep homebridge
root@OpenWrt:/usr/bin# /bin/ps |fgrep avahi-daemon
 928 nobody    1892 S    avahi-daemon: ruroot@OpenWrt:/usr/share#nning [OpenWrt.local]
 2469 root      1200 S    fgrep avahi-daemon
root@OpenWrt:/usr/bin# avahi-browse -at|fgrep _hap._tcp
+  wlan0 IPv4 Homebridge                                    _hap._tcp            local
+  wlan1 IPv4 Homebridge                                    _hap._tcp            local
+ eth0.1 IPv4 Homebridge                                    _hap._tcp            local
+ br-lan IPv4 Homebridge                                    _hap._tcp            local
+   eth1 IPv4 Homebridge                                    _hap._tcp            local
+   eth0 IPv4 Homebridge                                    _hap._tcp            local
root@OpenWrt:/usr/share# logread|fgrep homebridge
Fri Mar 30 08:49:23 2018 daemon.err homebridge[2433]: npm info it worked if it ends with ok
Fri Mar 30 08:49:23 2018 daemon.err homebridge[2433]: npm info using npm@5.6.0
Fri Mar 30 08:49:23 2018 daemon.err homebridge[2433]: npm info using node@v8.10.0
Fri Mar 30 08:49:23 2018 daemon.err homebridge[2433]: npm info ok
Fri Mar 30 08:49:24 2018 daemon.info homebridge[2433]: [Fri Mar 30 2018 08:49:24 GMT+0000 (UTC)] Loaded plugin: homebridge-mi-aqara
Fri Mar 30 08:49:24 2018 daemon.info homebridge[2433]: [Fri Mar 30 2018 08:49:24 GMT+0000 (UTC)] Registering platform 'homebridge-mi-aqara.MiAqaraPlatform'
Fri Mar 30 08:49:24 2018 daemon.info homebridge[2433]: [Fri Mar 30 2018 08:49:24 GMT+0000 (UTC)] ---
Fri Mar 30 08:49:24 2018 daemon.info homebridge[2433]: [Fri Mar 30 2018 08:49:24 GMT+0000 (UTC)] Loaded config.json with 0 accessories and 1 platforms.
Fri Mar 30 08:49:24 2018 daemon.info homebridge[2433]: [Fri Mar 30 2018 08:49:24 GMT+0000 (UTC)] ---
Fri Mar 30 08:49:25 2018 daemon.info homebridge[2433]: [Fri Mar 30 2018 08:49:25 GMT+0000 (UTC)] Loading 1 platforms...
Fri Mar 30 08:49:25 2018 daemon.info homebridge[2433]: [Fri Mar 30 2018 08:49:25 GMT+0000 (UTC)] [MiAqaraPlatform] Initializing MiAqaraPlatform platform...
Fri Mar 30 08:49:25 2018 daemon.info homebridge[2433]: [Fri Mar 30 2018 08:49:25 GMT+0000 (UTC)] [MiAqaraPlatform] [INFO]**************************************************************
Fri Mar 30 08:49:25 2018 daemon.info homebridge[2433]: [Fri Mar 30 2018 08:49:25 GMT+0000 (UTC)] [MiAqaraPlatform] [INFO]           MiAqaraPlatform v0.6.8 By YinHang
Fri Mar 30 08:49:25 2018 daemon.info homebridge[2433]: [Fri Mar 30 2018 08:49:25 GMT+0000 (UTC)] [MiAqaraPlatform] [INFO]  GitHub: https://github.com/YinHangCode/homebridge-mi-aqara
Fri Mar 30 08:49:25 2018 daemon.info homebridge[2433]: [Fri Mar 30 2018 08:49:25 GMT+0000 (UTC)] [MiAqaraPlatform] [INFO]                                         QQ Group: 107927710
Fri Mar 30 08:49:25 2018 daemon.info homebridge[2433]: [Fri Mar 30 2018 08:49:25 GMT+0000 (UTC)] [MiAqaraPlatform] [INFO]**************************************************************
Fri Mar 30 08:49:25 2018 daemon.info homebridge[2433]: [Fri Mar 30 2018 08:49:25 GMT+0000 (UTC)] [MiAqaraPlatform] [INFO]start success...
Fri Mar 30 08:49:25 2018 daemon.info homebridge[2433]: [Fri Mar 30 2018 08:49:25 GMT+0000 (UTC)] [MiAqaraPlatform] [INFO]config gateways: -------------
Fri Mar 30 08:49:25 2018 daemon.info homebridge[2433]: [Fri Mar 30 2018 08:49:25 GMT+0000 (UTC)] [MiAqaraPlatform] [INFO]binding to the default interface
Fri Mar 30 08:49:25 2018 daemon.info homebridge[2433]: Setup Payload:
Fri Mar 30 08:49:26 2018 daemon.info homebridge[2433]: [Fri Mar 30 2018 08:49:26 GMT+0000 (UTC)] [MiAqaraPlatform] [INFO]server is listening on port 9898.
Fri Mar 30 08:49:26 2018 daemon.info homebridge[2433]: [Fri Mar 30 2018 08:49:26 GMT+0000 (UTC)] Homebridge is running on port 51826.
root@OpenWrt:/usr/bin# netstat -tunlp
Active Internet connections (only servers)
Proto Recv-Q Send-Q Local Address           Foreign Address         State       PID/Program name    
tcp        0      0 0.0.0.0:80              0.0.0.0:*               LISTEN      911/uhttpd
tcp        0      0 192.168.1.140:53        0.0.0.0:*               LISTEN      1535/dnsmasq
tcp        0      0 192.168.1.130:53        0.0.0.0:*               LISTEN      1535/dnsmasq
tcp        0      0 192.168.1.120:53        0.0.0.0:*               LISTEN      1535/dnsmasq
tcp        0      0 192.168.1.110:53        0.0.0.0:*               LISTEN      1535/dnsmasq
tcp        0      0 127.0.0.1:53            0.0.0.0:*               LISTEN      1535/dnsmasq
tcp        0      0 222.195.67.42:53        0.0.0.0:*               LISTEN      1535/dnsmasq
tcp        0      0 192.168.1.1:53          0.0.0.0:*               LISTEN      1535/dnsmasq
tcp        0      0 0.0.0.0:22              0.0.0.0:*               LISTEN      880/dropbear
tcp        0      0 :::80                   :::*                    LISTEN      911/uhttpd
tcp        0      0 :::51826                :::*                    LISTEN      2433/homebridge
tcp        0      0 2001:da8:d800:501:c604:15ff:fe90:38d6:53 :::*                    LISTEN      1535/dnsmasq
[... cut ...]
udp        0      0 ::1:53                  :::*                                1535/dnsmasq
udp        0      0 fe80::c404:15ff:fe90:38d5:53 :::*                                1535/dnsmasq
udp        0      0 fe80::c604:15ff:fe90:38d6:53 :::*                                1535/dnsmasq
udp        0      0 fe80::c404:15ff:fe90:38d5:53 :::*                                1535/dnsmasq
udp        0      0 fe80::c604:15ff:fe90:38d7:53 :::*                                1535/dnsmasq
udp        0      0 fe80::c604:15ff:fe90:38d5:53 :::*                                1535/dnsmasq
```

nxhack, 2018-03-30:
```
tcp        0      0 :::51826                :::*                    LISTEN      2433/homebridge
```
OK

> Head to your iOS device and launch the Home app.
> Tap the Add button in the top right corner of the screen (looks like a plus sign).
> Tap Add Accessory.
> Tap on the Homebridge accessory (mine's just called Homebridge).
> You will see an alert that says, "This accessory is not certified and may not work reliably with HomeKit." tap Add Anyway.
> Tap Enter Code Manually at the bottom of the screen.
> Enter the HomeKit code for your Homebridge accessory. You can find it in the Terminal window underneath the text that reads, "Scan this code with your HomeKit App on your iOS device to pair with Homebridge."

https://www.imore.com/how-connect-non-homekit-devices-homekit-using-homebridge

Raccooon, 2018-03-30:
yes
I can find a near by homebridge
when I scan the picture or enter the code
It will start pairing and but can't finish
After minutes,
Couldn't add homebridge
home couldn't connet to this accessory

nxhack, 2018-03-30:
Please refer to here and try some.
https://github.com/nfarina/homebridge/issues

Raccooon, 2018-03-30:
Hey, nxhack
  I have tried some ways, and find this problem
  can you give some adivce?
  ```
EventedHTTPServer [::ffff:192.168.1.160] New connection from client +16m
  EventedHTTPServer [::ffff:192.168.1.160] HTTP server listening on port 34455 +6ms
  EventedHTTPServer [::ffff:192.168.1.160] HTTP request: /pair-setup +30ms
  HAPServer [A2:35:ED:EF:AC:26] HAP Request: POST /pair-setup +3ms
  HAPServer [A2:35:ED:EF:AC:26] Pair step 1/5 +9ms
  EventedHTTPServer [::ffff:192.168.1.160] HTTP Response is finished +3m
  EventedHTTPServer [::ffff:192.168.1.160] Client connection error: This socket has been ended by the other party +16ms
  EventedHTTPServer [::ffff:192.168.1.160] Client connection closed +4ms
  EventedHTTPServer [::ffff:192.168.1.160] HTTP connection was closed +7ms
  EventedHTTPServer [::ffff:192.168.1.160] HTTP server was closed +4ms

```

Raccooon, 2018-03-30:
`HAPServer [A2:35:ED:EF:AC:26] Pair step 1/5 +72ms`
It hangs on this line
After serveral minutes,
`EventedHTTPServer [::ffff:192.168.1.160] HTTP Response is finished +3m`

nxhack, 2018-03-30:
Great!
I think that it is necessary to analyze step by step with tcpdump etc. as you are doing now. Please do your best!

Raccooon, 2018-03-31:
Thank you for your encouragement.
May I ask you a question, is there any firewall settings you I have done for homebridge?

[14 more replies]

closed 2018-04-16 by nxhack