nxhack/openwrt-node-packages #274: libmodus install

p3x-robot, 2018-04-12:
Ciao NXHACK!

How are you?
I spotted an error in LEDE 17.01.4 (or maybe it is only working on OpenWrt?)
```text
Making all in doc
a2x --doctype manpage --format manpage -alibmodbus_version=3.1.4 modbus_close.txt
a2x: ERROR: "xmllint" --nonet --noout --valid "/build/source/build_dir/target-arm_cortex-a53+neon-vfpv4_musl-1.1.16_eabi/node-modbus-0.0.16/libmodbus/doc/modbus_close.xml" returned non-zero exit status 4
Makefile:645: recipe for target 'modbus_close.3' failed
make[6]: *** [modbus_close.3] Error 1
Makefile:498: recipe for target 'all-recursive' failed
make[5]: *** [all-recursive] Error 1
Makefile:384: recipe for target 'all' failed
make[4]: *** [all] Error 2
make[4]: Leaving directory '/build/source/build_dir/target-arm_cortex-a53+neon-vfpv4_musl-1.1.16_eabi/node-modbus-0.0.16/libmodbus'
npm info lifecycle modbus@0.0.16~install: Failed to exec install script
npm ERR! code ELIFECYCLE
npm ERR! errno 2
npm ERR! modbus@0.0.16 install: `./make_libmodbus.sh && node-gyp configure && node-gyp build`
npm ERR! Exit status 2
npm ERR! 
npm ERR! Failed at the modbus@0.0.16 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
```

nxhack, 2018-04-13:
I do not know what is going one. Please tell me about the situation.

p3x-robot, 2018-04-16:
i do not use so i guess for now i leave it.

closed 2018-04-16 by p3x-robot