nxhack/openwrt-node-packages #318: net-ping module

bastup, 2018-06-08:
Hello NXhack!

I am currently trying to get the net-ping module from npm to work on my archer c7.

The net-ping module is dependent on "raw-socket" packet, which requires some compiling at installation.

I tried to run NPM install net-ping and have python-light installed but there is not enough space on the device:
gyp ERR! stack Error: ENOSPC: no space left on device

Is there some way to get around this or is it possible to add "NaN", "raw-socket" and "net-ping" to the choice of addable packets?

Thanks again for your great work!

nxhack, 2018-06-11:
Hi @bastup 

https://github.com/nxhack/openwrt-node-packages/tree/master/node-net-ping

Please test.

bastup, 2018-06-11:
First tests are perfect,

//Testing ping,
> pingNow();
undefined
> 8.8.8.8: Alive(time: 18)
> pingNow();
undefined
> 8.8.8.8: Alive(time: 18)
> pingNow();
undefined
> 8.8.8.8: Alive(time: 18)

//Testing Traceroute:
8.8.4.4: 5.150.204.17 (ttl=1 ms=7)
8.8.4.4: 5.150.204.14 (ttl=2 ms=2)
8.8.4.4: 85.24.151.202 (ttl=3 ms=12)
8.8.4.4: 46.59.113.72 (ttl=4 ms=12)
8.8.4.4: 46.59.112.95 (ttl=5 ms=13)
8.8.4.4: 46.59.117.31 (ttl=6 ms=13)
8.8.4.4: 108.170.253.161 (ttl=7 ms=17)
8.8.4.4: 209.85.240.81 (ttl=8 ms=16)
8.8.4.4: 8.8.4.4 (ttl=9 ms=15)
8.8.4.4: Done

Great work!!

Is it possible to get an explanation how you solved this problem?

Close at will

nxhack, 2018-06-12:
Did you build in a cross environment or native environment?
My Makefile is for the OpenWrt SDK on cross environment.

>  gyp ERR! stack Error: ENOSPC: no space left on device

If you are doing a native build on the target machine, you may run out of disk space.

bastup, 2018-06-12:
I tried to build in a native env. (on the mips-device).
I read some on cross-platform but I didn't find any good way to do it with npm.

I'll read more on cross-platform compilation.

closed 2018-06-12 by bastup