nxhack/openwrt-node-packages #329: (weird error) root@digi:~# node -v v10.5.0 root@digi:~# npm -v node[19702]: ../src/node_platform.cc:262:std::shared_ptr<node::PerIsolatePlatformData> node::NodePlatform::ForIsolate(v8::Isolate*): Assertion `data' failed. Aborted root@digi:~# npm node[19710]: ../src/node_platform.cc:262:std::shared_ptr<node::PerIsolatePlatformData> node::NodePlatform::ForIsolate(v8::Isolate*): Assertion `data' failed. Aborted

p3x-robot, 2018-06-27:
Ciao!

Do you have an idea what this error is???

PatrikX3

p3x-robot, 2018-06-27:
Actaully Node is not wokring on 17.01.4 at all. `nodejs` is not working at all either.

p3x-robot, 2018-06-27:
# Linksys 1900ACS / 3200ACM
```
root@digi:~# node -v
v10.5.0
root@digi:~# npm -v
node[19702]: ../src/node_platform.cc:262:std::shared_ptr<node::PerIsolatePlatformData> node::NodePlatform::ForIsolate(v8::Isolate*): Assertion `data' failed.
Aborted
root@digi:~# npm
node[19710]: ../src/node_platform.cc:262:std::shared_ptr<node::PerIsolatePlatformData> node::NodePlatform::ForIsolate(v8::Isolate*): Assertion `data' failed.
Aborted
root@digi:~# node
node[19809]: ../src/node_platform.cc:262:std::shared_ptr<node::PerIsolatePlatformData> node::NodePlatform::ForIsolate(v8::Isolate*): Assertion `data' failed.
Aborted
root@digi:~# node -v
v10.5.0
root@digi:~#
```

# D-LINK DIR860L

```
Downloading http://cdn.corifeus.com/lede/17.01.4/packages/mipsel_24kc/node/node-yarn_1.7.0-1_mipsel_24kc.ipk
Configuring node.
Configuring node-npm.
Configuring node-yarn.
root@d-link:~# node -v
v10.5.0
root@d-link:~# npm -v
node[26862]: ../src/node_platform.cc:262:std::shared_ptr<node::PerIsolatePlatformData> node::NodePlatform::ForIsolate(v8::Isolate*): Assertion `data' failed.
Aborted
root@d-link:~# node -v
v10.5.0
root@d-link:~# yarn
node[26978]: ../src/node_platform.cc:262:std::shared_ptr<node::PerIsolatePlatformData> node::NodePlatform::ForIsolate(v8::Isolate*): Assertion `data' failed.
Aborted
root@d-link:~#
```

p3x-robot, 2018-06-27:
here is the error
https://github.com/nodejs/node/blob/v10.5.0/src/node_platform.cc#L262
Some asserting.

https://github.com/nodejs/node/issues/21562

https://github.com/nodejs/node/issues/21562#issuecomment-400654840

p3x-robot, 2018-06-27:
create a core dump as well:
https://github.com/nodejs/node/issues/21562#issuecomment-400686867

https://cdn.corifeus.com/archive/node.1530108405.24874.6.core

nxhack, 2018-06-28:
Hi @p3x-robot 

I do not know if this information is useful. This is the place I can do for now.
```
$ ~/openwrt/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-7.3.0_musl_eabi/bin/arm-openwrt-linux-gdb ./out/Release/node ~/node.1530108405.24874.6.core 
GNU gdb (GDB) 8.0.1
Copyright (C) 2017 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=x86_64-pc-linux-gnu --target=arm-openwrt-linux-muslgnueabi".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./out/Release/node...done.

warning: exec file is newer than core file.
[New LWP 24874]
[New LWP 24875]
[New LWP 24877]
[New LWP 24876]
[New LWP 24878]
Core was generated by `node'.
Program terminated with signal SIGABRT, Aborted.
#0  0xb6fbed5c in ?? ()
[Current thread is 1 (LWP 24874)]
(gdb) list
1	#include <features.h>
2	
3	#define START "_start"
4	
5	#include "crt_arch.h"
6	
7	int main();
8	void _init() __attribute__((weak));
9	void _fini() __attribute__((weak));
10	_Noreturn int __libc_start_main(int (*)(), int, char **,
(gdb) t 2
[Switching to thread 2 (LWP 24875)]
#0  0x007c1c0c in v8::internal::ICStats::End() ()
(gdb) t 3
[Switching to thread 3 (LWP 24877)]
#0  0xb6fca2c8 in ?? ()
(gdb) t 4
[Switching to thread 4 (LWP 24876)]
#0  0xb6fca2c8 in ?? ()
(gdb) t 5
[Switching to thread 5 (LWP 24878)]
#0  0xb6fca2c8 in ?? ()
(gdb) 
```

p3x-robot, 2018-06-28:
For you it works `10.5` ???

p3x-robot, 2018-06-28:
ok i talk to nodejs github. i see how to debug now with arm, thanks @nxhack !!! 🥇

p3x-robot, 2018-06-28:
I enabled, I just buld the debug flag nodejs and then the symbols will be in gdb right? same node executable i guess.

p3x-robot, 2018-06-28:
The problem I cannot create a debug build I get this error:
```
/build/source/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/node-v10.5.0/out/Debug/obj.target/v8_base/deps/v8/src/heap/factory.o:/build/source/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/node-v10.5.0/out/../deps/v8/src/heap/factory.cc:1070: more undefined references to `void v8::internal::String::WriteToFlat<unsigned char>(v8::internal::String*, unsigned char*, int, int)' follow
collect2: error: ld returned 1 exit status
node.target.mk:156: recipe for target '/build/source/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/node-v10.5.0/out/Debug/node' failed
make[4]: *** [/build/source/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/node-v10.5.0/out/Debug/node] Error 1
rm ae87dc835bea2d326910f4833d37b21735164336.intermediate
Makefile:91: recipe for target 'node_g' failed
make[3]: *** [node_g] Error 2
make[3]: Leaving directory '/build/source/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/node-v10.5.0'
Makefile:203: recipe for target '/build/source/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/node-v10.5.0/.built' failed
make[2]: *** [/build/source/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/node-v10.5.0/.built] Error 2
make[2]: Leaving directory '/build/source/feeds/node/node'
package/Makefile:105: recipe for target 'package/feeds/node/node/compile' failed
make[1]: *** [package/feeds/node/node/compile] Error 2
make[1]: Leaving directory '/build/source'
/build/source/include/toplevel.mk:198: recipe for target 'package/feeds/node/node/compile' failed
make: *** [package/feeds/node/node/compile] Error 2
docker@4d82cb762091:/build/source$
```

What it could be? Some patch missing?

nxhack, 2018-06-28:
It is necessary to use gdb of the target machine on the target machine.

p3x-robot, 2018-06-28:
but i cannot build a debug build

```
docker@4d82cb762091:/build/source$ cat .config | grep NODE
# CONFIG_BUSYBOX_DEFAULT_FEATURE_TOUCH_NODEREF is not set
# CONFIG_NODEJS_6 is not set
# CONFIG_NODEJS_8 is not set
# CONFIG_NODEJS_9 is not set
CONFIG_NODEJS_10=y
CONFIG_NODEJS_ICU_NONE=y
# CONFIG_NODEJS_ICU_SMALL is not set
# CONFIG_NODEJS_ICU_SYSTEM is not set
CONFIG_NODEJS_DEBUG=y
docker@4d82cb762091:/build/source$
```

p3x-robot, 2018-06-28:
I used it the router but i still have no debug symbols:
```
root@digi:/opt/router-scripts-openwrt# opkg install gdb
Installing gdb (7.11.1-1) to root...
Downloading http://cdn.corifeus.com/lede/17.01.4/packages/arm_cortex-a9_vfpv3/base/gdb_7.11.1-1_arm_cortex-a9_vfpv3.ipk
Configuring gdb.
root@digi:/opt/router-scripts-openwrt# gdb node /opt/router-scripts-openwrt/test/node.4309.6.1530167810.core
GNU gdb (GDB) 7.11.1
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "arm-openwrt-linux".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from node...(no debugging symbols found)...done.
[New LWP 4309]
[New LWP 4310]
[New LWP 4312]
[New LWP 4311]
[New LWP 4313]
Core was generated by `node'.
Program terminated with signal SIGABRT, Aborted.
#0  0xb6f2bd5c in ?? ()
[Current thread is 1 (LWP 4309)]
(gdb) list
No symbol table is loaded.  Use the "file" command.
(gdb)
```

nxhack, 2018-06-28:
You are building node.js, Therefore you have unstripped binary (./out/Release/node). So on target machine, run, core dump, and then use it with gdb.

If you want more information, please build 'node_g'.

p3x-robot, 2018-06-28:
but when i create a CONFIG_NODEJS_DEBUG=y it does build it shows this error:

```make package/feeds/node/node/{clean,prepare,compile} package/index V=s```

```
/build/source/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/node-v10.5.0/out/Debug/obj.target/v8_base/deps/v8/src/heap/factory.o:/build/source/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/node-v10.5.0/out/../deps/v8/src/heap/factory.cc:1070: more undefined references to `void v8::internal::String::WriteToFlat<unsigned char>(v8::internal::String*, unsigned char*, int, int)' follow
collect2: error: ld returned 1 exit status
node.target.mk:156: recipe for target '/build/source/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/node-v10.5.0/out/Debug/node' failed
make[4]: *** [/build/source/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/node-v10.5.0/out/Debug/node] Error 1
rm ae87dc835bea2d326910f4833d37b21735164336.intermediate
Makefile:91: recipe for target 'node_g' failed
make[3]: *** [node_g] Error 2
make[3]: Leaving directory '/build/source/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/node-v10.5.0'
Makefile:203: recipe for target '/build/source/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/node-v10.5.0/.built' failed
make[2]: *** [/build/source/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/node-v10.5.0/.built] Error 2
make[2]: Leaving directory '/build/source/feeds/node/node'
package/Makefile:105: recipe for target 'package/feeds/node/node/compile' failed
make[1]: *** [package/feeds/node/node/compile] Error 2
make[1]: Leaving directory '/build/source'
/build/source/include/toplevel.mk:198: recipe for target 'package/feeds/node/node/compile' failed
make: *** [package/feeds/node/node/compile] Error 2
docker@4d82cb762091:/build/source$
```

**Do you know what I mean?***

p3x-robot, 2018-06-28:
**recipe for target 'node_g' failed** :(

nxhack, 2018-06-28:
My side too,  Now building node v10.x  with debug build. Please wait as it will take time.

p3x-robot, 2018-06-28:
if i very nicely ask to build an arm_cortex-a9_vfpv3 node 10.5 ipk? for me it gives this error as i wrote above.

p3x-robot, 2018-06-28:
```
/build/source/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/node-v10.5.0/
out/Debug/obj.target/v8_base/deps/v8/src/heap/factory.o:
/build/source/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/
node-v10.5.0/out/../deps/v8/src/heap/factory.cc:1070: 
more undefined references to 
`void v8::internal::String::WriteToFlat<unsigned char>(v8::internal::String*, unsigned char*, int, int)' 
follow collect2: error: ld returned 1 exit status
```

p3x-robot, 2018-06-28:
looks like this file missing:
```
fatal error: opening dependency file 
/build/source/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/node-v10.5.0
/out/Release/.deps//build/source/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/node-v10.5.0
/out/Release/obj.target/v8_base/deps/v8/src/wasm/wasm-code-manager.o.d.raw: 
No such file or directory
```

missing file`wasm-code-manager`

nxhack, 2018-06-29:
On OpenWrt Trunk arm_cortex-a9+vfpv3_musl_eabi gcc7.3.0, node_g build fine.
Next I will create an environment of 17.01 and test it. Please wait.

p3x-robot, 2018-06-29:
do you know, how i can execute an arm node with linux intel cpu and execute with arm build? (without running on the router because i am havent deployed openwrt 18 , still on lede 17 stable).
i see gdb, but is not giving me execution on a differenet a platform, do you know what i mean and how?
THANKS SO MUCH!

nxhack, 2018-06-29:
First of all, I am trying to verify the build error you reported. My node packages are made for OpenWrt Trunk. To that end, I am first testing whether I can successfully build in that environment. Since it was able to build normally, Next I have just started building tests since building the build environment of 17.01. So, if I have build errors according to your report I will repair. After that, please use YOUR own node_g build and YOU DO debug in the target environment.

p3x-robot, 2018-06-29:
@nxhack but i am using `17.01.4` and the `node_g` does not build as the error shows aboce... :(

p3x-robot, 2018-06-29:
the node 17.0.4 i can build without debug but i get that error when i execute node...

nxhack, 2018-06-29:
It takes long time to build. so do not rush, ;)

p3x-robot, 2018-06-29:
ok, thanks so much!

nxhack, 2018-06-29:
on 17.01.x , caught same error.

I will check. It is probably attributed to binutils or gcc version.

p3x-robot, 2018-06-29:
ahh so awesome, thanks so muc @nxhack !!! 🥇

nxhack, 2018-06-30:
OpenWrt 17.01.4, binutils 2.27, gcc 6.3.0 , node_g build fine.

p3x-robot, 2018-06-30:
what does it mean?
i should create a new docker with these versions (binutils 2.27, gcc 6.3.0, node_g) ?

I build with this Dockerfile:
https://github.com/patrikx3/openwrt-insomnia/blob/master/openwrt-base/Dockerfile#L21

Should I change something?

nxhack, 2018-06-30:
menuconfig

Advanced configuration options (for developers)  --->
Toolchain Options  --->
 Binutils Version
 GCC compiler Version

p3x-robot, 2018-07-01:
ok found, rebuilding 15 hours, helps for testing.

i guess with openwrt 18 we will dont need setting, right?

p3x-robot, 2018-07-01:
but it means my firmware has to be re-installed as well right? i guess the whole firmware is different, right?

p3x-robot, 2018-07-01:
ciao!
i had a built original build, so then i changed `.config` and added the `binutils` and `gcc` version as you said, but sadly, i upgraded the firmware and installed `node 10.5`, i get the same error.

do you think it is the problem i updated the `.config` and the re-built as it was?

do you think if i create a virgin empty `bin` folder, then it will work?

p3x-robot, 2018-07-01:
i create a `virgin` build now. i hope it will work, before was working cool, now the new `18-rc1` and the stable `lede` with `nodejs 10.5` is none working. sadly. :(

p3x-robot, 2018-07-03:
@nxhack i am still afraid to use 18.06-rc on my linksys and d-link routers, but the nodejs 10.5 works with openwrt latest tag:?

p3x-robot, 2018-07-05:
@nxhack do you think 10.6 will work on LEDE?

p3x-robot, 2018-07-11:
yes on stable lede works on linksys 3200acm, testing mips.

p3x-robot, 2018-07-11:
Testing D-LINK DIR860L B1, MIPS is confirmed working nodejs 10.6, we have threads in nodejs on a small router :) :1st_place_medal:

nxhack, 2018-07-11:
Refer: https://github.com/nodejs/node/issues/21562
Close.

closed 2018-07-11 by nxhack