nxhack/openwrt-node-packages #330: (Error compiling) 'to_string' is not member of 'std' - Node v8.11.3

SharkSlayer, 2018-07-03:
Hello sir,
I'm trying to build OpenWRT Chaos Calmer 15.05  with OpenWRT nodejs packages
My build system : Ubuntu linux 16.04.1
In menuconfig, i select Languages > Node.js > Configuration -> version v8.x
After a while, i encounter this problem and i dont know how to fix it.
Can you help me ?

> **../deps/v8/src/ic/ic.cc: In member function 'void v8::internal::IC::TraceIC(const char*, v8::internal::Handle<v8::internal::Object>, v8::internal::IC::State, v8::internal::IC::State)':
../deps/v8/src/ic/ic.cc:164:29: error: 'to_string' is not a member of 'std'
     ic_info.instance_type = std::to_string(map->instance_type());
                                          ^
deps/v8/src/v8_base.target.mk:587: recipe for target '/home/minh-pc/openwrt/build_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/node-v8.11.3/out/Release/obj.target/v8_base/deps/v8/src/ic/ic.o' failed
make[5]: *** [/home/minh-pc/openwrt/build_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/node-v8.11.3/out/Release/obj.target/v8_base/deps/v8/src/ic/ic.o] Error 1
rm b7996b0bf538570a55414ede3051cb470ece29b3.intermediate
Makefile:88: recipe for target 'node' failed
make[4]: *** [node] Error 2
make[4]: Leaving directory '/home/minh-pc/openwrt/build_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/node-v8.11.3'
Makefile:159: recipe for target '/home/minh-pc/openwrt/build_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/node-v8.11.3/.built' failed**

nxhack, 2018-07-03:
Chaos Calmer, please read this document.

https://github.com/nxhack/openwrt-node-packages/blob/for-15.05/README.md

nxhack, 2018-07-03:
Since Chaos Calmer is old, I will not do maintenance frequently in the future.

SharkSlayer, 2018-07-03:
I have MediaTek Linkit Smart 7688 and i need firmware with node js v8.x on it. Through MediaTek website they suggest me to use OpenWRT Chaos Calmer 15.05.
I follow your instructions in here :
[https://github.com/nxhack/openwrt-node-packages/issues/8#issuecomment-316937931](url)

But it doesn't work. I did exactly but got so many errors

nxhack, 2018-07-05:
Hi @SharkSlayer 

The build will succeed with this procedure.
https://github.com/nxhack/openwrt-node-packages/wiki

But I have not tested whether it will run on the target machine. Because I do not have a target machine.
Please let me know if it works.

SharkSlayer, 2018-07-06:
Thanks for your reply @nxhack 
Did you build it with node.js 8.11.3 ?
Well, It worked with node.js 6.14.3 for me but not 8.11.3.
I'll try it again.

closed 2018-07-09 by nxhack