nxhack/openwrt-node-packages #379: ideino, johnny-five or cylon

scenaristeur, 2018-09-01:
Hi @nxhack , i've now succeded installing LEDE18.06, on my yun, 
& expand exroot ( there was a problem for mounting ext4 or f2fs partition, but opkg install f2fs-tools solves the problem) 
ArduinoOs is ok but not Luci :-(...

but I have some questions : 
node V10 installed -> but no npm to install other node_modules ???

How to access Arduino pins from nodejs ? 
i used to use ideino with LININOIO but there was a 'lininoio start' command to replace the Arduino sketch and map the gpio pins to linux 
Here I only see gpio21,22,23 in /sys/class/gpio ?? but not all the pins from the yun...

Do you know how to access them from johnny-five or cylon ???

Do you have any example of using one of this libs to access all arduino pins ???

Thx
David

nxhack, 2018-09-03:
I saw that your yun is rev2. Please check the datasheet and yun driver code.

https://content.arduino.cc/assets/ArduinoYunRev2-schematic.pdf
https://github.com/openwrt/openwrt/blob/openwrt-18.06/target/linux/ar71xx/files/arch/mips/ath79/mach-arduino-yun.c

On my old yun with original firmware.
```
root@Arduino:~# ll /sys/class/gpio/
total 0
drwxr-xr-x  2 root root    0 Jan  1  1970 ./
drwxr-xr-x 24 root root    0 Jan  1  1970 ../
--w-------  1 root root 4096 Jan  1  1970 export
lrwxrwxrwx  1 root root    0 Jan  1  1970 gpio21 -> ../../devices/virtual/gpio/gpio21/
lrwxrwxrwx  1 root root    0 Jan  1  1970 gpio22 -> ../../devices/virtual/gpio/gpio22/
lrwxrwxrwx  1 root root    0 Jan  1  1970 gpio23 -> ../../devices/virtual/gpio/gpio23/
lrwxrwxrwx  1 root root    0 Jan  1  1970 gpiochip0 -> ../../devices/virtual/gpio/gpiochip0/
--w-------  1 root root 4096 Jan  1  1970 unexport
```

and
Vanilla OpenWrt Linio-IO kernel module is not supported.

nxhack, 2018-11-21:
I will close it. Please do not hesitate to reopen.

closed 2018-11-21 by nxhack