nxhack/openwrt-node-packages #464: BAD NEWS: remove mips support

nxhack, 2019-02-22:
Upstream of node.js, support for mips seems to be removed in the future. I plan to maintain mips as much as possible but may not be able to do so depending on the circumstances.

https://github.com/nodejs/node/issues/26179
https://github.com/nodejs/node/pull/26192

nxhack, 2019-04-25:
The mips support has been removed in v12.0.0.
I'm reverting the deleted codes.

nxhack, 2019-04-26:
Modules that will break in Node 12

node-authenticate-pam
node-bignum
node-bleacon
node-bluetooth-hci-socket
node-ejdb
node-ffi
node-fuse-bindings
node-hashtable
node-mdns
node-mknod
node-modbus
node-net-ping 
node-node-expat
node-node-hid
node-node-hid-stream
node-node-red-contrib-modbus
node-openzwave-shared
node-pi-spi
node-rpio
node-sleep
node-sqlite3
node-serialport-bindings
  node-serialport
  node-arduino-firmata	
  node-firmata
  node-johnny-five
  node-node-enocean-utils

p3x-robot, 2019-06-29:
how can you are saying nodejs 12 dropped mips, i just built mt7621 (ramips):
https://cdn.corifeus.com/openwrt/18.06.3/packages/mipsel_24kc/node/

and it works. how come it still works?

p3x-robot, 2019-06-30:
ok, got it you have the patches for mips. thanks

nxhack, 2019-07-01:
The cross compilation environment is broken in v12.5.0.
The situation is difficult.

p3x-robot, 2019-07-01:
ok, i am happy, that i am buying a  new linksys wrt32x, because i will not use the d-link dir 860d b1 anymore, it looks like the last was in nodejs 12.4.0 for the ramips build....

p3x-robot, 2019-07-01:
that is why you reverted to 12.4.0?

nxhack, 2019-07-01:
@p3x-robot 
As it was likely to take time to support v12.5.0, I made a dedicated branch. For that purpose, I restored the master branch and the development branch.
It is the latest version that v12.4.0 is supported.

nxhack, 2019-07-03:
build: add mips64el support

https://github.com/nodejs/node/pull/27992

p3x-robot, 2019-07-03:
so it will still not support ramips  mt7621 , right?

p3x-robot, 2019-07-03:
it is only for mipsel64 i suppose.

nxhack, 2019-07-03:
Exactly.
mips32 will not be considered much.
I will try my best as much as possible.

nxhack, 2019-11-27:
After v8-7.9, mips(eb) and mips64(eb) will not work.

alenteria, 2020-08-17:
I successfully compiled nodejs v10 for mipsel_24kc (newifi D2).
But the only problem I have right now is it's really slow. CPU and ram consumptions are at less than 50% only. Simple http ajax request, express response at 2-3 seconds.

nxhack, 2020-09-28:
Verified with version 14.12.0. Unfortunately, I confirmed that it does not work on mips(eb) and mips64(eb).

closed 2021-05-06 by nxhack