nxhack/openwrt-node-packages #465: Node-Red stuck on startup

salmanisd, 2019-02-25:
Hello...
I built the latest Node-Red version and installed it on my router with MIPS 24KEc.
If i start it with typing `node-red` on the terminal, i get 
```
Welcome to Node-RED
===================

25 Feb 10:22:45 - [info] Node-RED version: v0.19.5
25 Feb 10:22:45 - [info] Node.js  version: v8.10.0
25 Feb 10:22:45 - [info] Linux 4.14.95 mipsel LE

```

and thats it.It doesn't proceed further and I cannot also reach the node-red UI  through my PC.
Can't find any logs and dont know how to debug this issue.

nxhack, 2019-02-25:
Hi @salmanisd 
```
25 Feb 11:22:59 - [info] Palette editor disabled : npm command not found
```
The npm command is a separate package(node-npm). What happens after installing the npm package?

nxhack, 2019-02-25:
I will correct the dependency of npm in the node-red package.

salmanisd, 2019-02-26:
> Hi @salmanisd
> 
> ```
> 25 Feb 11:22:59 - [info] Palette editor disabled : npm command not found
> ```
> 
> The npm command is a separate package(node-npm). What happens after installing the npm package?

The npm package is already installed
```
root@OpenWrt:~# npm -v
5.6.0
```
Although it took more than 40 secs wait just to display the npm version.

nxhack, 2019-02-27:
Hi @salmanisd 

> Although it took more than 40 secs wait just to display the npm version.

Oh! 40 secs...

(I think that you already know...)
Is there enough memory?
Is there enough CPU power?

And... Please change the settings etc. often pointed out when the operation of npm is slow.
```
npm set progress=false
npm config set registry http://registry.npmjs.org/
```

salmanisd, 2019-02-27:
Hi 

I changed npm settings like you suggested but no effect. I guess the MIPS CPU I have on the router is just not able to run node-red successfully.
Thanks for your help.

closed 2019-02-27 by salmanisd