nxhack/openwrt-node-packages #470: v10.15.1: make_unique is not a member of std

sukovec, 2019-02-26:
Hello!

I'm trying to compile newest version of LTS nodejs , but it's not working, probably due to older version of compiler. The first compile error is following: 

    ../src/cares_wrap.cc: In function 'void node::cares_wrap::{anonymous}::Query(const v8::FunctionCallbackInfo<v8::Value>&)':
    ../src/cares_wrap.cc:1799:15: error: 'make_unique' is not a member of 'std'
       auto wrap = std::make_unique<Wrap>(channel, req_wrap_obj);

It looks like `make_unique` is c++14 feature, which is not enabled by default with `gcc 4.8.4-2`. I can, of course, install newer version of gcc, but I'm trying to stay as close as possible to https://github.com/CZ-NIC/turris-os (it's compiled in docker, based on ubuntu-trusty image). 

I would expect, that `./configure` script will take care of this and will issue error or automatically switch to `-std=c++14`, but it's not happening and compilation is interrupted after a while. Is there something to do to _make node compile again_ without radical changes in OS/toolchain?

PS: Version 10.13.0 works fine.
PSS: Thanks for your great work!

nxhack, 2019-02-27:
Hi @sukovec 

I think your analysis is correct.

However, considering the future, I think it is correct to update the version of gcc.

I checked the method of trying to solve it on the spot but I gave up so I had to take a lot of hands on "deps/v8".

Sorry, I couldn't help you.

sukovec, 2019-02-27:
Hi! 

You are right. It looks that ubuntu trusty is on end of its LTS, so I will try to move on onto newer Ubuntu (with newer gcc) and I will also try to convince CZ-NIC developers to move on newer base image. 

Thank you!

closed 2019-02-27 by sukovec