nxhack/openwrt-node-packages #487: Recursive dependency

cconde, 2019-03-28:
Hi,

I'm receiving these messages:
```
tmp/.config-package.in:21532:   symbol PACKAGE_node-cylon-gpio is selected by PACKAGE_node-cylon-gpio
tmp/.config-package.in:21565:error: recursive dependency detected!
```
I'm not using it, so I'm uninstalling the module, but I thought it may be worth to notify it.

Thanks for such good job,
Ivan

nxhack, 2019-04-03:
It is possible that there is a conflict with the OpenWrt package feed.

Please try it
```
rm ./package/feeds/packages/node
rm ./package/feeds/packages/node-*
./scripts/feeds install -a -p node
```

nxhack, 2019-04-17:
I will close the issue. Don't hesitate to reopen it if you still have the issue.

closed 2019-04-17 by nxhack