nxhack/openwrt-node-packages #519: Languages > Node.js folder Empty

sunzongzheng, 2019-05-30:
[image]
Why empty? I can not choose one version to compile.

nxhack, 2019-05-30:
Probably You don't use openwrt 18.06 version upper, or it's an unsupported CPU ARCH.

Support for the old openwrt has been dropped to support the latest node.js.

Please consider using this:  https://github.com/nxhack/openwrt-node-packages/tree/for-15.05
(not updated frequently)

sunzongzheng, 2019-05-30:
I run it on wsl. ubuntu 16.04 
[image]

Can i do a cross compilation? I can not change the target system after pressing enter
[image]

nxhack, 2019-05-30:
Please read : https://openwrt.org/docs/guide-developer/build-system/use-buildsystem

nxhack, 2019-05-30:
And  Please read : https://github.com/nxhack/openwrt-node-packages#illegal-instruction-issue

nxhack, 2019-05-30:
And : https://forum.openwrt.org/t/fun-project-build-lede-using-bash-on-ubuntu-on-windows-wsl-windows-subsystem-for-linux/3819

nxhack, 2019-05-30:
And : https://github.com/Microsoft/WSL/issues/432

nxhack, 2019-05-30:
I recommend using VirtualBox.

nxhack, 2019-07-26:
I will close the issue. Don't hesitate to reopen it if you still have the issue.

closed 2019-07-26 by nxhack