nxhack/openwrt-node-packages #526: Cannot find Homebridge in iOS Home app

NN708, 2019-06-13:
I am using OpenWrt 18.06.2 on Netgear R6220, with FPU emulator enabled. I compiled OpenWrt with node, node-npm, node-hap-nodejs, node-homebridge and avahi-dbus-daemon. However, after installation, I cannot find Homebridge in my iOS Home app.
I run `logread | fgrep homebridge` and `logread | fgrep avahi-daemon`. It seems that they are working normally:
```
root@OpenWrt:~# logread | fgrep homebridge
daemon.err homebridge[1590]: npm info it worked if it ends with ok
daemon.err homebridge[1590]: npm info using npm@6.9.0
daemon.err homebridge[1590]: npm info using node@v10.16.0
daemon.err homebridge[1590]: npm timing npm Completed in 13741ms
daemon.err homebridge[1590]: npm info ok
daemon.err homebridge[1590]: config.json (/usr/share/homebridge/config.json) not found.
daemon.err homebridge[1590]: No plugins found. See the README for information on installing plugins.
daemon.info homebridge[1590]: Setup Payload:
daemon.info homebridge[1590]: X-HM://0023*********
daemon.info homebridge[1590]: Scan this code with your HomeKit app on your iOS device to pair with Homebridge: *****
daemon.info homebridge[1590]: Or enter this code with your HomeKit app on your iOS device to pair with Homebridge: ***-**-***
daemon.info homebridge[1590]: Homebridge is running on port *****.
```

```
daemon.info avahi-daemon[1064]: Found user 'nobody' (UID *****) and group 'nogroup' (GID *****).
daemon.info avahi-daemon[1064]: Successfully dropped root privileges.
daemon.info avahi-daemon[1064]: avahi-daemon 0.6.32 starting up.
daemon.warn avahi-daemon[1064]: WARNING: No NSS support for mDNS detected, consider installing nss-mdns!
daemon.info avahi-daemon[1064]: No service file found in /etc/avahi/services.
daemon.info avahi-daemon[1064]: Network interface enumeration completed.
daemon.info avahi-daemon[1064]: Registering HINFO record with values 'MIPS'/'LINUX'.
daemon.info avahi-daemon[1064]: Server startup complete. Host name is OpenWrt.local. Local service cookie is **********.
daemon.info avahi-daemon[1064]: Joining mDNS multicast group on interface br-lan.IPv4 with address 192.168.1.1.
daemon.info avahi-daemon[1064]: New relevant interface br-lan.IPv4 for mDNS.
daemon.info avahi-daemon[1064]: Registering new address record for 192.168.1.1 on br-lan.IPv4.
daemon.info avahi-daemon[1064]: Joining mDNS multicast group on interface eth0.IPv6 with address fe80::12da:43ff:****:****.
daemon.info avahi-daemon[1064]: New relevant interface eth0.IPv6 for mDNS.
daemon.info avahi-daemon[1064]: Registering new address record for fe80::12da:43ff:****:**** on eth0.*.
daemon.info avahi-daemon[1064]: Joining mDNS multicast group on interface eth0.2.IPv6 with address fe80::12da:43ff:****:****.
daemon.info avahi-daemon[1064]: New relevant interface eth0.2.IPv6 for mDNS.
daemon.info avahi-daemon[1064]: Registering new address record for fe80::12da:43ff:****:**** on eth0.2.*.
daemon.info avahi-daemon[1064]: Joining mDNS multicast group on interface br-lan.IPv6 with address fe80::12da:43ff:****:****.
daemon.info avahi-daemon[1064]: New relevant interface br-lan.IPv6 for mDNS.
daemon.info avahi-daemon[1064]: Registering new address record for fe80::12da:43ff:****:**** on br-lan.*.
daemon.info avahi-daemon[1064]: Leaving mDNS multicast group on interface br-lan.IPv6 with address fe80::12da:43ff:****:****.
daemon.info avahi-daemon[1064]: Joining mDNS multicast group on interface br-lan.IPv6 with address fd25:****:****::1.
daemon.info avahi-daemon[1064]: Registering new address record for fd25:****:****::1 on br-lan.*.
daemon.info avahi-daemon[1064]: Withdrawing address record for fe80::12da:43ff:****:**** on br-lan.
daemon.info avahi-daemon[1064]: Joining mDNS multicast group on interface wlan0.IPv6 with address fe80::12da:43ff:****:****.
daemon.info avahi-daemon[1064]: New relevant interface wlan0.IPv6 for mDNS.
daemon.info avahi-daemon[1064]: Registering new address record for fe80::12da:43ff:****:**** on wlan0.*.
daemon.info avahi-daemon[1064]: Joining mDNS multicast group on interface wlan1.IPv6 with address fe80::12da:43ff:****:****.
daemon.info avahi-daemon[1064]: New relevant interface wlan1.IPv6 for mDNS.
daemon.info avahi-daemon[1064]: Registering new address record for fe80::12da:43ff:****:**** on wlan1.*.
```
I also tested the connection with Homebridge port and the port was open. Then I checked with `avahi-browse -at` and found out that there were nothing about Homebridge.
However, my Raspberry Pi can still run Homebridge normally. What’s the problem? Did I omit any required packages?

nxhack, 2019-06-13:
Hi @NN708 

Does homebridge bind with the intended interface?

please check.
```
opkg update
opkg install avahi-utils
avahi-browse -t _hap._tcp
```

NN708, 2019-06-13:
It returns nothing. What should I do to bind with the interface?

nxhack, 2019-06-13:
Can you see the QR code that homebridge outputs in logread?

NN708, 2019-06-13:
Yes, I can. I scanned the QR code with Home app, then it was searching for accessory, but finally no accessory was found.

nxhack, 2019-06-13:
hmm...
You should see "_hap._tcp" if homebrige is started correctly. So I think homebridge is not running.

In my test environment:
```
~ # avahi-browse -t _hap._tcp
+ br-lan IPv4 Homebridge-CAD8                               _hap._tcp            local
```
```
~ # avahi-browse -rt _hap._tcp
+ br-lan IPv4 Homebridge-CAD8                               _hap._tcp            local
= br-lan IPv4 Homebridge-CAD8                               _hap._tcp            local
   hostname = [CC_22_3D_E3_CE_30.local]
   address = [172.17.0.2]
   port = [33291]
   txt = ["sh=hJpPnw==" "sf=1" "ci=2" "ff=0" "s#=1" "c#=2" "id=CC:22:3D:E3:CE:30" "pv=1.0" "md=Homebridge"]
```

NN708, 2019-06-13:
I run `ps | grep homebridge` and that shows it is running. Also, the port of Homebridge is seen open on another computer.

nxhack, 2019-06-13:
hmm... It is a curious condition.
```
avahi-browse -t _hap._tcp
```
It does not work if  _hap._tcp service is not registered.

nxhack, 2019-06-13:
Please check the simple check of avahi.
```
opkg install avahi-daemon-service-ssh
/etc/init.d/avahi-daemon restart
avahi-browse -at
```

NN708, 2019-06-13:
There are something about OpenWrt:
```
root@OpenWrt:~# avahi-browse -at
+  wlan1 IPv6 OpenWrt                                       _ssh._tcp            local
+  wlan0 IPv6 OpenWrt                                       _ssh._tcp            local
+ br-lan IPv6 OpenWrt                                       _ssh._tcp            local
+ br-lan IPv4 OpenWrt                                       _ssh._tcp            local
+   eth0 IPv6 OpenWrt                                       _ssh._tcp            local
```

nxhack, 2019-06-13:
It's a homebridge side problem because avahi seems to be working.

nxhack, 2019-06-13:
Which network interface do you want to use?

> br-lan IPv4 OpenWrt

correct?

NN708, 2019-06-13:
Yes

NN708, 2019-06-13:
I'm trying to find the Homebridge log file. Homebridge Wiki says:
>Homebridge logs to two places:
/var/log/homebridge.log
/var/log/homebridge.err

But I can't find these files. Where are they on OpenWrt?

nxhack, 2019-06-13:
Please try this

https://www.reddit.com/r/homebridge/comments/aiexkx/problems_connecting_homebridge_to_homekit/

nxhack, 2019-06-13:
> But I can't find these files. Where are they on OpenWrt?

All logs are in ring buffer memory. Using logread command is the Openwrt style. :)

NN708, 2019-06-13:
I tried to add `"mdns": {"interface": "192.168.1.1"}` to my config.json, but in `avahi-browse -a` there are still nothing. The good news is, the Home app prompts "Uncertified Accessory" now, but after I press "Add Anyway", it says "Home couldn't connect to this accessory".

nxhack, 2019-06-13:
How about this command
```
avahi-browse -t _hap._tcp
```

Check  accessories and platforms in config.json.
Try restarting your homebridge, see below
https://github.com/nfarina/homebridge#my-ios-app-cant-find-homebridge

NN708, 2019-06-13:
I have already read and tried it, but no use. I think the problem is that Homebridge didn't broadcast via avahi, but it didn't give out any error message...

nxhack, 2019-06-13:
[FYI]
The default config.json path of the OpenWrt homebridge package is
```
/usr/share/homebridge/config.json
```

and The privilege to operate is "homebridge" user.

nxhack, 2019-06-13:
check
```
/etc/avhi/avahi-daemon.conf
```
specify 
```
allow-interfaces=br-lan
``` 
https://forum.archive.openwrt.org/viewtopic.php?id=56615

NN708, 2019-06-13:
It didn't solve the problem. And there's a warning with avahi-daemon:
```
*** WARNING: Detected another IPv4 mDNS stack running on this host. This makes mDNS unreliable and is thus not recommended. ***
```

nxhack, 2019-06-13:
```
 killall avahi-daemon
/etc/init.d/avahi-daemon start
```

nxhack, 2019-06-13:
Interesting... homebridge bind udp 5353

This procedure may be necessary if you want to restart avahi-daemon.
```
/etc/init.d/homebridge stop
/etc/init.d/avahi-daemon restart
/etc/init.d/homebridge start
```

NN708, 2019-06-13:
Now the Home app can discover Homebridge on OpenWrt, so perhaps it is not the problem with avahi...
What's more, I tried to run Homebridge my Raspberry Pi again and there's no _hap._tcp in `avahi-browse -a`, but I can connect to it normally.

nxhack, 2019-06-13:
It does not appear in "**-a**" even in my environment. Please check with "**-t _hap._tcp**"

nxhack, 2019-06-13:
@NN708 
If possible, I hope you publish a summary of the solution.

Thank you.

NN708, 2019-06-13:
There isn't any solution so far... Home app can only "discover" Homebridge, but can't connect to it...

NN708, 2019-06-13:
Home app can find the accessory:
[image]
Then I try to connect to it:
[image]
[image]
But failed:
[image]
Looks like some connection issues between Home app and Homebridge...

nxhack, 2019-06-13:
Ping youre iPhone from the OpenWrt host and check continuity.

https://www.iphonefaq.org/archives/971437

nxhack, 2019-06-14:
This may also be useful for problem solving.

iPhone app - Discovery - DNS-SD Browser
https://apps.apple.com/us/app/discovery-dns-sd-browser/id305441017

nxhack, 2019-06-14:
A common thing to try in case of home bridge connection problem is
disable IPv6, 
restart the router, 
disable IGMP snooping setting on the router.

NN708, 2019-06-14:
Thank you for your suggestions. Ping is OK, and I found that when I try to connect Homebridge in my Home app, the CPU usage increases to 50% (normally it is about 2%). That indicates there’s no connection problem between iOS and Homebridge.
Now I guess, perhaps the issue is related to the low performance of my router. It takes the router more than 20 seconds to start the Homebridge process, which is normally under 5 seconds on my Raspberry Pi. If Homebridge doesn’t respond in time, the Home app may think the connection is lost.

nxhack, 2019-06-14:
I have a similar experience. It tries to register to home.app many times and succeeds. It works well once registered.

nxhack, 2019-06-14:
There are also problems with key generation, so can you try it?
```
opkg install haveged
```

NN708, 2019-06-14:
I tried to use haveged, but it didn’t solve the problem.

nxhack, 2019-06-17:
I tested with my home router.

CPU: QCA9563 @ 775 MHz
Mem: 128MB

I have run homebridge in debug mode.
As a result, connected to HAPServer from the iOS side and found that the session timed out while homebridge was generating a key pair.

It seems to be a specification on the iOS side.
https://developer.apple.com/documentation/foundation/nsurlsessionconfiguration/1408259-timeoutintervalforrequest?language=objc

homebridge works fine with raspberry pi zero w + openwrt (head).
Also it worked in the past with AR9331 400MHz 64MB. (lede-17.01, node-v6, homebridge[unknown old version])

NN708, 2019-06-17:
I found that the problem also exists in pairing with accessories (using plugin). The pairing timed out on OpenWrt, but works fine on Raspberry Pi.

NN708, 2019-06-17:
I also tried to pair on Raspberry Pi first, and then copy Homebridge /persist files to OpenWrt, but it didn’t work.

nxhack, 2019-06-17:
Stop using the avahi package and I will try "mdnsresponder" package or "umdns" package.
It will take some time to test.

nxhack, 2019-06-18:
It seems that there has been a change in hap-nodejs. The old precompiled binary module is no longer used. There seems to be a difference in performance.

https://github.com/KhaosT/HAP-NodeJS/commit/47966d13cfea35ce7acb4b0780faaae745eef69c#diff-b9cfc7f2cdf78a7f4b91a753d10865a2
https://github.com/KhaosT/HAP-NodeJS/commit/2a66d6fa58a51ae87346dfcae063696d0352b47f#diff-b9cfc7f2cdf78a7f4b91a753d10865a2

NN708, 2019-06-18:
So we have to compile TweetNaCl.js first?

nxhack, 2019-06-18:
Changed to be able to select the old hap-nodejs module. (I have not tested it in my environment yet.)

https://github.com/nxhack/openwrt-node-packages/blob/master/node-hap-nodejs/Makefile

NN708, 2019-06-18:
OK, I will try it later.

NN708, 2019-06-19:
I tried HAP-NodeJS 0.4.49, but it has the same problem.

nxhack, 2019-07-01:
I tested on raspberry pi zero w with openwrt head and homebridge of community package, but I could connect easily.
As you may have guessed it is a problem with floating point computing power at key generation.

I will mention in README.md that it often does not work in environments where mips fpu emu is used.

nxhack, 2019-11-27:
I will close the issue. Don't hesitate to re-open it if you still have the issue.

closed 2019-11-27 by nxhack

nature2608, 2020-01-04:
I have the same problems like yours,i used [homekit-bridge](https://github.com/NRCHKB/node-red-contrib-homekit-bridged/issues/191),have you successfuly run it in openwrt? @NN708 
I have tried run it on ar9558 and mt7621 devices,but it still can not be connected.

NN708, 2020-01-04:
Sadly, no. I’m using my Raspberry Pi instead.

nature2608, 2020-01-04:
what a shame.I still can not figure out why after trying so many ways.