nxhack/openwrt-node-packages #530: node-pty

rgillan, 2019-06-21:
Hello,
We have tried to cross compile node-pty but keep running into a problem that the MS folks seem to think it's in the cross compile chain.  We are able to cross compile other modules (e.g. websocket) with no issue on the target so we're pretty stumped.  Any ideas or help would be greatly appreciated; node-pty is the last of the code we need to port to OpenWRT.

https://github.com/microsoft/node-pty/issues/325

Thanks
Rob

nxhack, 2019-06-21:
Hi @rgillan 

made it.
https://github.com/nxhack/openwrt-node-packages/tree/master/node-pty

Work for me.
sample https://www.npmjs.com/package/node-pty#example-usage
```bash
~ # cat /etc/openwrt_*
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='SNAPSHOT'
DISTRIB_REVISION='r10210-09c6885'
DISTRIB_TARGET='brcm2708/bcm2710'
DISTRIB_ARCH='aarch64_cortex-a53'
DISTRIB_DESCRIPTION='OpenWrt SNAPSHOT r10210-09c6885'
DISTRIB_TAINTS='no-all'
r10210-09c6885
~ # uname -a
Linux 688644c1be94 4.15.0-1041-aws #43-Ubuntu SMP Thu Jun 6 13:40:52 UTC 2019 aarch64 GNU/Linux
~ # node index.js 
bash-5.0# ls
ls
index.js
bash-5.0# ls
index.js
bash-5.0# exit
^C
~ # 
```

rgillan, 2019-06-21:
Fantastic and appreciate the promptness.  I'll run this up on our target on Monday and report back.  I've also started a complete rebuild of our dev/cross compile machine (want to find out what went wrong as well).  Cheers /r

rgillan, 2019-06-24:
Morikawa-san, we've managed to create an image with node-pty compiled in.  About to start testing from node itself.  One question, it appears that the build process installs the pre-compiled node modules into /usr/lib/node (rather than the typical /usr/lib/node_modules).  Is this something that's configurable in the process?  Very clean and simple process, thanks so much

nxhack, 2019-06-24:
> One question, it appears that the build process installs the pre-compiled node modules into /usr/lib/node (rather than the typical /usr/lib/node_modules).

Specification of the openwrt community nodejs package. It is to distinguish what is managed by the package.

rgillan, 2019-06-24:
makes sense (we're relative newbies in OpenWRT world but I expect not for long).

So node-pty is throwing an error when I think it's trying to open the default path:
execvp(3) failed.: No such file or directory

I'm in communications with the MS folks who manage the module and I'll loop back here once we find the cause.

Thanks for all your help

rgillan, 2019-06-24:
Ok, our fault.  Our node.js call to the module was expecting bash, not ash, as the shell.  All great and working now (thanks so much).

closed 2019-06-24 by rgillan