nxhack/openwrt-node-packages #533: Websocket module

rgillan, 2019-06-24:
Hi, so there's another module that would be great to be added to the repo.  The websocket module has two optional compiled files (bufferutil.node, validation.node) that are used if present (or defaults to pure js if not albeit with some decrease in performance).

If this module could be added (not urgent) it would be fantastic.  Appreciate all your help with these.

Cheers
Rob

nxhack, 2019-06-24:
https://github.com/nxhack/openwrt-node-packages/tree/master/node-bufferutil
https://github.com/nxhack/openwrt-node-packages/tree/master/node-utf-8-validate

rgillan, 2019-06-24:
Hi, and again the promptness is awesome.  I'll move these into the websocket module, test and let you know (I see you had already added ws, so these came from there originally as you know).  Cheers /r

rgillan, 2019-06-25:
Hi, so they both load ok, but it looks like the websocket author made some changes to the original modules, so bufferutil is throwing an error.  Maybe inclusion of the websocket module here might be easier, but happy however it works out (as mentioned, these are optional modules and just improve performance) . Cheers and thanks again for all your efforts, it is very much appreciated.

nxhack, 2019-07-26:
I will close the issue. Don't hesitate to reopen it if you still have the issue.

rgillan, 2019-07-29:
So it turns out that those two inclusions don't work with the npm websocket module (as we think the author has made some changes to the code, so we can't simply move them into the build directory).  Is it possible for you to add the websocket module completely here at the high level?
Thanks again, you're a great help.

nxhack, 2019-07-30:
Oh... It was about the 'websocket' module, not the 'ws' module.

I made.

https://github.com/nxhack/openwrt-node-packages/tree/master/node-websocket

rgillan, 2019-07-30:
Awesome.  We'll add it to our planned build/test next week and let you know.  Appreciate the responsiveness and help

nxhack, 2019-11-27:
I will close the issue. Don't hesitate to re-open it if you still have the issue.

closed 2019-11-27 by nxhack