nxhack/openwrt-node-packages #536: @nxhack do you use multiple cores when you build this repo or just one core?

p3x-robot, 2019-07-01:


nxhack, 2019-07-01:
@p3x-robot 
My build environment is AWS EC2 m4.large. It is 2 core.
I'm using build toolchain as it is. So I think it's '-j1'.

p3x-robot, 2019-07-01:
so -j1 means 2 cores? just kinda weird, i tought 2 cores means -j2, but i suppose then -j1 means 0 and 1 - eg 2?

nxhack, 2019-07-01:
https://github.com/openwrt/openwrt/blob/master/include/package.mk#L23

p3x-robot, 2019-07-01:
the problem is, that when i build multiple cores in openwrt (when i build all packages), except for nodejs it works, because it knows about the dependencies, is it possible to add all packages to have a dependencies so that npm should be have a dependecies on node and the rest should be have dependencies on nodejs and npm, because then it would works with multiple cores, right now my problem is that my build would take 4 hours, but because this nodejs repo is not using these dependencies , it tries to build for example node-cpx before npm or nodejs is available so it gives an error.
i would love to do it, just tell me where i could put these, because then it would work with parallel building. do you know what i mean?

nxhack, 2019-07-01:
To build node-cpx, you need npm of host build. It may not be possible to build npm packages until npm can build it.

p3x-robot, 2019-07-01:
but can i somehow add a dependencies (nodejs, npm) on all these nodejs packages as a dependencies, because when i try to build with multiple cores/jobs it will throw an error because it starts to build any eg node-cpx, because nodejs and npm is not built.

if i would add these packages as a dependency on nodejs and npm, then the multiple jobs would work, do you know what i mean? right now because it only works with 1 job, i would like to build the node feeds with multiple jobs.
i do in openwrt ` IGNORE_ERRORS=1 make V=s  -j4` but it is not working with multiple jobs and only because of the nodejs packages all the rest is working (they have dependencies).

can we transform it the packages to be a dependencies of nodejs and npm? do you know what i mean?

nxhack, 2019-07-01:
The dependencies are explicitly described in the Makefile of npm modules.
```
PKG_BUILD_DEPENDS:=node/host
```

p3x-robot, 2019-07-01:
i see, but for some reason it is not working, maybe the node/host is built, but the npm is not built still, can we add the dependency for npm as well?

p3x-robot, 2019-07-01:
i see the error on the multiple jobs it is missing NPM... that is my issue

p3x-robot, 2019-07-01:
do you want to show me what the error is?

nxhack, 2019-07-01:
Yes Please.

p3x-robot, 2019-07-01:
ok, i am sending it to you , now building

p3x-robot, 2019-07-01:
here are the errors, it is missing the npm:
```text
make[3]: Entering directory '/build/source/feeds/node/node-browserify'
mkdir -p /build/source/dl
SHELL= flock /build/source/tmp/.browserify-16.2.3.tgz.flock -c '        /build/source/scripts/download.pl "/build/source/dl$
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://registry.npmjs.org/browserify/-/browserify-16.2.3.tgz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
^M  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0^M  0     0    0     0    0     0      0   $
install -d -m0755 /build/source/build_dir/hostpkg/node-browserify-16.2.3
touch /build/source/build_dir/hostpkg/node-browserify-16.2.3/.prepareddaa78577660823ecdee8eab81e37069d_6664517399ebbbc92a37$
(cd /build/source/build_dir/hostpkg/node-browserify-16.2.3/; if [ -x configure ]; then cp -fpR /build/source/scripts/config$
touch /build/source/build_dir/hostpkg/node-browserify-16.2.3/.configured
npm_config_nodedir=/build/source/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/usr/ npm_config_prefix=/build/source/stag$
bash: npm: command not found
Makefile:88: recipe for target '/build/source/build_dir/hostpkg/node-browserify-16.2.3/.built' failed
make[3]: *** [/build/source/build_dir/hostpkg/node-browserify-16.2.3/.built] Error 127
make[3]: Leaving directory '/build/source/feeds/node/node-browserify'
time: package/feeds/node/node-browserify/host-compile#0.16#0.03#5.77
```
```text
make[3]: Entering directory '/build/source/feeds/node/node-cpx'
mkdir -p /build/source/dl
SHELL= flock /build/source/tmp/.cpx-1.5.0.tgz.flock -c '        /build/source/scripts/download.pl "/build/source/dl" "cpx-1$
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://registry.npmjs.org/cpx/-/cpx-1.5.0.tgz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
^M  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0^M  0     0    0     0    0     0      0   $
install -d -m0755 /build/source/build_dir/hostpkg/node-cpx-1.5.0
touch /build/source/build_dir/hostpkg/node-cpx-1.5.0/.preparedb030254d5cc72eb029e366440a378efb_6664517399ebbbc92a37c5bb081b$
(cd /build/source/build_dir/hostpkg/node-cpx-1.5.0/; if [ -x configure ]; then cp -fpR /build/source/scripts/config.{guess,$
touch /build/source/build_dir/hostpkg/node-cpx-1.5.0/.configured
npm_config_nodedir=/build/source/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/usr/ npm_config_prefix=/build/source/stag$
bash: npm: command not found
Makefile:88: recipe for target '/build/source/build_dir/hostpkg/node-cpx-1.5.0/.built' failed
make[3]: *** [/build/source/build_dir/hostpkg/node-cpx-1.5.0/.built] Error 127
make[3]: Leaving directory '/build/source/feeds/node/node-cpx'
time: package/feeds/node/node-cpx/host-compile#0.14#0.02#5.73
```

nxhack, 2019-07-01:
Oh! understood.
Those npm modules also build the host package itself.
I will fix it from now on.

p3x-robot, 2019-07-01:
would you tell me when it is working? then i would try to build with multiple jobs? no rush, just asking.

nxhack, 2019-07-01:
@p3x-robot 
merged.  please test.

p3x-robot, 2019-07-01:
building

p3x-robot, 2019-07-01:
now, it is a different problem:
right now it looks like the node package is still building as the log is growing and now, i get this error:
```text
docker@7c0d52556493:/build/source/logs/package/feeds/node/node-browserify$ cat host-compile.txt 
make[3]: Entering directory '/build/source/feeds/node/node-browserify'
mkdir -p /build/source/dl
SHELL= flock /build/source/tmp/.browserify-16.2.3.tgz.flock -c '  	/build/source/scripts/download.pl "/build/source/dl" "browserify-16.2.3.tgz" "275b7748d4d169dd4fcc49c651ba944da798297e28b1197b2ecb1affd0f62b04" "" "https://registry.npmjs.org/browserify/-/"    '
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://registry.npmjs.org/browserify/-/browserify-16.2.3.tgz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  165k  100  165k    0     0  30076      0  0:00:05  0:00:05 --:--:-- 41080
install -d -m0755 /build/source/build_dir/hostpkg/node-browserify-16.2.3
touch /build/source/build_dir/hostpkg/node-browserify-16.2.3/.prepareddaa78577660823ecdee8eab81e37069d_6664517399ebbbc92a37c5bb081b5c53
(cd /build/source/build_dir/hostpkg/node-browserify-16.2.3/; if [ -x configure ]; then cp -fpR /build/source/scripts/config.{guess,sub} /build/source/build_dir/hostpkg/node-browserify-16.2.3// && CC="ccache gcc" CFLAGS="-O2 -I/build/source/staging_dir/host/include -I/build/source/staging_dir/hostpkg/include -I/build/source/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/host/include" CXX="ccache g++" CPPFLAGS="-I/build/source/staging_dir/host/include -I/build/source/staging_dir/hostpkg/include -I/build/source/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/host/include" LDFLAGS="-L/build/source/staging_dir/host/lib -L/build/source/staging_dir/hostpkg/lib -L/build/source/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/host/lib" CONFIG_SHELL="/usr/bin/env bash"  bash ./configure --target=x86_64-pc-linux-gnu --host=x86_64-pc-linux-gnu --build=x86_64-pc-linux-gnu --program-prefix="" --program-suffix="" --prefix=/build/source/staging_dir/hostpkg --exec-prefix=/build/source/staging_dir/hostpkg --sysconfdir=/build/source/staging_dir/hostpkg/etc --localstatedir=/build/source/staging_dir/hostpkg/var --sbindir=/build/source/staging_dir/hostpkg/bin ; fi )
touch /build/source/build_dir/hostpkg/node-browserify-16.2.3/.configured
npm_config_nodedir=/build/source/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/usr/ npm_config_prefix=/build/source/staging_dir/hostpkg/ npm_config_cache=/build/source/tmp/npm-cache-7X52ReTGa5 npm_config_tmp=/build/source/tmp/npm-tmp-7X52ReTGa5 npm install -g /build/source/dl/browserify-16.2.3.tgz
npm info it worked if it ends with ok
npm info using npm@6.9.0
npm info using node@v12.4.0
npm timing stage:loadCurrentTree Completed in 68ms
npm timing stage:loadIdealTree:cloneCurrentTree Completed in 0ms
npm timing stage:loadIdealTree:loadShrinkwrap Completed in 1ms
npm http fetch GET 200 https://registry.npmjs.org/JSONStream 5131ms
npm http fetch GET 200 https://registry.npmjs.org/assert 5166ms
npm http fetch GET 200 https://registry.npmjs.org/crypto-browserify 36ms
npm http fetch GET 200 https://registry.npmjs.org/crypto-browserify/-/crypto-browserify-3.12.0.tgz 38ms
npm http fetch GET 200 https://registry.npmjs.org/defined 36ms
npm http fetch GET 200 https://registry.npmjs.org/deps-sort 28ms
npm http fetch GET 200 https://registry.npmjs.org/domain-browser 39ms
npm http fetch GET 200 https://registry.npmjs.org/deps-sort/-/deps-sort-2.0.0.tgz 26ms
npm http fetch GET 200 https://registry.npmjs.org/duplexer2 25ms
npm http fetch GET 200 https://registry.npmjs.org/events 25ms
npm http fetch GET 200 https://registry.npmjs.org/glob 29ms
npm http fetch GET 200 https://registry.npmjs.org/has 22ms
npm http fetch GET 200 https://registry.npmjs.org/htmlescape 25ms
npm http fetch GET 200 https://registry.npmjs.org/https-browserify 24ms
npm http fetch GET 200 https://registry.npmjs.org/htmlescape/-/htmlescape-1.1.1.tgz 29ms
npm http fetch GET 200 https://registry.npmjs.org/inherits 34ms
npm http fetch GET 200 https://registry.npmjs.org/insert-module-globals 29ms
npm http fetch GET 200 https://registry.npmjs.org/labeled-stream-splicer 23ms
npm http fetch GET 200 https://registry.npmjs.org/mkdirp 25ms
npm http fetch GET 200 https://registry.npmjs.org/module-deps 36ms
npm http fetch GET 200 https://registry.npmjs.org/mkdirp/-/mkdirp-0.5.1.tgz 34ms
npm http fetch GET 200 https://registry.npmjs.org/os-browserify 23ms
npm http fetch GET 200 https://registry.npmjs.org/parents 25ms
npm http fetch GET 200 https://registry.npmjs.org/path-browserify 29ms
npm http fetch GET 200 https://registry.npmjs.org/process 26ms
npm http fetch GET 200 https://registry.npmjs.org/punycode 22ms
npm http fetch GET 200 https://registry.npmjs.org/querystring-es3 23ms
npm http fetch GET 200 https://registry.npmjs.org/read-only-stream 24ms
npm http fetch GET 200 https://registry.npmjs.org/readable-stream 27ms
npm http fetch GET 200 https://registry.npmjs.org/resolve 29ms
npm http fetch GET 200 https://registry.npmjs.org/shasum 27ms
npm http fetch GET 200 https://registry.npmjs.org/shell-quote 26ms
npm http fetch GET 200 https://registry.npmjs.org/shasum/-/shasum-1.0.2.tgz 35ms
npm http fetch GET 200 https://registry.npmjs.org/stream-browserify 33ms
npm http fetch GET 200 https://registry.npmjs.org/stream-http 28ms
[... cut ...]
npm ERR! A complete log of this run can be found in:
npm ERR!     /build/source/tmp/npm-cache-XAqNO1klQO/_logs/2019-07-01T12_43_53_520Z-debug.log
Makefile:89: recipe for target '/build/source/build_dir/hostpkg/node-cpx-1.5.0/.built' failed
make[3]: *** [/build/source/build_dir/hostpkg/node-cpx-1.5.0/.built] Error 254
make[3]: Leaving directory '/build/source/feeds/node/node-cpx'
time: package/feeds/node/node-cpx/host-compile#5.41#1.21#117.24
docker@7c0d52556493:/build/source/logs/package/feeds/node/node-cpx$ 
```

p3x-robot, 2019-07-01:
what is good, that now it builds node and npm (the log is growing still), but it is weird why it is still not working perfectly like all dependencies (all other packages).
[image]

before, it was building node-browserify and node-cpx without even having a node folder.

p3x-robot, 2019-07-01:
in the node folder what is happening is there are to log files:
[image]

the host-compile.txt is done:
[image]

but for the compile.txt is still growing, it is still at some point:
[image]

do you have an idea why it is happening?

p3x-robot, 2019-07-01:
the log error.txt file right now looks like this:
```text
   ERROR: package/feeds/packages/protobuf-c [host] failed to build.
   ERROR: package/feeds/packages/hamlib failed to build.
   ERROR: package/feeds/packages/open2300 failed to build.
   ERROR: package/feeds/packages/classpath failed to build.
   ERROR: package/feeds/packages/openssh failed to build (build variant: with-pam).
   ERROR: package/feeds/node/node-cpx [host] failed to build.
   ERROR: package/feeds/node/node-browserify [host] failed to build.
```

p3x-robot, 2019-07-01:
now finally, the node compile.txt is ready, now i am expecting the new packages should work:
[image]

p3x-robot, 2019-07-01:
the rest as it works as i was doing it in serial, but instead of 17 hours it took 4 hours...
looks like somehow the compile.txt should be done before the npm packages should be building.
do you know what i mean?

p3x-robot, 2019-07-01:
https://cdn.corifeus.com/openwrt/18.06.4/targets/mvebu/error-packages.txt -  you  can see were serial and no host job error
https://cdn.corifeus.com/openwrt/18.06.4/targets/mvebu/error-packages.txt - you can see were multiple jobs and 2 host job npm error, weird, huh?

[image]

nxhack, 2019-07-01:
https://github.com/nxhack/openwrt-node-packages/issues/464#issuecomment-486926142

p3x-robot, 2019-07-02:
but how come it is still not building? some host compile error:
```text
  ERROR: package/feeds/node/node-cpx [host] failed to build.
  ERROR: package/feeds/node/node-browserify [host] failed to build.
```

And it builds before node is totally compiling. I can ee the log the node compile.txt is still growing and that why is has a host faild to build. that is the question. when i build serially it works 100%.

p3x-robot, 2019-07-02:
i think you did not look at my comments careful. i run 4 jobs and it is still compiling node , when it start building `node-cpx` and `node-browserify`, these packages are still not correct from the building point. these should start building when `build` is 100% built, not during. the rest is all good.

nxhack, 2019-07-02:
@p3x-robot 

I changed the part that seems to be related.
Test please.

p3x-robot, 2019-07-02:
ok, building, thanks

p3x-robot, 2019-07-02:
weird, same error:
```text
   ERROR: package/feeds/node/node-authenticate-pam failed to build.
   ERROR: package/feeds/node/node-bignum failed to build.
   ERROR: package/feeds/node/node-bluetooth-hci-socket failed to build.
   ERROR: package/feeds/node/node-browserify [host] failed to build.
   ERROR: package/feeds/node/node-cpx [host] failed to build.
   ERROR: package/feeds/node/node-ejdb failed to build.
   ERROR: package/feeds/node/node-fuse-bindings failed to build.
   ERROR: package/feeds/node/node-hashtable failed to build.
   ERROR: package/feeds/node/node-mdns failed to build.
   ERROR: package/feeds/node/node-mknod failed to build.
   ERROR: package/feeds/node/node-openzwave-shared failed to build.
   ERROR: package/feeds/node/node-pi-spi failed to build.
   ERROR: package/feeds/node/node-pty failed to build.
   ERROR: package/feeds/node/node-rpio failed to build.
   ERROR: package/feeds/node/node-statvfs failed to build.
   ERROR: package/feeds/packages/classpath failed to build.
   ERROR: package/feeds/packages/dmapd failed to build.
   ERROR: package/feeds/packages/grilo-plugins failed to build.
   ERROR: package/feeds/packages/hamlib failed to build.
   ERROR: package/feeds/packages/lcdgrilo failed to build.
   ERROR: package/feeds/packages/libmraa failed to build.
   ERROR: package/feeds/packages/libupm failed to build.
   ERROR: package/feeds/packages/open2300 failed to build.
   ERROR: package/feeds/packages/openssh failed to build (build variant: with-pam).
   ERROR: package/feeds/packages/perl-device-usb failed to build.
   ERROR: package/feeds/packages/perl-inline [host] failed to build.
   ERROR: package/feeds/packages/protobuf-c [host] failed to build.
   ERROR: package/feeds/routing/pimbd failed to build.
   ERROR: package/network/services/omcproxy failed to build.
```

`browserify` and `cpx` is giving a `host failed to build` and it is because node is the `host-compile` is done but, then actual final `compile` is not done.

the host-compile is done at 08:06 AM, but the compile is actually is done at 09:08 AM:
[image]

nxhack, 2019-07-03:
Please show me detailed log on the construction of "node-cpx".

p3x-robot, 2019-07-03:
ok, i deleted, but i am sending you in about 2-3 hours.

p3x-robot, 2019-07-03:
ok, same result. what is see is happening is, that there are 2 logs for node:
compile.txt, host-compile.txt
it completes the host-compile.txt, when that is done, it start to do the next node package -> node-browserfy and i can see the node/compile.txt is starts to fill the log there when node-browserify starts and it giving an error, then going to node-cpx, same error [host] fialed built error.

how can we trigger to build the node/compile.txt before it starts the node-cpx and the other, here is the log, but it is the same, but in the package error.txt is shows it is a host error:

error.txt
```text
  ERROR: package/feeds/packages/protobuf-c [host] failed to build.
   ERROR: package/feeds/packages/hamlib failed to build.
   ERROR: package/feeds/packages/open2300 failed to build.
   ERROR: package/feeds/packages/classpath failed to build.
   ERROR: package/feeds/packages/openssh failed to build (build variant: with-pam).
   ERROR: package/feeds/node/node-cpx [host] failed to build.
   ERROR: package/feeds/node/node-browserify [host] failed to build.
```

node-browserfy
```text
docker@1c55aa7c49a2:/build/source/logs/package/feeds/node/node-browserify$ cat host-compile.txt 
make[3]: Entering directory '/build/source/feeds/node/node-browserify'
mkdir -p /build/source/dl
SHELL= flock /build/source/tmp/.browserify-16.2.3.tgz.flock -c '  	/build/source/scripts/download.pl "/build/source/dl" "browserify-16.2.3.tgz" "275b7748d4d169dd4fcc49c651ba944da798297e28b1197b2ecb1affd0f62b04" "" "https://registry.npmjs.org/browserify/-/"    '
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://registry.npmjs.org/browserify/-/browserify-16.2.3.tgz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  165k  100  165k    0     0  30133      0  0:00:05  0:00:05 --:--:-- 41190
install -d -m0755 /build/source/build_dir/hostpkg/node-browserify-16.2.3
touch /build/source/build_dir/hostpkg/node-browserify-16.2.3/.prepareddaa78577660823ecdee8eab81e37069d_6664517399ebbbc92a37c5bb081b5c53
(cd /build/source/build_dir/hostpkg/node-browserify-16.2.3/; if [ -x configure ]; then cp -fpR /build/source/scripts/config.{guess,sub} /build/source/build_dir/hostpkg/node-browserify-16.2.3// && CC="ccache gcc" CFLAGS="-O2 -I/build/source/staging_dir/host/include -I/build/source/staging_dir/hostpkg/include -I/build/source/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/host/include" CXX="ccache g++" CPPFLAGS="-I/build/source/staging_dir/host/include -I/build/source/staging_dir/hostpkg/include -I/build/source/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/host/include" LDFLAGS="-L/build/source/staging_dir/host/lib -L/build/source/staging_dir/hostpkg/lib -L/build/source/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/host/lib" CONFIG_SHELL="/usr/bin/env bash"  bash ./configure --target=x86_64-pc-linux-gnu --host=x86_64-pc-linux-gnu --build=x86_64-pc-linux-gnu --program-prefix="" --program-suffix="" --prefix=/build/source/staging_dir/hostpkg --exec-prefix=/build/source/staging_dir/hostpkg --sysconfdir=/build/source/staging_dir/hostpkg/etc --localstatedir=/build/source/staging_dir/hostpkg/var --sbindir=/build/source/staging_dir/hostpkg/bin ; fi )
touch /build/source/build_dir/hostpkg/node-browserify-16.2.3/.configured
npm_config_nodedir=/build/source/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/usr/ npm_config_prefix=/build/source/staging_dir/hostpkg/ npm_config_cache=/build/source/tmp/npm-cache-uQvGFM3NE7 npm_config_tmp=/build/source/tmp/npm-tmp-uQvGFM3NE7 npm install -g /build/source/dl/browserify-16.2.3.tgz
npm info it worked if it ends with ok
npm info using npm@6.9.0
npm info using node@v12.4.0
npm timing stage:loadCurrentTree Completed in 71ms
npm timing stage:loadIdealTree:cloneCurrentTree Completed in 0ms
npm timing stage:loadIdealTree:loadShrinkwrap Completed in 1ms
npm http fetch GET 200 https://registry.npmjs.org/assert 5144ms
npm http fetch GET 200 https://registry.npmjs.org/JSONStream 5155ms
npm http fetch GET 200 https://registry.npmjs.org/crypto-browserify 33ms
npm http fetch GET 200 https://registry.npmjs.org/defined 29ms
npm http fetch GET 200 https://registry.npmjs.org/crypto-browserify/-/crypto-browserify-3.12.0.tgz 33ms
npm http fetch GET 200 https://registry.npmjs.org/domain-browser 24ms
npm http fetch GET 200 https://registry.npmjs.org/duplexer2 35ms
npm http fetch GET 200 https://registry.npmjs.org/events 22ms
npm http fetch GET 200 https://registry.npmjs.org/glob 26ms
npm http fetch GET 200 https://registry.npmjs.org/has 24ms
npm http fetch GET 200 https://registry.npmjs.org/deps-sort 912ms
npm http fetch GET 200 https://registry.npmjs.org/deps-sort/-/deps-sort-2.0.0.tgz 37ms
npm http fetch GET 200 https://registry.npmjs.org/htmlescape 794ms
npm http fetch GET 200 https://registry.npmjs.org/https-browserify 24ms
npm http fetch GET 200 https://registry.npmjs.org/htmlescape/-/htmlescape-1.1.1.tgz 36ms
npm http fetch GET 200 https://registry.npmjs.org/inherits 28ms
npm http fetch GET 200 https://registry.npmjs.org/insert-module-globals 805ms
npm http fetch GET 200 https://registry.npmjs.org/labeled-stream-splicer 805ms
npm http fetch GET 200 https://registry.npmjs.org/mkdirp 25ms
npm http fetch GET 200 https://registry.npmjs.org/mkdirp/-/mkdirp-0.5.1.tgz 48ms
npm http fetch GET 200 https://registry.npmjs.org/os-browserify 22ms
[... cut ...]
npm ERR! A complete log of this run can be found in:
npm ERR!     /build/source/tmp/npm-cache-ngU09tQCnK/_logs/2019-07-03T05_18_53_015Z-debug.log
Makefile:92: recipe for target '/build/source/build_dir/hostpkg/node-cpx-1.5.0/.built' failed
make[3]: *** [/build/source/build_dir/hostpkg/node-cpx-1.5.0/.built] Error 254
make[3]: Leaving directory '/build/source/feeds/node/node-cpx'
time: package/feeds/node/node-cpx/host-compile#5.20#1.29#118.59
docker@1c55aa7c49a2:/build/source/logs/package/feeds/node/node-cpx$ 
```

nxhack, 2019-07-03:
hmm
> npm ERR! enoent ENOENT: no such file or directory, rename '/build/source/staging_dir/hostpkg/lib/node_modules/cpx/node_modules/arr-flatten' -> '/build/source/staging_dir/hostpkg/lib/node_modules/cpx/node_modules/.arr-flatten.DELETE'

I will check it

p3x-robot, 2019-07-03:
i installed this package on my workstation and i get tons of warning of `rename`:
```text
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@^1.0.0 (node_modules/chokidar/node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.2.9: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"x64"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: abbrev@1.1.1 (node_modules/fsevents/node_modules/abbrev):
npm WARN enoent SKIPPING OPTIONAL DEPENDENCY: ENOENT: no such file or directory, rename '/home/patrikx3/Downloads/package/node_modules/fsevents/node_modules/abbrev' -> '/home/patrikx3/Downloads/package/node_modules/fsevents/node_modules/.abbrev.DELETE'
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: ansi-regex@2.1.1 (node_modules/fsevents/node_modules/ansi-regex):
npm WARN enoent SKIPPING OPTIONAL DEPENDENCY: ENOENT: no such file or directory, rename '/home/patrikx3/Downloads/package/node_modules/fsevents/node_modules/ansi-regex' -> '/home/patrikx3/Downloads/package/node_modules/fsevents/node_modules/.ansi-regex.DELETE'
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: aproba@1.2.0 (node_modules/fsevents/node_modules/aproba):
npm WARN enoent SKIPPING OPTIONAL DEPENDENCY: ENOENT: no such file or directory, rename '/home/patrikx3/Downloads/package/node_modules/fsevents/node_modules/aproba' -> '/home/patrikx3/Downloads/package/node_modules/fsevents/node_modules/.aproba.DELETE'
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: balanced-match@1.0.0 (node_modules/fsevents/node_modules/balanced-match):
npm WARN enoent SKIPPING OPTIONAL DEPENDENCY: ENOENT: no such file or directory, rename '/home/patrikx3/Downloads/package/node_modules/fsevents/node_modules/balanced-match' -> '/home/patrikx3/Downloads/package/node_modules/fsevents/node_modules/.balanced-match.DELETE'
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: chownr@1.1.1 (node_modules/fsevents/node_modules/chownr):
npm WARN enoent SKIPPING OPTIONAL DEPENDENCY: ENOENT: no such file or directory, rename '/home/patrikx3/Downloads/package/node_modules/fsevents/node_modules/chownr' -> '/home/patrikx3/Downloads/package/node_modules/fsevents/node_modules/.chownr.DELETE'
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: code-point-at@1.1.0 (node_modules/fsevents/node_modules/code-point-at):
npm WARN enoent SKIPPING OPTIONAL DEPENDENCY: ENOENT: no such file or directory, rename '/home/patrikx3/Downloads/package/node_modules/fsevents/node_modules/code-point-at' -> '/home/patrikx3/Downloads/package/node_modules/fsevents/node_modules/.code-point-at.DELETE'
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: concat-map@0.0.1 (node_modules/fsevents/node_modules/concat-map):
npm WARN enoent SKIPPING OPTIONAL DEPENDENCY: ENOENT: no such file or directory, rename '/home/patrikx3/Downloads/package/node_modules/fsevents/node_modules/concat-map' -> '/home/patrikx3/Downloads/package/node_modules/fsevents/node_modules/.concat-map.DELETE'
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: console-control-strings@1.1.0 (node_modules/fsevents/node_modules/console-control-strings):
npm WARN enoent SKIPPING OPTIONAL DEPENDENCY: ENOENT: no such file or directory, rename '/home/patrikx3/Downloads/package/node_modules/fsevents/node_modules/console-control-strings' -> '/home/patrikx3/Downloads/package/node_modules/fsevents/node_modules/.console-control-strings.DELETE'
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: core-util-is@1.0.2 (node_modules/fsevents/node_modules/core-util-is):
npm WARN enoent SKIPPING OPTIONAL DEPENDENCY: ENOENT: no such file or directory, rename '/home/patrikx3/Downloads/package/node_modules/fsevents/node_modules/core-util-is' -> '/home/patrikx3/Downloads/package/node_modules/fsevents/node_modules/.core-util-is.DELETE'
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: deep-extend@0.6.0 (node_modules/fsevents/node_modules/deep-extend):
npm WARN enoent SKIPPING OPTIONAL DEPENDENCY: ENOENT: no such file or directory, rename '/home/patrikx3/Downloads/package/node_modules/fsevents/node_modules/deep-extend' -> '/home/patrikx3/Downloads/package/node_modules/fsevents/node_modules/.deep-extend.DELETE'
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: delegates@1.0.0 (node_modules/fsevents/node_modules/delegates):
npm WARN enoent SKIPPING OPTIONAL DEPENDENCY: ENOENT: no such file or directory, rename '/home/patrikx3/Downloads/package/node_modules/fsevents/node_modules/delegates' -> '/home/patrikx3/Downloads/package/node_modules/fsevents/node_modules/.delegates.DELETE'
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: detect-libc@1.0.3 (node_modules/fsevents/node_modules/detect-libc):
npm WARN enoent SKIPPING OPTIONAL DEPENDENCY: ENOENT: no such file or directory, rename '/home/patrikx3/Downloads/package/node_modules/fsevents/node_modules/detect-libc' -> '/home/patrikx3/Downloads/package/node_modules/fsevents/node_modules/.detect-libc.DELETE'
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fs.realpath@1.0.0 (node_modules/fsevents/node_modules/fs.realpath):
npm WARN enoent SKIPPING OPTIONAL DEPENDENCY: ENOENT: no such file or directory, rename '/home/patrikx3/Downloads/package/node_modules/fsevents/node_modules/fs.realpath' -> '/home/patrikx3/Downloads/package/node_modules/fsevents/node_modules/.fs.realpath.DELETE'
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: has-unicode@2.0.1 (node_modules/fsevents/node_modules/has-unicode):
[... cut ...]
npm WARN enoent SKIPPING OPTIONAL DEPENDENCY: ENOENT: no such file or directory, rename '/home/patrikx3/Downloads/package/node_modules/fsevents/node_modules/wrappy' -> '/home/patrikx3/Downloads/package/node_modules/fsevents/node_modules/.wrappy.DELETE'
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: yallist@3.0.3 (node_modules/fsevents/node_modules/yallist):
npm WARN enoent SKIPPING OPTIONAL DEPENDENCY: ENOENT: no such file or directory, rename '/home/patrikx3/Downloads/package/node_modules/fsevents/node_modules/yallist' -> '/home/patrikx3/Downloads/package/node_modules/fsevents/node_modules/.yallist.DELETE'
```

Is in it something with the rename that is not working with the building?

nxhack, 2019-07-03:
I did not found root cause, but I changed a little. Please test if you can.

p3x-robot, 2019-07-03:
building

p3x-robot, 2019-07-03:
ok, the `node-cpx` now built.

so does `node-browserify` has the same changes as in `node-cpx`?

the only thing i did, i used 6 cores, instead of 4...

now, only `node-browserify` could not build:
```text
docker@bd99082b3202:/build/source/logs/package/feeds/node/node-browserify$ cat host-compile.txt 
make[3]: Entering directory '/build/source/feeds/node/node-browserify'
mkdir -p /build/source/dl
SHELL= flock /build/source/tmp/.browserify-16.2.3.tgz.flock -c '  	/build/source/scripts/download.pl "/build/source/dl" "browserify-16.2.3.tgz" "275b7748d4d169dd4fcc49c651ba944da798297e28b1197b2ecb1affd0f62b04" "" "https://registry.npmjs.org/browserify/-/"    '
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://registry.npmjs.org/browserify/-/browserify-16.2.3.tgz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  165k  100  165k    0     0  30091      0  0:00:05  0:00:05 --:--:-- 41110
install -d -m0755 /build/source/build_dir/hostpkg/node-browserify-16.2.3
touch /build/source/build_dir/hostpkg/node-browserify-16.2.3/.prepareddaa78577660823ecdee8eab81e37069d_6664517399ebbbc92a37c5bb081b5c53
(cd /build/source/build_dir/hostpkg/node-browserify-16.2.3/; if [ -x configure ]; then cp -fpR /build/source/scripts/config.{guess,sub} /build/source/build_dir/hostpkg/node-browserify-16.2.3// && CC="ccache gcc" CFLAGS="-O2 -I/build/source/staging_dir/host/include -I/build/source/staging_dir/hostpkg/include -I/build/source/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/host/include" CXX="ccache g++" CPPFLAGS="-I/build/source/staging_dir/host/include -I/build/source/staging_dir/hostpkg/include -I/build/source/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/host/include" LDFLAGS="-L/build/source/staging_dir/host/lib -L/build/source/staging_dir/hostpkg/lib -L/build/source/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/host/lib" CONFIG_SHELL="/usr/bin/env bash"  bash ./configure --target=x86_64-pc-linux-gnu --host=x86_64-pc-linux-gnu --build=x86_64-pc-linux-gnu --program-prefix="" --program-suffix="" --prefix=/build/source/staging_dir/hostpkg --exec-prefix=/build/source/staging_dir/hostpkg --sysconfdir=/build/source/staging_dir/hostpkg/etc --localstatedir=/build/source/staging_dir/hostpkg/var --sbindir=/build/source/staging_dir/hostpkg/bin ; fi )
touch /build/source/build_dir/hostpkg/node-browserify-16.2.3/.configured
CFLAGS="-O2 -I/build/source/staging_dir/host/include -I/build/source/staging_dir/hostpkg/include -I/build/source/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/host/include" CPPFLAGS="-I/build/source/staging_dir/host/include -I/build/source/staging_dir/hostpkg/include -I/build/source/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/host/include" CXXFLAGS="" LDFLAGS="-L/build/source/staging_dir/host/lib -L/build/source/staging_dir/hostpkg/lib -L/build/source/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/host/lib" npm_config_nodedir=/build/source/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/usr/ npm_config_prefix=/build/source/staging_dir/hostpkg/ npm_config_cache=/build/source/tmp/npm-cache-48A1jLCVhj npm_config_tmp=/build/source/tmp/npm-tmp-48A1jLCVhj npm install -g /build/source/dl/browserify-16.2.3.tgz
npm info it worked if it ends with ok
npm info using npm@6.9.0
npm info using node@v12.4.0
npm timing stage:loadCurrentTree Completed in 85ms
npm timing stage:loadIdealTree:cloneCurrentTree Completed in 1ms
npm timing stage:loadIdealTree:loadShrinkwrap Completed in 1ms
npm http fetch GET 200 https://registry.npmjs.org/JSONStream 5141ms
npm http fetch GET 200 https://registry.npmjs.org/assert 5153ms
npm http fetch GET 200 https://registry.npmjs.org/crypto-browserify 29ms
npm http fetch GET 200 https://registry.npmjs.org/defined 23ms
npm http fetch GET 200 https://registry.npmjs.org/crypto-browserify/-/crypto-browserify-3.12.0.tgz 42ms
npm http fetch GET 200 https://registry.npmjs.org/deps-sort 37ms
npm http fetch GET 200 https://registry.npmjs.org/domain-browser 28ms
npm http fetch GET 200 https://registry.npmjs.org/deps-sort/-/deps-sort-2.0.0.tgz 44ms
npm http fetch GET 200 https://registry.npmjs.org/duplexer2 30ms
npm http fetch GET 200 https://registry.npmjs.org/events 51ms
npm http fetch GET 200 https://registry.npmjs.org/glob 45ms
npm http fetch GET 200 https://registry.npmjs.org/has 23ms
npm http fetch GET 200 https://registry.npmjs.org/htmlescape 23ms
npm http fetch GET 200 https://registry.npmjs.org/https-browserify 61ms
npm http fetch GET 200 https://registry.npmjs.org/htmlescape/-/htmlescape-1.1.1.tgz 45ms
npm http fetch GET 200 https://registry.npmjs.org/inherits 25ms
npm http fetch GET 200 https://registry.npmjs.org/insert-module-globals 38ms
npm http fetch GET 200 https://registry.npmjs.org/labeled-stream-splicer 30ms
npm http fetch GET 200 https://registry.npmjs.org/mkdirp 32ms
npm http fetch GET 200 https://registry.npmjs.org/module-deps 27ms
npm http fetch GET 200 https://registry.npmjs.org/mkdirp/-/mkdirp-0.5.1.tgz 29ms
npm http fetch GET 200 https://registry.npmjs.org/os-browserify 26ms
npm http fetch GET 200 https://registry.npmjs.org/parents 27ms
npm http fetch GET 200 https://registry.npmjs.org/path-browserify 26ms
npm http fetch GET 200 https://registry.npmjs.org/process 24ms
npm http fetch GET 200 https://registry.npmjs.org/punycode 31ms
npm http fetch GET 200 https://registry.npmjs.org/querystring-es3 22ms
[... cut ...]
npm ERR!     /build/source/tmp/npm-cache-48A1jLCVhj/_logs/2019-07-03T10_51_31_120Z-debug.log
Makefile:94: recipe for target '/build/source/build_dir/hostpkg/node-browserify-16.2.3/.built' failed
make[3]: *** [/build/source/build_dir/hostpkg/node-browserify-16.2.3/.built] Error 254
make[3]: Leaving directory '/build/source/feeds/node/node-browserify'
time: package/feeds/node/node-browserify/host-compile#4.63#0.96#125.98
docker@bd99082b3202:/build/source/logs/package/feeds/node/node-browserify$ 
```

nxhack, 2019-07-03:
Thank you for the test.
Next I will change the package structure. It will take some time.

p3x-robot, 2019-07-05:
with 8 cores, an no host error:
```text
   ERROR: package/feeds/node/node-authenticate-pam failed to build.
   ERROR: package/feeds/node/node-bignum failed to build.
   ERROR: package/feeds/node/node-bluetooth-hci-socket failed to build.
   ERROR: package/feeds/node/node-ejdb failed to build.
   ERROR: package/feeds/node/node-fuse-bindings failed to build.
   ERROR: package/feeds/node/node-hashtable failed to build.
   ERROR: package/feeds/node/node-mdns failed to build.
   ERROR: package/feeds/node/node-mknod failed to build.
   ERROR: package/feeds/node/node-openzwave-shared failed to build.
   ERROR: package/feeds/node/node-pi-spi failed to build.
   ERROR: package/feeds/node/node-pty failed to build.
   ERROR: package/feeds/node/node-rpio failed to build.
   ERROR: package/feeds/node/node-statvfs failed to build.
   ERROR: package/feeds/packages/classpath failed to build.
   ERROR: package/feeds/packages/dmapd failed to build.
   ERROR: package/feeds/packages/grilo-plugins failed to build.
   ERROR: package/feeds/packages/hamlib failed to build.
   ERROR: package/feeds/packages/lcdgrilo failed to build.
   ERROR: package/feeds/packages/libmraa failed to build.
   ERROR: package/feeds/packages/libupm failed to build.
   ERROR: package/feeds/packages/open2300 failed to build.
   ERROR: package/feeds/packages/openssh failed to build (build variant: with-pam).
   ERROR: package/feeds/packages/perl-device-usb failed to build.
   ERROR: package/feeds/packages/perl-parse-recdescent [host] failed to build.
   ERROR: package/feeds/packages/protobuf-c [host] failed to build.
   ERROR: package/feeds/routing/pimbd failed to build.
   ERROR: package/network/services/omcproxy failed to build.
```

p3x-robot, 2019-07-06:
this package is perfect, i used 9 jobs and no host error at all.

closed 2019-07-06 by p3x-robot